shivasiddharth/GassistPi #915: Can't get the Installer to work properly after inputting info

Cave300, 2019-06-13:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Board Info:
Board Info:
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000004c0b919

Os-Info:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
```

### Describe the bug:
```
Once i run the Installer executable it asks "Enter the your full credential file name including the path and .json extension: " when i put what i feel should be there it just results in the installation to say "Error: Got unexpected extra argument Error: Got unexpected extra argument (/home/pi/client_secret_227937457499-o0ujo46l648uth9enjl6d4ohll1pnre6.apps.googleusercontent.com.json" extension when it asks for it. (also i hope the bot doesn't close this due to it "not following the template" when in the first one i sent it was in the format just modified to remove unnecessary questions.)

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
[... 247 lines cut ...]
### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
No Log file was found while searching sorry for the inconvenience.

Cave300, 2019-06-14:
oh... okay. i'll try and fix this in a bit to see if it works. busy at the moment. Thanks for the quick reply. I'll comment when i've tried it.

Cave300, 2019-06-16:
Okay, haven't tried it yet sorry, but i haven't been able to try it due to a family member currently being hospitalized. please don't close this issue as it isn't resolved just yet . Thanks and have a good day!

Cave300, 2019-06-18:
it worked. up until i pasted the auth code. Once i did it just says no such file or directory. for some reason it came up with `ValueError: Please supply either code or Authorization_response parameters`
Should i try running it again? i'm gonna try and redo the setup to see if it changes.

Cave300, 2019-06-18:
ok, so it got father this time, but it just gave errors for a bit then ended.

Cave300, 2019-06-21:
I'm pretty sure i'm gonna have to retry doing everything... well i'll try it in a bit i guess.

shivasiddharth, 2019-06-24:
I really need to take a look at the terminal ouput. That conveys more info to me than the description of the error in English.

gnanasreekar, 2019-07-05:
Can you tell me the exact format of the input given when the script is executed 
i've tried /home/pi/...........json but it dint work
i got file does not exist error

shivasiddharth, 2019-07-05:
if you get a file does not exist prompt then it means that the credentials file not located in the directory that you have mentioned or the credentials file's name is different from what you have supplied.

gnanasreekar, 2019-07-05:
it is in /home/pi
i've specified the correct location

gnanasreekar, 2019-07-05:
I literally seriously appreciate your fast reply..means a lot

gnanasreekar, 2019-07-05:
pi@raspberrypi:~ $ sudo  ./GassistPi/scripts/gassist-installer.sh   
This script must run as pi, trying to change user...

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com.json 

Enter the your Google Cloud Console Project-Id: homeassi-d4aeb

Enter the modelid that was generated in the actions console: homeassi-d4aeb-home-assi-nvalbc

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://deb.nodesource.com/node_10.x stretch InRelease                          
Hit:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease                      
Reading package lists... Done                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u1).
libssl-dev is already the newest version (1.1.0k-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u1).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.7-0+rpt1+deb9u1).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

===========Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Ok-Google hotword==========
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (19.1.1)
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (41.0.1)
Requirement already up-to-date: wheel in ./env/lib/python3.5/site-packages (0.33.4)
[... 149 lines cut ...]
Requirement already satisfied: requests>=2.0.0 in ./env/lib/python3.5/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (2.22.0)
Requirement already satisfied: oauthlib>=3.0.0 in ./env/lib/python3.5/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (3.0.2)
Requirement already satisfied: protobuf>=3.6.0 in ./env/lib/python3.5/site-packages (from googleapis-common-protos>=1.5.2->google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (3.8.0)
Requirement already satisfied: asn1crypto>=0.21.0 in ./env/lib/python3.5/site-packages (from cryptography>=1.3.4; extra == "secure"->urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (0.24.0)
Requirement already satisfied: pycparser in ./env/lib/python3.5/site-packages (from CFFI>=1.0->sounddevice<0.4,>=0.3.7; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (2.19)
Requirement already satisfied: pyasn1<0.5.0,>=0.4.1 in ./env/lib/python3.5/site-packages (from pyasn1-modules>=0.2.1->google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (0.4.5)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in ./env/lib/python3.5/site-packages (from requests>=2.0.0->requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (3.0.4)
Requirement already satisfied: setuptools in ./env/lib/python3.5/site-packages (from protobuf>=3.6.0->googleapis-common-protos>=1.5.2->google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5. [... cut ...]
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/tool/__main__.py", line 100, in main
    scopes=scope
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 180, in from_client_secrets_file
    with open(client_secrets_file, 'r') as json_file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com.json'
pi@raspberrypi:~ $

gnanasreekar, 2019-07-05:
my home folder

pi@raspberrypi:~ $ cd /home/pi
pi@raspberrypi:~ $ ls
audiosetup
client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com .json
credentials.json
Desktop
Documents
Downloads
env
GassistPi
MagicMirror
MagPi
modelid.txt
Music
Pictures
Public
Templates
Videos

shivasiddharth, 2019-07-05:
> my home folder
pi@raspberrypi:~ $ cd /home/pi
pi@raspberrypi:~ $ ls
audiosetup
client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com .json

I see a space between "com" and ".json" which is missing in your argument.

gnanasreekar, 2019-07-05:
Same error

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com.json             

Enter the your Google Cloud Console Project-Id: homeassi-d4aeb

Enter the modelid that was generated in the actions console: homeassi-d4aeb-home-assi-nvalbc

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_10.x stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u1).
libssl-dev is already the newest version (1.1.0k-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u1).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.7-0+rpt1+deb9u1).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

===========Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Ok-Google hotword==========
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (19.1.1)
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (41.0.1)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))': /simple/wheel/
Requirement already up-to-date: wheel in ./env/lib/python3.5/site-packages (0.33.4)
[... 149 lines cut ...]
Requirement already satisfied: oauthlib>=3.0.0 in ./env/lib/python3.5/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (3.0.2)
Requirement already satisfied: requests>=2.0.0 in ./env/lib/python3.5/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (2.22.0)
Requirement already satisfied: protobuf>=3.6.0 in ./env/lib/python3.5/site-packages (from googleapis-common-protos>=1.5.2->google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (3.8.0)
Requirement already satisfied: pycparser in ./env/lib/python3.5/site-packages (from CFFI>=1.0->sounddevice<0.4,>=0.3.7; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (2.19)
Requirement already satisfied: asn1crypto>=0.21.0 in ./env/lib/python3.5/site-packages (from cryptography>=1.3.4; extra == "secure"->urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (0.24.0)
Requirement already satisfied: pyasn1>=0.1.3 in ./env/lib/python3.5/site-packages (from rsa>=3.1.4->google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (0.4.5)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in ./env/lib/python3.5/site-packages (from requests>=2.0.0->requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (3.0.4)
[... cut ...]
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/tool/__main__.py", line 100, in main
    scopes=scope
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 180, in from_client_secrets_file
    with open(client_secrets_file, 'r') as json_file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com.json'
pi@raspberrypi:~ $

gnanasreekar, 2019-07-05:
Without space

pi@raspberrypi:~ $ sudo  ./GassistPi/scripts/gassist-installer.sh 
This script must run as pi, trying to change user...

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_442256639019-1h0hnco4h8hoeguaviq05sebc9vhin20.apps.googleusercontent.com .json

Error: Got unexpected extra argument (.json)
pi@raspberrypi:~ $

gnanasreekar, 2019-07-05:
Got it..THanks a ton

gnanasreekar, 2019-07-05:
It's asking me to open the Google home app 
but the home app does not find any devices

gnanasreekar, 2019-07-05:
Where is the config file

shivasiddharth, 2019-07-05:
Please check the src sub-folder

gnanasreekar, 2019-07-05:
i cant find the gpio 05 and 06 pins

shivasiddharth, 2019-07-05:
These are really very basic things and if you look at it properly you should find it https://github.com/shivasiddharth/GassistPi/blob/f4f7971adc13af484310e6cf535490b633e05e52/src/config.yaml#L101 

Do you see a 05 and 06 under this ?

gnanasreekar, 2019-07-05:
I'm now to this, i ain't getting it

gnanasreekar, 2019-07-05:
And is there any way that I can communicate with Arduino UNO r3 using HC-05 Bluetooth module to send strings so that I can control the lights and fans of my room

gnanasreekar, 2019-07-06:
Can you tell me how to get the API key from google apis,i'm unable to find it

shivasiddharth, 2019-07-06:
You can directly send bluetooth command from Pi itself. Some posted that long back. Search for that in the issues.

alias13e, 2021-03-21:
Hi, I have a similar problem that comes up after entering the model ID and the installer runs.  I get the long url to get the authorization code, but when I go and sign in, it says access denied and tells me to email the developer to get permission and directs me to my own email. Seems like somewhere in the cloud console needs to be clicked, but I’ve been through the tutorial twice and I don’t see what I’m missing. 

Thanks in advance your work here is amazing!