shivasiddharth/GassistPi #918: Installation Issue (Read Time out)

Friday8229, 2019-06-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000054341172

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[... 201 lines cut ...]

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
NO LOGS

shivasiddharth, 2019-06-28:
It is due to a very slow internet or internet connection interruption. There is nothing that can be done from my side for this.

closed 2019-06-28 by shivasiddharth

Friday8229, 2019-06-30:
I do not think that this is a network problem I tried with a faster internet and got another error that is 

pi@raspberrypi:~ $ sudo  ./GassistPi/scripts/gassist-installer.sh   
This script must run as pi, trying to change user...

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_656068528074-3id67a127rbf5159l4rgvdjbqcfhe9dc.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: zero-38896

Enter the modelid that was generated in the actions console: zero-38896-zero-38896-y1hare

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease    
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease      
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u1).
libssl-dev is already the newest version (1.1.0j-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u1).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.7-0+rpt1+deb9u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

===========Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Ok-Google hotword==========
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (19.1.1)
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (41.0.1)
[... 132 lines cut ...]
Collecting beautifulsoup4 (from MechanicalSoup>=0.4.0->gmusicapi>=11.1.1->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/1d/5d/3260694a59df0ec52f8b4883f5d23b130bc237602a1411fa670eae12351e/beautifulsoup4-4.7.1-py3-none-any.whl
Collecting ifaddr (from zeroconf>=0.17.7->PyChromecast>=2.3.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 9))
  Using cached https://www.piwheels.org/simple/ifaddr/ifaddr-0.1.6-py3-none-any.whl
Collecting cachetools>=2.0.0 (from google-auth>=1.4.1->google-api-python-client>=1.7.5->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 13))
  Using cached https://files.pythonhosted.org/packages/2f/a6/30b0a0bef12283e83e58c1d6e7b5aabc7acfc4110df81a4471655d33e704/cachetools-3.1.1-py2.py3-none-any.whl
Collecting pytz (from google-api-core[grpc]<2.0.0dev,>=1.6.0->google-cloud-texttospeech==0.3.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 25))
  Using cached https://files.pythonhosted.org/packages/3d/73/fe30c2daaaa0713420d0382b16fbb761409f532c56bdcc514bf7b6262bb6/pytz-2019.1-py2.py3-none-any.whl
Collecting googleapis-common-protos!=1.5.4,<2.0dev,>=1.5.3 (from google-api-core[grpc]<2.0.0dev,>=1.6.0->google-cloud-texttospeech==0.3.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 25))
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))': /simple/googleapis-common-protos/
[... cut ...]
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    lxml from https://www.piwheels.org/simple/lxml/lxml-4.3.4-cp35-cp35m-linux_armv7l.whl#sha256=0eaeef1fa87a2ef490acd39acc236b0e48324c728a9f92bc1e8d44a610d8aeaf (from MechanicalSoup>=0.4.0->gmusicapi>=11.1.1->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 5)):
        Expected sha256 0eaeef1fa87a2ef490acd39acc236b0e48324c728a9f92bc1e8d44a610d8aeaf
             Got        1f9446db068e20ab758c308dcb0f9ea5ff6cdc454d3add6337d22340f93839ab

shivasiddharth, 2019-06-30:
>>I do not think that this is a network problem I tried with a faster internet and got another error that is

Clearly it has not timed out with a faster network. So the previous error that you reported was a network issue.  Clear your temporary files for a clean install of gmusicapi