shivasiddharth/GassistPi #920: GassistPi starts with mic muted and mutes again after commands

tmdelia, 2019-07-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000046807142

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
[... 121 lines cut ...]
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: g-pi-a2db1-g-pi-ovbx76
device_id: D10F19973BFF3D10FCBA16BAA5C3D4E5

    This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": true}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  

[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3366348/GassistPi.log)

Daandeve, 2019-07-17:
I have exactly the same problem, any fix for this issue?

shivasiddharth, 2019-07-17:
Went through #879 ?

tmdelia, 2019-07-17:
Yes, been through #879. It's a similar but different issue.

Daandeve, 2019-07-17:
Ah, yes i see now. But when the mic is muted the custom voice wakeup commands do not work...

shivasiddharth, 2019-07-17:
@tmdelia , @Daandeve ,

1. In the config file check if you have disabled Ok_Google hotword, if you have disabled it then, the assistant will start muted only. 

2. If you have not disabled it and still if assistant is starting muted, open the /home/pi/ directory and choose to view hidden files and then delete the .mute file. Now if you restart your assistant it should start with mic turned on.

3. If you dont have the .mute file, you need to connect a pushbutton to the GPIO indicated in the README to unmute it.

4. If you cannot connect a pushbutton first disable the Ok_Google hotword in the config file, start the assistant, close the assistant, enable the OK_Google hotword again and restart the assistant.

Daandeve, 2019-07-17:
The Ok_Google word works when it is not muted. But when it is muted the script should recognize my own hotwords right?

tmdelia, 2019-07-18:
Okay_Google hotword is enabled:

#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'

------

When I start the assistant:

ON_MUTED_CHANGED:
  {"is_muted": true}

------

No .mute file in /home/pi, no pushbutton to connect to the GPIO.

Tried #4, still staring muted.

shivasiddharth, 2019-07-18:
After this line https://github.com/shivasiddharth/GassistPi/blob/fceb968663c6d80172f4535bb8ba8c5170e2dd12/src/main.py#L300

add 
`self.assistant.set_mic_mute(False)`

Now start the assistant. After starting the assistant, without closing the assistant remove that line in the script and save it.

Daandeve, 2019-07-18:
That works for me!

But you said in #879 that when Ok_Google is disabled your custom wakeword sould activate it, even when it is muted

tmdelia, 2019-07-18:
Thanks, worked for me too!

tmdelia, 2019-07-18:
@Daandeve The custom wakeword will still activate it, even when it is muted. You will just get a response right before the ding "the mic is back on" and then after the response "the mic is turned off", which is a bit annoying, but it is what it is. You can set all the instances of self.assistant.set_mic_mute(True) to self.assistant.set_mic_mute(False) in main.py if you don't want the mic to mute when using a custom wakeword, but I don't know if there is a bigger reason for muting the mic only when custom wakeword is on and Ok Google is disabled.

Daandeve, 2019-07-18:
Yeah i know but my custom wake words ain't working.
The standard ones don't work, i also created my own but they don't seem to work also...

tmdelia, 2019-07-18:
Alexa doesn't work for you? Snoboy didn't work for me, and I created a custom wakeword, but it only works some of the time of since I am the only user using the wakeword so it doesn't have many users, but Alexa works every time for me. You could also try downloading a wakeword that has a bunch of users and it will be more likely to work.

Daandeve, 2019-07-18:
Yeah i tried jarvis but no success :((

tmdelia, 2019-07-18:
Hmmm... yeah odd then. Ok Google works everytime for you?

Daandeve, 2019-07-18:
Yeah every time.

tmdelia, 2019-07-18:
Hmmm... this seems to be a different issue, related to the custom wakeword itself and not the mic muting issue.

shivasiddharth, 2019-07-18:
Other than muting the assistant, there is no way to turn Ok_Google off. This option is for all those privacy minded people. 

As the primary issue has been resolved closing this thread. Check out #505  for custom wakeword thingy.

closed 2019-07-18 by shivasiddharth