shivasiddharth/GassistPi #924: Segmentation fault with hotword

sethlloydcollins, 2019-07-10:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-07-10:
Can you try using Raspbian stretch ? Buster seems to be having issues.

sethlloydcollins, 2019-07-11:
Stretch has solved this issue. Now getting the service stopping when I issue a "set radio volume to 50%". Worth mentioning that "set volume to 50%" doesnt change radio (or any audio that is playing) volume.

shivasiddharth, 2019-07-11:
Thats a wrong command for setting audio. Please go through the readme for correct syntax. https://github.com/shivasiddharth/GassistPi#ccontrolling-media-by-voice

shivasiddharth, 2019-07-12:
Can you go into this path `/home/pi/env/lib/python3.7/site-packages/google/assistant/library/` get the assistant.py file and __init__ file and attach it here. I need these to see as to whats causing it to crash in buster.

sethlloydcollins, 2019-07-13:
Unfortunately I dont have the buster image anymore. I started afresh with stretch. 

Ps this issue can be closed, it is resolved

closed 2019-07-13 by sethlloydcollins

neethilal, 2019-07-29:
> Unfortunately I dont have the buster image anymore. I started afresh with stretch.
> 
> Ps this issue can be closed, it is resolved

pls sent me the link for downloading stretch neethilalbck@gmail.com