shivasiddharth/GassistPi #925: 33/5000 When I run the assistant, I have an error

Mezirx, 2019-07-16:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 000000004f588fe8

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:
```
[... 235 lines cut ...]
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9999] Unanticipated host error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1026, in <module>
    logger.exception(error)
NameError: name 'logger' is not defined
```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  

The content of this file is empty

shivasiddharth, 2019-07-16:
Please check your snowboy models that you have entered in your config file.

Mezirx, 2019-07-16:
I did not make any changes to the configuration file, I configured everything according to the instructions.

shivasiddharth, 2019-07-16:
Did you install pulseaudio ? If you did please remove that.

Mezirx, 2019-07-17:
when I enter
`apt-get remove --purge pulseaudio`
I will get the answer

`The "pulseaudio" package is not installed, so it will not be removed`

shivasiddharth, 2019-07-17:
Try using a different Mic.

serhatkimya, 2019-07-26:
Now İ have same error

Mezirx, 2019-07-26:
I checked on another sound card and microphone, I still have this error.

serhatkimya, 2019-07-26:
in this errors line 987 and 1024 ,so i understant that 987 is about "credentials, device_model_id" problem, but never changed anything and why this problem become?

gsg2016, 2019-07-29:
Hi Guys
I'm getting a similar error, 

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-07-29 22:00:24,613 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01354368] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0136aff0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
[... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Fatal Python error: Segmentation fault

Current thread 0x76f86ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 98 in __init__
  File "/home/pi/GassistPi/src/main.py", line 987 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Segmentation fault

can anyone help?

JahFyahh, 2021-03-12:
Hi Guys,

Any updates on this issue? I installed it all today (multiple times) but when starting using the service i'm also getting these errors. It seems to be restarting itself every minute. 

I'm using a Raspberry Pi Zero W with the 2 mic respeaker. Both the sound and mic work. When the service starts, Google says hi and one minute later again and this continues (nobody is that friendly).

This error keeps repeating itself:

Mar 12 19:18:06 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 12 19:18:32 pi-google-assist python[1682]: /home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255:            UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
Mar 12 19:18:32 pi-google-assist python[1682]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
Mar 12 19:18:32 pi-google-assist python[1682]: 2021-03-12 19:18:32,949 - gmusicapi.Mobileclient1 (shared:176) [WARNING           ]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Mar 12 19:18:39 pi-google-assist python[1682]: [01ccaa60] vlcpulse audio output error: PulseAudio server connection fa           ilure: Connection refused
Mar 12 19:18:40 pi-google-assist python[1682]: [01cd1130] vlcpulse audio output error: PulseAudio server connection fa           ilure: Connection refused
Mar 12 19:18:40 pi-google-assist python[1682]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 12 19:18:50 pi-google-assist python[1682]: INFO:snowboy:Keyword 1 detected at time: 2021-03-12 19:18:50
Mar 12 19:18:51 pi-google-assist python[1682]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_Be           ginPolling: Assertion `ret == self->nfds' failed.
Mar 12 19:18:51 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 12 19:18:51 pi-google-assist systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 12 19:18:51 pi-google-assist systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 12 19:18:51 pi-google-assist systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 24.
Mar 12 19:18:51 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.

Hope to hear something soon.

Thanks

JahFyahh, 2021-03-13:
So i ran the command `/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/Extras/gmusicauth.py` and got rid of the mobileclient.cred error, but genereted a whole new one. This time google is returning "Unusual traffic from your computer network". Any thougts?

```
Mar 13 12:26:39 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 13 12:27:13 pi-google-assist python[1058]: Traceback (most recent call last):
Mar 13 12:27:13 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/shared.py", line 216, in perform
Mar 13 12:27:13 pi-google-assist python[1058]:     response.raise_for_status()
Mar 13 12:27:13 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/requests/models.py", line 943, in raise_for_status
Mar 13 12:27:14 pi-google-assist python[1058]:     raise HTTPError(http_error_msg, response=self)
Mar 13 12:27:14 pi-google-assist python[1058]: requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo?alt=json&hl=en_US&dv=0&tier=fr
Mar 13 12:27:14 pi-google-assist python[1058]: During handling of the above exception, another exception occurred:
Mar 13 12:27:14 pi-google-assist python[1058]: Traceback (most recent call last):
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/GassistPi/src/pushbutton.py", line 41, in <module>
Mar 13 12:27:14 pi-google-assist python[1058]:     from actions import say
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
Mar 13 12:27:14 pi-google-assist python[1058]:     from mediaplayer import api
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/GassistPi/src/mediaplayer.py", line 23, in <module>
Mar 13 12:27:14 pi-google-assist python[1058]:     for device in api.get_registered_devices():
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/mobileclient.py", line 880, in get_registered_devices
Mar 13 12:27:14 pi-google-assist python[1058]:     res = self._make_call(mobileclient.GetDeviceManagementInfo)
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/mobileclient.py", line 43, in _make_call
Mar 13 12:27:14 pi-google-assist python[1058]:     return super()._make_call(protocol, *args, **kwargs)
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/shared.py", line 83, in _make_call
Mar 13 12:27:14 pi-google-assist python[1058]:     return protocol.perform(self.session, self.validate, *args, **kwargs)
Mar 13 12:27:14 pi-google-assist python[1058]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/shared.py", line 224, in perform
Mar 13 12:27:14 pi-google-assist python[1058]:     raise CallFailure(err_msg, call_name)
Mar 13 12:27:14 pi-google-assist python[1058]: gmusicapi.exceptions.CallFailure: GetDeviceManagementInfo: 404 Client Error: Not Found for url: https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo?alt=json&hl=en_US&dv=0&tier=fr
Mar 13 12:27:14 pi-google-assist python[1058]: (requests kwargs: {'method': 'GET', 'url': 'https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo', 'params': {'alt': 'json', 'hl': 'en_US', 'dv': 0, 'tier': 'fr'}, 'headers': {'Authorization': '<omitted>'}})
Mar 13 12:27:14 pi-google-assist python[1058]: (response was: '<html><head><meta http-equiv="content-type" content="text/html; charset=utf-8"/><title>Sorry...</title><style> body { font-family: verdana, arial, sans-serif; background-color: #fff; color: #000; }</style></head><body><div><table><tr><td><b><font face=sans-serif size=10><font color=#4285f4>G</font><font color=#ea4335>o</font><font color=#fbbc05>o</font><font color=#4285f4>g</font><font color=#34a853>l</font><font color=#ea4335>e</font></font></b></td><td style="text-align: left; vertical-align: bottom; padding-bottom: 15px; width: 50%"><div style="border-bottom: 1px solid #dfdfdf;">Sorry...</div></td></tr></table></div><div style="margin-left: 4em;"><h1>We\'re sorry...</h1><p>... but your computer or network may be sending automated queries. To protect our users, we can\'t process your request right now.</p></div><div style="margin-left: 4em;">See <a href="https://support.google.com/websearch/answer/86640">Google Help</a> for more information.<br/><br/></div><div style="text-align: center; border-top: 1px solid #dfdfdf;"><a href="https://www.google.com">Google Home</a></div></body></html>')
Mar 13 12:27:15 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Mar 13 12:27:15 pi-google-assist systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Mar 13 12:27:15 pi-google-assist systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 13 12:27:15 pi-google-assist systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 2.
Mar 13 12:27:15 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.

```

shivasiddharth, 2021-03-13:
This is really a old thread. A lot of things have changed since the OP opened up this issue. @JahFyahh open up a new issue.

closed 2021-03-13 by shivasiddharth