shivasiddharth/GassistPi #931: Segmentation fault when reinstalling with previously working credentials

shankargopal, 2019-07-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000098fd4098

```
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:

I had a working installation of GassistPi and it was running great. Then by accident I broke the memory card on which it was installed. I purchased a new memory card and did the following:

* Used the Raspbian install files I had already downloaded and installed
* Cloned GassistPi from git into the home directory
* Used the credentials file I had already created for Google
* Used the project ID and model ID I had already created
* Ran the audio script
* Ran the installation script

Now, every time I run the script, I get something like this:

```Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 225: 16249 Segmentation fault      googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).

Yes. I tried setting the locale to UTF8 in the console, and tried generating a new model ID and using that.  Same error (though the number before `Segmentation fault` changed, the line number etc. remained the same). 

### 2. Is the issue related to assistant installation ?

Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
As pasted above
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
N/A
### 4. Is the assistant service restarting automatically?
N/A
### If answer to question 4 is Yes, then paste the contents of the terminal below.
N/A
### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)
N/A
### Paste the command below, that crashed the assistant
N/A

### Paste the contents of the terminal below
N/A
### Attach the log file named Gassistpi.log
No such file

shivasiddharth, 2019-07-19:
> Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).

Duplicate of #917  Duplicate of #924

closed 2019-07-19 by shivasiddharth

shankargopal, 2019-07-19:
My apologies, I missed those two somehow and saw only the older ones.

On Fri, 19 Jul 2019 00:19:50 -0700
shivasiddharth <notifications@github.com> wrote:

> > Did you go through or search the issues section to check if your
> > issue was already discussed (either currently open issues or closed
> > issues).  
> 
> Duplicate of #917  Duplicate of #924 
> 

-- 

‌‌

shivasiddharth, 2019-07-19:
Seems like a python error. Hope to have it fixed soon.

Banhammer1919, 2019-09-05:
I get the same error when installing gassist under raspbian buster on the Pi4.  

./GassistPi/scripts/gassist-installer.sh: line 225: 9098 Segmentation fault googlesamples-assistant-hotword --project_id $ projid --device_model_id $modelid 
Is there a solution to the problem?