shivasiddharth/GassistPi #933: The credentials.json does not have its name

jbrepogmailcom, 2019-07-19:
The sentence "Place the credentials.json file in/home/${USER}/ directory DO NOT RENAME" is VERY confusing. When downloaded from google, the filename is not credentials.json, but something like client_secret_xxxxxxxxx.json. So should the filename be changed to credentials.json or not?

closed 2019-07-19 by close-issue-app[bot]

shivasiddharth, 2019-07-19:
It clearly asks you to not rename it.

jbrepogmailcom, 2019-07-19:
Hello,

I agree, it clearly asks me not to rename credentials.json file, but the file downloaded from google is like client_secret_913316289274-qh85gbsrdrjgesn6ipcomgf6tksku9dp.apps.googleusercontent.com <http://client_secret_913316289274-qh85gbsrdrjgesn6ipcomgf6tksku9dp.apps.googleusercontent.com/>.json.  So should I rename it to credentials.json and then do not rename it any further, or should I keep long name and ignore the sentence in tutorial that I should not rename file credentials.json - because there is no file like that?

Thanks, Jan

shivasiddharth, 2019-07-19:
https://github.com/shivasiddharth/GassistPi/commit/fceb968663c6d80172f4535bb8ba8c5170e2dd12 Hope it is clear now