shivasiddharth/GassistPi #934: Project_id from side channel and project_id from url do not match

jbrepogmailcom, 2019-07-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000335907fa
pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
### Describe the bug:
```
At end of installation, after entering code from web, I get error "Project_id from side channel and project_id from url do not match". The project ID that is contained in json file is R2D2-190720, but when getting code from URL in the url the  project is called "project 506305001375".
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Enter the authorization code: 4/igHmhN3IADAEOuSpN5H9bZ2tPtwvG1FMoMF5AqZeZ3mQqdoN0trEAKA
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
INFO:root:Connecting to embeddedassistant.googleapis.com
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
INFO:root:Registering device
ERROR:root:Failed to register device: {
  "error": {
    "code": 400,
    "message": "Could not create the device instance. Project_id from side channel and project_id from url do not match.",
    "status": "INVALID_ARGUMENT"
  }
}

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
It played intro sound, said "Hi" and then crashed:

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id R2D2-190720  --device-model-id r2d2-190720-raspi0w-7glm3f
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-07-19 13:10:07,774 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[00d9c850] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[00dd5270] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
ERROR:root:Failed to register device: {
  "error": {
    "code": 400,
    "message": "Could not create the device instance. Project_id from side channel and project_id from url do not match.",
    "status": "INVALID_ARGUMENT"
  }
}
```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  

no such log there

shivasiddharth, 2019-07-19:
> pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id R2D2-190720  --device-model-id r2d2-190720-raspi0w-7glm3f

Your project-id is in CAPS but the project id in the model-id is in lower case. Is that how you normally have ? 

Check if your credentials file is from the same project that you are using here. If you open your credentials file, you can find your model id inside.

jbrepogmailcom, 2019-07-19:
Great, it is fixed. I named the project in CAPS, but Google renamed it to lowercase and project ID has to be entered in lowercase despite the fact Google displays it in Uppercase...

closed 2019-07-19 by jbrepogmailcom