shivasiddharth/GassistPi #937: Installation on Orange Pi fails

jbrepogmailcom, 2019-07-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@orangepizero:~$ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: Allwinner sun8i Family
Revision	: 0000
Serial		: 02c00042f859dcc8
pi@orangepizero:~$ cat /etc/os-release  
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

### Describe the bug:
```
Error during installation, see terminal output:

Collecting cryptography==2.4.2 (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 28))
  Downloading https://files.pythonhosted.org/packages/f3/39/d3904df7c56f8654691c4ae1bdb270c1c9220d6da79bd3b1fbad91afd0e1/cryptography-2.4.2.tar.gz (468kB)
     |████████████████████████████████| 471kB 2.5MB/s 
  Installing build dependencies ... error
  ERROR: Complete output from command /home/pi/env/bin/python /home/pi/env/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-xcn7tvxj/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=18.5' wheel 'cffi>=1.7,!=1.11.3; python_implementation != '"'"'PyPy'"'"'':
  ERROR: Traceback (most recent call last):
    File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/home/pi/env/lib/python3.7/site-packages/pip/__main__.py", line 16, in <module>
      from pip._internal import main as _main  # isort:skip # noqa
    File "/home/pi/env/lib/python3.7/site-packages/pip/_internal/__init__.py", line 4, in <module>
      import locale
    File "/usr/lib/python3.7/locale.py", line 16, in <module>
      import re
    File "/usr/lib/python3.7/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
  ----------------------------------------
ERROR: Command "/home/pi/env/bin/python /home/pi/env/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-xcn7tvxj/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=18.5' wheel 'cffi>=1.7,!=1.11.3; python_implementation != '"'"'PyPy'"'"''" failed with error code 1 in None
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-07-19:
Run
sudo apt-get install libffi-dev
sudo apt-get install libssl-dev

And the. Re-run the installer.

jbrepogmailcom, 2019-07-19:
libffi-dev is already the newest version (3.2.1-9).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
libssl-dev is already the newest version (1.1.1c-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

shivasiddharth, 2019-07-19:
https://github.com/shivasiddharth/GassistPi#only-oses-suported-are You are running an OS that i dont support. I cannot do anything more. I already told you that https://github.com/shivasiddharth/GassistPi/issues/936#issuecomment-513355285 Get a working Armbian stretch

closed 2019-07-19 by shivasiddharth

jbrepogmailcom, 2019-07-19:
This was different board. I am sorry for bugging you and thanks for your time.

shivasiddharth, 2019-07-19:
Source and linux kernels are upstream. All boards will have a version of Buster. I am 100% ready to help people, i have attended to each and every issue posted by everyone, at the same time i also expect people to put in some effort.

jbrepogmailcom, 2019-07-19:
Armbian Stretch for Orange Pi Zero is hard to find at this time. Only Buster is offered for downloads

shivasiddharth, 2019-07-19:
Check #820 one person has got Stretch kernel 4.14.