shivasiddharth/GassistPi #938: Raspberry Zero - Illegal Instruction in Stretch

jbrepogmailcom, 2019-07-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000335907fa
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
### Describe the bug:
```
At end of installation there is Illegal instruction. Did not occur in Buster (but there was another problem):
Enter the authorization code: 4/igHwBhxvE7k986eBGCozJO4rvfkEbN6vSJzZz_DCnrEwx99zYmTtd1g
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 225: 24310 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
see above
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-07-20:
I need to see the full terminal content. Can you share that.

jbrepogmailcom, 2019-07-20:
Hello, here is full installation text:

Enter the your full credential file name including the path and .json extension: client_secret_506305001375-hoj9q92e70laicrn4vhg0nrp4c9hcmhc.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: r2d2-190720

Enter the modelid that was generated in the actions console: r2d2-190720-raspi0w-7glm3f

Hit:1 http://raspbian.raspberrypi.org/raspbian <http://raspbian.raspberrypi.org/raspbian> stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian <http://archive.raspberrypi.org/debian> stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u1).
libssl-dev is already the newest version (1.1.0k-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u1).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.7-0+rpt1+deb9u1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

==========Your board does not support Ok-Google Hotword. You need to trigger the assistant using pushbutton/custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Pushbutton/Custom-wakeword==========
Looking in indexes: https://pypi.org/simple <https://pypi.org/simple>, https://www.piwheels.org/simple <https://www.piwheels.org/simple>
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (19.1.1)
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (41.0.1)
Requirement already up-to-date: wheel in ./env/lib/python3.5/site-packages (0.33.4)
Looking in indexes: https://pypi.org/simple <https://pypi.org/simple>, https://www.piwheels.org/simple <https://www.piwheels.org/simple>
Collecting git+https://github.com/plamere/spotipy.git <git+https://github.com/plamere/spotipy.git> (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 17))
[... 122 lines cut ...]
Requirement already satisfied: requests-oauthlib>=0.7.0 in ./env/lib/python3.5/site-packages (from google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.5.1) (1.2.0)
Requirement already satisfied: CFFI>=1.0 in ./env/lib/python3.5/site-packages (from sounddevice<0.4,>=0.3.7; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (1.12.3)
Requirement already satisfied: googleapis-common-protos>=1.5.2 in ./env/lib/python3.5/site-packages (from google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (1.6.0)
Requirement already satisfied: grpcio>=1.3.5 in ./env/lib/python3.5/site-packages (from google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (1.22.0)
Requirement already satisfied: six in ./env/lib/python3.5/site-packages (from pathlib2<3,>=2.3.0; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (1.12.0)
Requirement already satisfied: pyOpenSSL>=0.14; extra == "secure" in ./env/lib/python3.5/site-packages (from urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (19.0.0)
Requirement already satisfied: idna>=2.0.0; extra == "secure" in ./env/lib/python3.5/site-packages (from urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (2.8)
[... cut ...] tzeiwVeM8v2QNCx1la79lVS4iLn3sMY&access_type=offline&prompt=consent&code_challenge_method=S256>
Enter the authorization code: 4/igHwBhxvE7k986eBGCozJO4rvfkEbN6vSJzZz_DCnrEwx99zYmTtd1g
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 225: 24310 Illegal instruction     googlesamples-assistant-pushtotalk --project-id $projid --device-model-id $modelid

Tuxliri, 2019-09-09:
I have the same problem on my raspberry pi 1b