shivasiddharth/GassistPi #940: RuntimeError: Python 3 was configured to use ASCII

jbrepogmailcom, 2019-07-20:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
Orange Pi Zero, latest Armbian Stretch
### Describe the bug:
```
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.  The recommendation about setting environment variables did not help. See terminal output
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Requirement already satisfied: setuptools in ./env/lib/python3.5/site-packages (from protobuf>=3.6.0->googleapis-common-protos>=1.5.2->google-assistant-grpc==0.2.1; extra == "samples"->google-assistant-sdk[samples]==0.5.1) (41.0.1)
Traceback (most recent call last):
  File "/home/pi/env/bin/google-oauthlib-tool", line 10, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 676, in main
    _verify_python3_env()
  File "/home/pi/env/lib/python3.5/site-packages/click/_unicodefun.py", line 118, in _verify_python3_env
    'for mitigation steps.' + extra)
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.  Consult http://click.pocoo.org/python3/for mitigation steps.

This system supports the C.UTF-8 locale which is recommended.
You might be able to resolve your issue by exporting the
following environment variables:

    export LC_ALL=C.UTF-8
    export LANG=C.UTF-8

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-07-20:
Run:

```
export LC_ALL=C.UTF-8
export LANG=C.UTF-8
```

jbrepogmailcom, 2019-07-20:
I have done that and then ran installer and it did not help. Should it probably be done after python switches to different environment?

jbrepogmailcom, 2019-07-20:
OK echo $LC_ALL returned just C, so I fixed it and running installer again...

jbrepogmailcom, 2019-07-20:
fixed

closed 2019-07-20 by jbrepogmailcom