shivasiddharth/GassistPi #943: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes.

tertianalpaca01, 2019-07-22:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pycryptodomex==3.7.2 from https://www.piwheels.org/simple/pycryptodomex/pycryptodomex-3.7.2-cp35-cp35m-linux_armv7l.whl#sha256=8bc23bc7ccec688f9ea3f4431a9c8785040d786fa5e36aec4a023b37c71297ae (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 22)):
        Expected sha256 8bc23bc7ccec688f9ea3f4431a9c8785040d786fa5e36aec4a023b37c71297ae
             Got        32ed51fe00e32a96bbf8bde0f1a1e68b8a548100de5cb43a2eb510eecf403275

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-07-22:
Go through this piece.

https://stackoverflow.com/questions/40183108/python-packages-hash-not-matching-whilst-installing-using-pip

tertianalpaca01, 2019-07-22:
Thanks Shivasiddharth/Gassistpi for your reply. Issue sorted out.

closed 2019-07-22 by shivasiddharth