shivasiddharth/GassistPi #949: YouTube

GeekTeacher, 2019-08-04:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-08-04:
There is no place in actions script to enter the API. You have got all the instructions wrong. Go through the latest set of instructions.

GeekTeacher, 2019-08-04:
These are YOUR instructions on https://www.instructables.com/id/Google-Assistant-on-Pi-for-Home-Automation-and-Ent/ - I refer you to point number 5 where it clearly says:

 **5. Click on create credentials and select API Key and choose close. Make a note of the created API Key and enter it in the actions.py script at the indicated location.**

It's point number 7 that I don't see how to do.

But the main issue is that YouTube is not playing music when Play from YouTube command spoken. GA says 'Getting YouTube Link' and then stops with a 'encountered an exception, please check the logs'. 

Please help me sort this.

Here's the full list of points which I am following from Instructibles.com

Adding YouTube API and Generating API Key

1. Go to the projects page on your Google Cloud Console-> https://console.cloud.google.com/project

2. Select your project from the list.

3. On the left top corner, click on the hamburger icon or three horizontal stacked lines.

4. Move your mouse pointer over "API and services" and choose "credentials".

**5. Click on create credentials and select API Key and choose close. Make a note of the created API Key and enter it in the actions.py script at the indicated location.**

6. "From the API and services" option, select library and in the search bar type youtube, select "YouTube Data API v3" API and click on "ENABLE".

7. In the API window, click on "All API Credentials" and in the drop down, make sure to have a tick (check mark) against the API Key that you just generated.

Default keyword for playing music from YouTube without autoplay is Stream.

For example, Stream I got you command will fetch Bebe Rexha's I Got You from YouTube.

Default keyword for playing music from YouTube with autoplay is Autoplay and Stream.

For example, Autoplay and Stream I got you command will play the requested I Got You and after the end of the track will autoplay susequent tracks. The number of autoplay tracks has been limited to a maximum of 10. this can be changed the under the YouTube_Autoplay function in the actions.py script.

GeekTeacher, 2019-08-04:
If this is no longer the correct set of instructions for adding YouTube functionality to your implementation of Google Assistant on a Raspberry Pi, please could you send me a link to the correct ones. Thank-you.

shivasiddharth, 2019-08-04:
I have got this project listed in a number of sites (hackster, instructables, etc) as the project advances, it is humanly impossible to go back and makes changes in each and every post/site. That is why i ask people to follow the README which always has the up to date instruction.

GeekTeacher, 2019-08-04:
Thanks for your reply.  Let me say that I really do appreciate all the work you've done and are still doing on this project. However, this stuff is not always easy for newbies like me and so I'd appreciate your help.  What you may think is obvious isn't always, even when people (like me) have some knowledge. But we are not immersed in this stuff everyday and so it can be difficult.
I've been trying unsuccessfully to set up the Alexa voice assistant on Raspberry Pi for the last four weeks. I really can't see why a totally clean installation of everything from scratch with all the latest updates shouldn't work - but it doesn't for me, and I am pretty meticulous when it comes to entering code scripts and instructions.
Yesterday I finally gave up and installed Google Assistant using your files.  I eventually got the basic installation to work.
Now I want to install a command to play music, and YouTube seems the easiest option here. I found some of your instructions on Instructibles for adding YouTube functionality.  They were fine until point 7, as I have mentioned on the GitHub issues page.  I have looked at the readme.file in GassistPi folder and it contains no information, merely pointing to your blog.  I have been to your blog to find the same instructions with the exception of the API key now being added to config.yaml instead of actions.py.  Is config.yaml the correct place for the API key? I've added it there too now, but it still doesn't work/
I don't understand what the API window means and so I haven't been able to do it.     
   - In the API window, click on "All API Credentials" and in the drop down, make sure to have a tick (check mark) against the API Key that you just generated.   

I can't see 'All API Credentials' anywhere in the 'API and Services' section. Has Google changed the screen layout or something - which is the API Window? 
Anyway, the bottom line is that YouTube isn't working on my Google Assistant implementation and so any help would be appreciated.  If there is a read.me file that contains up to date info on adding YouTube functionality please could you send me it.
As always, I appreciate your help
Martyn Jones

shivasiddharth, 2019-08-04:
Can you manually run the assistant and share the terminal contents here. I will do my best to assist.

GeekTeacher, 2019-08-04:
Thanks, here it is although I'm not sure it helps.  The 'weather and date' voice assistant response is fine.  The Play on YouTube voice assistant response is 'Sorry, something went wrong - please give it another try'.
Since adding the api key to the config.yaml script, the response seems to have changed.  Before I did use to get something like 'Searching YouTube links', but it still didn't find the song.
In your video you seem to have u in front of the commands i.e. {u"text": u"play I got you"} which I don't have.
Looking forward to your reply.
Martyn
pi@raspberrypi:~ $ sudo systemctl stop gassistpi.servicepi@raspberrypi:~ $ source env/bin/activate(env) pi@raspberrypi:~ $ google-assistant-demo --device-model-id micromartechodot-micromartedchodot-h4s2dzdevice_model_id: micromartechodot-micromartedchodot-h4s2dzdevice_id: 10EB81A48EB0B1AA8A36CF64A72FE280

    This device is not registered. This means you will not be able to use    Device Actions or see your device in Assistant Settings. In order to    register this device follow instructions at:
    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device
ON_MUTED_CHANGED:  {"is_muted": false}ON_START_FINISHEDON_MEDIA_STATE_IDLE
ON_CONVERSATION_TURN_STARTEDON_END_OF_UTTERANCEON_END_OF_UTTERANCEON_RECOGNIZING_SPEECH_FINISHED:  {"text": "weather and date"}ON_RESPONDING_STARTED:  {"is_error_response": false}ON_RESPONDING_FINISHEDON_CONVERSATION_TURN_FINISHED:  {"with_follow_on_turn": false}

ON_CONVERSATION_TURN_STARTEDON_END_OF_UTTERANCEON_END_OF_UTTERANCEON_RECOGNIZING_SPEECH_FINISHED:  {"text": "play on YouTube I got you"}ON_RESPONDING_STARTED:  {"is_error_response": false}ON_RESPONDING_FINISHEDON_CONVERSATION_TURN_FINISHED:  {"with_follow_on_turn": false}

 
 

YouTube Exception Error · Issue #949 · shivasiddharth/GassistPi

Build software better, together

GitHub is where people build software. More than 36 million people use GitHub to discover, fork, and contribute ...
 |

 |

 |

GeekTeacher, 2019-08-04:
I noticed the 'This device is not registered' message when I sent you the console text in my previous message.  As I've been running GA as a service on boot up, I've not been seeing any text in the console and I'd forgotten about this because the basic GA functions are working.  Could this be the cause of the errors? I've since been trying to register this device but can't see how. I'm not sure you can register a device after it's been created, only as you are creating it. Should I create a new device and generate some more credentials? Or can you tell me how to register an existing device please?

shivasiddharth, 2019-08-04:
Your syntax for the youtube command is wrong. Can you get the correct syntax from here https://gassistpi.blogspot.com/p/how-to-use-customizations.html and then try

GeekTeacher, 2019-08-05:
OK, many thanks, it's appreciated.
Martyn

scientificpraveen, 2021-04-01:
> Your syntax for the youtube command is wrong. Can you get the correct syntax from here https://gassistpi.blogspot.com/p/how-to-use-customizations.html and then try

I am getting this error 

<HttpError 403 when requesting https://youtube.googleapis.com/youtube/v3/search?q=faded&part=id%2Csnippet&key=AIzaSyDmpLwvE5xdtdxoIM1qa7FJGPNDEXKUGBI&alt=json returned "The request cannot be completed because you have exceeded your <a href="/youtube/v3/getting-started#quota">quota</a>.". Details: "The request cannot be completed because you have exceeded your <a href="/youtube/v3/getting-started#quota">quota</a>.">

I checked the quota in cloud platform and it says 

YouTube Data API v3 | Queries per 100 seconds | One quota is within limit

Limit:3,000,000Current usage: 0(0%)
7 days peak: 0(0%)

Phil-MC, 2021-05-02:
I am also getting this error when trying to ask for a song from Youtube
I tried increasing my quota from 0 on the Google Project page but the system would not have it as I don't have a billing address. Which is not surprising!

Here is an extract from the syslog.

> May  2 15:29:33 SlideShow python[480]: congratulations
May  2 15:29:33 SlideShow python[480]: <HttpError 403 when requesting https://youtube.googleapis.com/youtube/v3/search?q=congratulations&part=id%2Csnippet&key=AIzaSyDTz5Wx9p33zezQ2HxT5qwndYBg_w1eCg4&alt=json returned "The request cannot be completed because you have exceeded your <a href="/youtube/v3/getting-started#quota">quota</a>.". Details: "[{'message': 'The request cannot be completed because you have exceeded your <a href="/youtube/v3/getting-started#quota">quota</a>.', 'domain': 'youtube.quota', 'reason': 'quotaExceeded'}]">