shivasiddharth/GassistPi #962: Add ubuntu x64 pc support

rt400, 2019-08-29:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 60
model name	: Intel(R) Core(TM) i5-4430 CPU @ 3.00GHz
stepping	: 3
microcode	: 0x27
cpu MHz		: 1261.439
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips	: 5986.35
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

```

### Describe the bug:
```
Add support for Ubuntu x64 pc
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes
### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
g++ -I../../ -O3 -fPIC -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++0x  -shared snowboy-detect-swig.o \
../..//lib/ubuntu64/libsnowboy-detect.a -L/usr/local/lib/python3.7/config-3.7m-x86_64-linux-gnu -L/usr/local/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm  -lm -ldl -lf77blas -lcblas -llapack -latlas -o _snowboydetect.so
/usr/bin/ld: /usr/local/lib/python3.7/config-3.7m-x86_64-linux-gnu/libpython3.7m.a(abstract.o): relocation R_X86_64_PC32 against symbol `PyObject_SetItem' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
Makefile:73: recipe for target '_snowboydetect.so' failed
make: *** [_snowboydetect.so] Error 1

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-08-29 16:39:44 IDT; 4s ago
  Process: 12996 ExecStart=/home/yuval/env/bin/python -u /home/yuval/GassistPi/src/main.py --device_model_id mej-e3a01-warehouseassisnt-sv293w (code=exited, status=1/FAILURE)
 Main PID: 12996 (code=exited, status=1/FAILURE)

Aug 29 16:39:44 server systemd[1]: gassistpi.service: Service hold-off time over, scheduling restart.
Aug 29 16:39:44 server systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Aug 29 16:39:44 server systemd[1]: Stopped Google Assistant-Voice.
Aug 29 16:39:44 server systemd[1]: gassistpi.service: Start request repeated too quickly.
Aug 29 16:39:44 server systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Aug 29 16:39:44 server systemd[1]: Failed to start Google Assistant-Voice.

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
 /home/$USER$/env/bin/python3 -u /home/$USER$/GassistPi/src/main.py --device_model_id 'device_model_id'
```

### Paste the contents of the terminal below
```
Traceback (most recent call last):
  File "/home/yuval/GassistPi/src/main.py", line 37, in <module>
    import snowboydecoder
  File "/home/yuval/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/yuval/GassistPi/src/snowboydetect.py", line 28, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/yuval/GassistPi/src/snowboydetect.py", line 24, in swig_import_helper
    _mod = imp.load_module('_snowboydetect', fp, pathname, description)
  File "/usr/local/lib/python3.7/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/local/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: dynamic module does not define module export function (PyInit__snowboydetect)

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

lordjoo, 2020-11-28:
Same here i want to test it on my ubuntu vm but I can't 💔

JLFN, 2020-12-22:
Did you try to disable snowboy from the code?

mc9625, 2021-04-18:
Actually I have the same issue on Raspberry OS 64bit