shivasiddharth/GassistPi #963: Fails to install, nothing happens when it says "testing the installed google assistant"

shankargopal, 2019-08-30:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000098fd4098

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
[... 260 lines cut ...]
### 4. Is the assistant service restarting automatically?
Yes/No - N/A

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
N/A
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)
N/A
### Paste the command below, that crashed the assistant
```
N/A
```

### Paste the contents of the terminal below
```
N/A
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
No file present

Nualp83, 2019-09-02:
My issue is the same except it has a segmentation fault right after i put in my authorization code.
Enter the authorization code: 4/q***************************
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 225:  2498 Segmentation fault      googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid

datawire1337, 2019-09-03:
Having the same issue after the installation:

> ./GassistPi/scripts/gassist-installer.sh: line 225: 11303 Segmentation fault      googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid

When running the assistant manually afterwards:

Fatal Python error: Segmentation fault

> Current thread 0x76f99ad0 (most recent call first):
>   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 98 in __init__
**>   File "/home/pi/GassistPi/src/main.py", line 987 in main
>   File "/home/pi/GassistPi/src/main.py", line 1024 in <module>**
> Segmentation fault

The lines point out to the device_model_id variable.

Banhammer1919, 2019-09-06:
Same issue here on raspian buster. /GassistPi/scripts/gassist-installer.sh: line 225: 11303 Segmentation fault googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid

Tuxliri, 2019-09-09:
I have the same problem, have you found a solution?

shivasiddharth, 2019-09-12:
Can you manually run the assistant ?

shivasiddharth, 2019-09-24:
@shankargopal , Do we have an update, coz I am surprised that GA is not working with even stretch.

shankargopal, 2019-09-24:
Very sorry am still traveling. Will give you an update by Thursday definitely. Thank you for all your work as always.

kuduvasen83, 2019-10-19:
Currently this Code is not working anymore. I have tried yesterday with a fresh copy of Raspberry OS.
1. After successful initialization it has given same error which has mention has Shankargopal, 

2. Then try to start manually with "googlesamples-assistant-hotword --project-id my-dev-project --device-model-id my-model" but it gave Segmentation fault. 

3. Finally I followed the instruction mentioned in google developer link "https://developers.google.com/assistant/sdk/guides/library/python/embed/install-sample". Then I can start the assistant with "googlesamples-assistant-hotword --project-id my-dev-project --device-model-id my-model"

4. Now I tried to enable & start as Service, but Status shows failed to start

5. Then I ran the cmd manually with python , python2, python2.7 , but it gave below error
	/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'pi-xxxx-googlepi-xxxx'
	Traceback (most recent call last):
	File "/home/pi/GassistPi/src/main.py", line 18, in <module>
    import faulthandler
	ImportError: No module named faulthandler
	
6. Then I ran the cmd manually with python , but gave below error
	/home/pi/env/bin/python3 -u /home/pi/GassistPi/src/main.py --device_model_id 'pi-xxxx-googlepi-xxxx'
	Traceback (most recent call last):
	  File "/home/pi/GassistPi/src/main.py", line 47, in <module>
		from google.cloud import speech
	  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech.py", line 19, in <module>
		from google.cloud.speech_v1 import SpeechClient
	  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/__init__.py", line 17, in <module>
		from google.cloud.speech_v1.gapic import speech_client
	  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/gapic/speech_client.py", line 22, in <module>
		import google.api_core.gapic_v1.client_info
	  File "/home/pi/env/lib/python3.7/site-packages/google/api_core/gapic_v1/__init__.py", line 16, in <module>
		from google.api_core.gapic_v1 import config
	  File "/home/pi/env/lib/python3.7/site-packages/google/api_core/gapic_v1/config.py", line 23, in <module>
		import grpc
	  File "/home/pi/env/lib/python3.7/site-packages/grpc/__init__.py", line 23, in <module>
		from grpc._cython import cygrpc as _cygrpc
	ImportError: /home/pi/env/lib/python3.7/site-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_exchange_8

do we need to install python 3.7? Please advise. Thanks in advance

kiam001, 2019-12-03:
Same Problem

commit 26e3fbcf (2020-02-18): Bumped the project to work with Raspbian Buster

Fixes #987  Fixes #982  Fixes #963  Fixes #976
closed 2020-02-18 by shivasiddharth