shivasiddharth/GassistPi #966: can gassistpi work offline without google assistant just executing local commands like controlling home with ESPs.

nclbill, 2019-09-24:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
assistant dont crash

```

### Paste the contents of the terminal below
assistant dont crash
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3649005/GassistPi.log)

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
# #

nclbill, 2019-09-24:
Hi Mrs can gassistpi work offline without google assistant just executing local commands like controlling home with ESPs.
(i already did the automatisations using ESPs, to control light, tv, ac, and publishing in mqtt locally from different sensors i used say query like you advised me for alert and perform actions when conditions is true , now my assistant is ready .)
is it possible to run it locally without internet, just custom actions, perhaps later activate google assistant with a custom action and stop it with an other custom action leaving the custom actions always on. tks

nclbill, 2019-09-24:
sorry i gess its not the right way to ask for feature request !!!!?

shivasiddharth, 2019-09-25:
Can you gitter for non-issue stuff. Its really difficult for me to monitor non-issue and issue stuff here.

closed 2019-09-25 by shivasiddharth