shivasiddharth/GassistPi #987: 'mock' error

adas4190, 2020-02-17:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000b2702ac6

```
Paste your Board and OS details here.   
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[... 778 lines cut ...]
Collecting future
  Downloading https://www.piwheels.org/simple/future/future-0.18.2-py3-none-any.whl (491 kB)
     |████████████████████████████████| 491 kB 1.4 MB/s
Collecting mock>=0.7.0
  Downloading https://www.piwheels.org/simple/mock/mock-4.0.1-py3-none-any.whl (28 kB)
ERROR: Package 'mock' requires a different Python: 3.5.3 not in '>=3.6'

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
ERROR: Package 'mock' requires a different Python: 3.5.3 not in '>=3.6'
```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
no log found

subham9777, 2020-02-17:
Exact same issue while installing the assistant using /home/pi/GassistPi/scripts/gassist-installer.sh it stucks at mock error as described above.

c-freitag, 2020-02-17:
Same here - manually installed python 3.7 to fulfill criteria and set as default, but no change. 
- What is the WHL file referring to?
- Is it still correct (as stated in the beginning of the instruction to **use "raspbian stretch"**?

adas4190, 2020-02-17:
yeah, I have tried that with no luck... I was able to install gassistapi like a year ago but no luck so far now

shivasiddharth, 2020-02-18:
Bump up the python version to 3.6 Follow the instructions here.
http://www.knight-of-pi.org/installing-python3-6-on-a-raspberry-pi/

divijbajaj5, 2020-02-18:
Still getting the same error after following above link

shivasiddharth, 2020-02-18:
@divijbajaj5 Please post the error. And mention what steps you followed and in what order. Bumping the python version will solve the issue. Open a terminal and type `python3 --version` to check the python version.

divijbajaj5, 2020-02-18:
Enter the modelid that was generated in the actions console: bivoiceai-mysmartspeaker-rvsr51

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease                  
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease               
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u5).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u2).
libssl-dev is already the newest version (1.1.0l-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1+deb9u2).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u2).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.8-0+deb9u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 249 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyaudio in /usr/local/lib/python3.6/site-packages (0.2.11)

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

===========Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Ok-Google hotword==========
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (20.0.2)
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (45.2.0)
Requirement already up-to-date: wheel in ./env/lib/python3.5/site-packages (0.34.2)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/plamere/spotipy.git (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 17))
  Cloning https://github.com/plamere/spotipy.git to /tmp/pip-req-build-gthmkcej
  Running command git clone -q https://github.com/plamere/spotipy.git /tmp/pip-req-build-gthmkcej
Collecting git+https://github.com/shivasiddharth/py-googletrans.git (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 23))
  Cloning https://github.com/shivasiddharth/py-googletrans.git to /tmp/pip-req-build-krvzgrdc
  Running command git clone -q https://github.com/shivasiddharth/py-googletrans.git /tmp/pip-req-build-krvzgrdc
[... 46 lines cut ...]
Collecting google-cloud-speech==0.36.0
  Using cached google_cloud_speech-0.36.0-py2.py3-none-any.whl (54 kB)
Collecting adafruit-io==2.1
  Using cached https://www.piwheels.org/simple/adafruit-io/adafruit_io-2.1-py3-none-any.whl (18 kB)
Collecting cryptography==2.4.2
  Using cached https://www.piwheels.org/simple/cryptography/cryptography-2.4.2-cp35-cp35m-linux_armv7l.whl (845 kB)
Collecting six>=1.10.0
  Using cached six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting python-dateutil
  Using cached python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
Collecting oauth2client>=1.1
  Downloading oauth2client-4.1.3-py2.py3-none-any.whl (98 kB)
     |████████████████████████████████| 98 kB 153 kB/s 
Collecting proboscis>=1.2.5.1
  Downloading https://www.piwheels.org/simple/proboscis/proboscis-1.2.6.0-py3-none-any.whl (27 kB)
Collecting future
  Using cached https://www.piwheels.org/simple/future/future-0.18.2-py3-none-any.whl (491 kB)
Collecting decorator>=3.3.1
  Using cached decorator-4.4.1-py2.py3-none-any.whl (9.2 kB)
Collecting mutagen>=1.34
  Using cached mutagen-1.44.0-py3-none-any.whl (212 kB)
Collecting mock>=0.7.0
  Using cached https://www.piwheels.org/simple/mock/mock-4.0.1-py3-none-any.whl (28 kB)
ERROR: Package 'mock' requires a different Python: 3.5.3 not in '>=3.6'
pi@raspberrypi:~ $

divijbajaj5, 2020-02-18:
python3 --version
Python 3.6.3
I guess we have to remove older one first ?

adas4190, 2020-02-18:
it worked for me, I did reboot after following procedure in the link provided. I have latest strech image fully updated and clean

adas4190, 2020-02-18:
I have succesfully installed it. Thank You!

shivasiddharth, 2020-02-18:
Will leave this issue open for others.

subham9777, 2020-02-18:
@adas4190 Can you please trace the exact steps you followed to solve the error after a clean build of the raspbian stretch?

divijbajaj5, 2020-02-18:
@shivasiddharth env lib shows 2 folders python 3.5 and python 3.6. So do I have to remove older version of python? 

@adas4190 I followed all the steps mentioned in the link. Please tell what exactly to be done?

adas4190, 2020-02-18:
I did nothing more than in the link. Don't worry about other python versions, once you add alias it will use 3.6 version. Just follow the procedure and try to install Gassistpi again and you should be ok

adas4190, 2020-02-18:
> @adas4190 Can you please trace the exact steps you followed to solve the error after a clean build of the raspbian stretch?

I did clean install, update system and follow procedure to update python (in the link provided ). 
Than follow procedure to install GassistPi

Voila 😊

divijbajaj5, 2020-02-18:
> Will leave this issue open for others.

Enter the modelid that was generated in the actions console: bivoiceai-mysmartspeaker-rvsr51

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease                  
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Fetched 15.0 kB in 9s (1,546 B/s)                               
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-10).
automake is already the newest version (1:1.15-6).
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.3).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u5).
libatlas-base-dev is already the newest version (3.10.3-1+rpi1).
libffi-dev is already the newest version (3.2.1-6).
libjack-jackd2-dev is already the newest version (1.9.10+20150825git1ed50c92~dfsg-5).
libmpg123-dev is already the newest version (1.23.8-1).
libpcre3 is already the newest version (2:8.39-3).
libpcre3-dev is already the newest version (2:8.39-3).
libsox-fmt-mp3 is already the newest version (14.4.1-5+deb9u2).
libssl-dev is already the newest version (1.1.0l-1~deb9u1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1+deb9u2).
mpg123 is already the newest version (1.23.8-1).
mplayer is already the newest version (2:1.3.0-6).
mpv is already the newest version (0.23.0-2+deb9u2).
nmap is already the newest version (7.40-1).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
python-setuptools is already the newest version (33.1.1-1).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-lxml is already the newest version (3.7.1-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-venv is already the newest version (3.5.3-1).
socat is already the newest version (1.7.3.1-2+deb9u1).
sox is already the newest version (14.4.1-5+deb9u2).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-5).
python-pip is already the newest version (9.0.1-2+rpt2).
python3-pip is already the newest version (9.0.1-2+rpt2).
vlc is already the newest version (3.0.8-0+deb9u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 249 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyaudio in /usr/local/lib/python3.6/site-packages (0.2.11)

===========================Checking OS Compatability=========================

===========You are running the installer on Stretch==========

===========Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword==========

===========GPIO pins can be used with the assistant==========

==========Changing particulars in service files==========

==========Changing particulars in service files for Ok-Google hotword==========
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.5/site-packages (20.0.2)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x75b16790>: Failed to establish a new connection: [Errno 101] Network is unreachable',)': /simple/setuptools/
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (45.2.0)
Requirement already up-to-date: wheel in ./env/lib/python3.5/site-packages (0.34.2)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/plamere/spotipy.git (from -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 17))
  Cloning https://github.com/plamere/spotipy.git to /tmp/pip-req-build-8fy75x2u
[... 46 lines cut ...]
  Using cached https://www.piwheels.org/simple/pycryptodomex/pycryptodomex-3.7.2-cp35-cp35m-linux_armv7l.whl (7.5 MB)
Collecting paho-mqtt==1.4.0
  Using cached https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.4.0-py3-none-any.whl (48 kB)
Collecting google-cloud-texttospeech==0.3.0
  Using cached google_cloud_texttospeech-0.3.0-py2.py3-none-any.whl (40 kB)
Collecting google-cloud-speech==0.36.0
  Using cached google_cloud_speech-0.36.0-py2.py3-none-any.whl (54 kB)
Collecting adafruit-io==2.1
  Using cached https://www.piwheels.org/simple/adafruit-io/adafruit_io-2.1-py3-none-any.whl (18 kB)
Collecting cryptography==2.4.2
  Using cached https://www.piwheels.org/simple/cryptography/cryptography-2.4.2-cp35-cp35m-linux_armv7l.whl (845 kB)
Collecting six>=1.10.0
  Using cached six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting python-dateutil
  Using cached python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
Collecting oauth2client>=1.1
  Using cached oauth2client-4.1.3-py2.py3-none-any.whl (98 kB)
Collecting proboscis>=1.2.5.1
  Using cached https://www.piwheels.org/simple/proboscis/proboscis-1.2.6.0-py3-none-any.whl (27 kB)
Collecting mock>=0.7.0
  Using cached https://www.piwheels.org/simple/mock/mock-4.0.1-py3-none-any.whl (28 kB)
ERROR: Package 'mock' requires a different Python: 3.5.3 not in '>=3.6'
pi@raspberrypi:~ $ python3 --version
Python 3.6.3
pi@raspberrypi:~ $

divijbajaj5, 2020-02-18:
> Enter the modelid that was generated in the actions console: bivoiceai-mysmartspeaker-rvsr51
>
> Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
> Get:2 [...]

@adas4190 @shivasiddharth Can't find any luck. It's still the same error despite doing everything.

shivasiddharth, 2020-02-18:
Hold on dont break your head. I am just in the processing of releasing the assistant for buster.

commit 26e3fbcf (2020-02-18): Bumped the project to work with Raspbian Buster

Fixes #987  Fixes #982  Fixes #963  Fixes #976
closed 2020-02-18 by shivasiddharth