shivasiddharth/GassistPi #988: Installation Error

Shidiqadm, 2020-02-25:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000108bcdb2
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[... 268 lines cut ...]

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-02-25:
Its a network issue. Retry after sometime.

Shidiqadm, 2020-02-25:
But I have tried for about 5 to 7 times for every time it says this also I tried changing the network.

subham9777, 2020-02-26:
@Shidiqadm I faced the same issue, if your network is unstable or the date of the raspberry pi is wrong or different this error shows up. Try for a stable network connection or change the date and time settings.

Shidiqadm, 2020-02-26:
ok I'll try that

Hoeby, 2020-02-27:
Also check your path for the client_secret file. Looks strange what you have written. When you put the file in the root of your user of will look like: /home/pi/client_secret.........json  

Change pi to your username.
This has nothing to with your error your have posted, but it will give an error when the install will run further

Shidiqadm, 2020-02-29:
now it gives me a error like this:
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    grpcio<2.0dev,>=1.8.2; extra == "grpc" from https://www.piwheels.org/simple/grpcio/grpcio-1.27.2-cp37-cp37m-linux_armv7l.whl#sha256=ae124c166243030bc2d67fa4f39c47790982efff9406d8ec2e43bfe4e614ab87 (from google-api-core[grpc]<2.0.0dev,>=1.6.0->google-cloud-texttospeech==0.3.0->-r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 25)):
        Expected sha256 ae124c166243030bc2d67fa4f39c47790982efff9406d8ec2e43bfe4e614ab87
             Got        46325bc94f6f388ca44288a7997c12f7f94928b130bcd58f749828b5c3665a14

closed 2020-04-07 by Shidiqadm