shivasiddharth/GassistPi #990: Malformed auth code error

Whoremoan, 2020-02-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial          : 000000000fbfde26
Model           : Raspberry Pi 2 Model B Rev 1.1

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"   
```
I had this running perfectly a few days ago, then after trying to get a LCD display to work everything got a bit messy, so I decided to wipe the Pi and start again. I did everything the same as I did at the weekend when I got it working, but now after entering the authorization code generated by google during the installation, I receive an error Malformed auth code error.
I have even tried deleting the google API stuff and starting again, but same result.
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes. Although I have found similar issues, they are quite old and don't seem to refer to anything that is valid anymore

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=1078799738607-rdgf9qlvce0etnrpjbnk9041rup3m7fl.apps.googleusercontent.com&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0%3Aoob&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fgcm&state=bKpCWDgNtZ4DHvZRmppqZbvpamvv2v&prompt=consent&access_type=offline
Enter the authorization code:  4/wwH0nNHPbsLTQLomg3DNHHqoH_8sStr8kF74OFHYfzEkSG69YSn7pZI                                                                                         Traceback (most recent call last):                                                                                                                                                 File "/home/pi/env/bin/google-oauthlib-tool", line 8, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/tool/__main__.py", line 106, in main
    creds = flow.run_console()
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/flow.py", line 404, in run_console
    self.fetch_token(code=code)
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/flow.py", line 281, in fetch_token
    self.client_config['token_uri'], **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/requests_oauthlib/oauth2_session.py", line 360, in fetch_token
    self._client.parse_request_body_response(r.text, scope=self.scope)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/clients/base.py", line 421, in parse_request_body_response
    self.token = parse_token_response(body, scope=scope)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 431, in parse_token_response
    validate_token_parameters(params)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 438, in validate_token_parameters
    raise_from_error(params.get('error'), params)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/errors.py", line 405, in raise_from_error
    raise cls(**kwargs)
oauthlib.oauth2.rfc6749.errors.InvalidGrantError: (invalid_grant) Malformed auth code.

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

Whoremoan, 2020-02-27:
OK, So again, I deleted my google project and remade it. Now when I enter the authorization code I get
Enter the authorization code: 4/xAH6ACqUi3tnXgiTOrgPlowJuHIrjxKiYtMJAn3jp8PSfoK6                                                                                                 xy8Ss24                                                                                                                                                                          Traceback (most recent call last):                                                                                                                                                 File "/home/pi/env/bin/google-oauthlib-tool", line 8, in <module>                                                                                                                  sys.exit(main())                                                                                                                                                               File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 722, in __                                                                                                 call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 697, in ma                                                                                                 in
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 895, in in                                                                                                 voke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 535, in in                                                                                                 voke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/tool/__mai                                                                                                 n__.py", line 106, in main
    creds = flow.run_console()
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/flow.py",                                                                                                  line 404, in run_console
    self.fetch_token(code=code)
  File "/home/pi/env/lib/python3.7/site-packages/google_auth_oauthlib/flow.py",                                                                                                  line 281, in fetch_token
    self.client_config['token_uri'], **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/requests_oauthlib/oauth2_sessio                                                                                                 n.py", line 360, in fetch_token
    self._client.parse_request_body_response(r.text, scope=self.scope)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/clients                                                                                                 /base.py", line 421, in parse_request_body_response
    self.token = parse_token_response(body, scope=scope)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/paramet                                                                                                 ers.py", line 431, in parse_token_response
    validate_token_parameters(params)
  File "/home/pi/env/lib/python3.7/site-packages/oauthlib/oauth2/rfc6749/paramet                                                                                                 ers.py", line 461, in validate_token_parameters
    raise w
Warning: Scope has changed from "https://www.googleapis.com/auth/assistant-sdk-p                                                                                                 rototype https://www.googleapis.com/auth/gcm" to "https://www.googleapis.com/aut                                                                                                 h/assistant-sdk-prototype".

shivasiddharth, 2020-02-28:
The forward slash "/" that you have in the auth code is causing the problem. You need not re install everything again. move in to the environment using:
`source env/bin/activate `
and run the authorization again

```
google-oauthlib-tool --scope https://www.googleapis.com/auth/assistant-sdk-prototype \
          --scope https://www.googleapis.com/auth/gcm \
          --save --headless --client-secrets "REPLACE THIS WITH PATH TO CREDENTIALS FILE"
```

Whoremoan, 2020-02-28:
Hi
Thank you for your quick response.
I tried to reauthorize as per your instructions but I got exactly the same results as my 2nd post above.
The auth. code that was generated by google still had a / in it.

Incidentally, if I try to manually start the GassistPi service, I get this error at the end.

usage: main.py [-h] [--device-model-id DEVICE_MODEL_ID]
               [--project-id PROJECT_ID] [--nickname NICKNAME]
               [--device-config DEVICE_CONFIG_FILE]
               [--credentials OAUTH2_CREDENTIALS_FILE] [--query QUERY] [-v]
main.py: error: argument --credentials: No such file or directory: "/home/pi/.config/google-oauthlib-tool/credentials.json"

But I'm guessing that is a result of my issue above?

Simon

Whoremoan, 2020-02-29:
Hi not sure if you have updated anything recently, but I’ve just wiped my SD card and started again from scratch. 
This time the installation went through perfect without issues and I’m back up and running. 
Great project. Keep up the good work :)

closed 2020-02-29 by Whoremoan