shivasiddharth/GassistPi #991: vlcpulse audio output error: PulseAudio server connection failure: Connection refused

rickyklu, 2020-03-01:
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 00000000569d590c
```
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:
```
I went through the initial set up and after running `sudo  ./GassistPi/scripts/gassist-installer.sh` and going through Steps 3-7 and verified it could make sound, hear my voice commands, and look up the weather.
[... 89 lines cut ...]
2020-02-29 19:56:32,040 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2020-02-29 19:56:32,055 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2020-02-29 19:56:32,331 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2020-02-29 19:56:32,404 : DEBUG : Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2020-02-29 19:56:32,424 : DEBUG : Starting new HTTPS connection (1): embeddedassistant.googleapis.com:443
2020-02-29 19:56:32,668 : DEBUG : https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/aiy-kit-2020-assist-pi-oh6hdc/devices/4906A5C9F0BDCE5D9CF71EDEA74696A5 HTTP/1.1" 400 None
2020-02-29 19:56:44,605 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2020-02-29 19:56:44,618 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2020-02-29 19:56:44,839 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2020-02-29 19:56:44,872 : DEBUG : Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2020-02-29 19:56:44,886 : DEBUG : Starting new HTTPS connection (1): embeddedassistant.googleapis.com:443
2020-02-29 19:56:45,095 : DEBUG : https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/aiy-kit-2020-assist-pi-oh6hdc/devices/4906A5C9F0BDCE5D9CF71EDEA74696A5 HTTP/1.1" 400 None
2020-02-29 19:56:56,487 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2020-02-29 19:56:56,500 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2020-02-29 19:56:56,730 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2020-02-29 19:56:56,765 : DEBUG : Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2020-02-29 19:56:56,777 : DEBUG : Starting new HTTPS connection (1): embeddedassistant.googleapis.com:443
2020-02-29 19:56:56,998 : DEBUG : https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/aiy-kit-2020-assist-pi-oh6hdc/devices/4906A5C9F0BDCE5D9CF71EDEA74696A5 HTTP/1.1" 400 None
2020-02-29 19:57:11,147 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2020-02-29 19:57:11,161 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2020-02-29 19:57:11,381 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2020-02-29 19:57:11,416 : DEBUG : Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2020-02-29 19:57:11,429 : DEBUG : Starting new HTTPS connection (1): embeddedassistant.googleapis.com:443
2020-02-29 19:57:11,635 : DEBUG : https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/aiy-kit-2020-assist-pi-oh6hdc/devices/4906A5C9F0BDCE5D9CF71EDEA74696A5 HTTP/1.1" 400 None
```

shivasiddharth, 2020-03-02:
Your manual start command is wrong. Please check the readme again. Did you by any chance the the service files as well ?

rickyklu, 2020-03-03:
Silly me for not reading clearly, I updated the command to manually start and it works fine.

As for the autostart, I checked the service file and some reason the project and model IDs didn't match so I manually changed them and autostart seems to be working now.

closed 2020-03-03 by rickyklu