shivasiddharth/GassistPi #993: Raspberry Pi 3 + Buster. Installation OK but GassistPi doesn't start after reboot

mc9625, 2020-03-06:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000007f7b9558
Model		: Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
I've installed GassistPi on a PI starting from a fresh Buster image. The installation went fine but after the reboot the service start but the device is not responding to hotword or button. Even stopping the service and try the manual start command doesn't work, it will gave me a bunch of errors.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id [...]

```

### Paste the contents of the terminal below
```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-06 19:35:35,405 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0125a8e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0126e368] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x76f34ad0 (most recent call first):
  File "/usr/lib/python3.7/subprocess.py", line 775 in __init__
  File "/usr/lib/python3.7/ctypes/util.py", line 291 in _findSoname_ldconfig
  File "/usr/lib/python3.7/ctypes/util.py", line 322 in find_library
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 317 in _load_lib
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 90 in __init__
  File "/home/pi/GassistPi/src/main.py", line 987 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-03-07:
Did the assistant start normally immediately after installation ? What is your audio setup ?

mc9625, 2020-03-07:
Yes the assistant started normally. I have the Google AYI Kit. Everything works fine if I do a test.
This is the arecord -l output:
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

and the aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

This is the /home/pi/.asoundrc

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2020-03-07:
Can you post output of asound.conf ?

mc9625, 2020-03-07:
#
#  ALSA library configuration file
#

# pre-load the configuration files

@hooks [
	{
		func load
		files [
			"/etc/alsa/conf.d"
			"/etc/asound.conf"
			"~/.asoundrc"
		]
		errors false
	}
]

# load card-specific configuration files (on request)

cards.@hooks [
	{
		func load
		files [
			{
				@func concat
				strings [
					{ @func datadir }
					"/cards/aliases.conf"
				]
			}
		]
	}
	{
		func load_for_all_cards
		files [
			{
				@func concat
				strings [
					{ @func datadir }
					"/cards/"
					{ @func private_string }
					".conf"
				]
			}
		]
		errors false
	}
]

#
# defaults
#

# show all name hints also for definitions without hint {} section
defaults.namehint.showall on
# show just basic name hints
defaults.namehint.basic on
# show extended name hints
defaults.namehint.extended on
#
defaults.ctl.card 0
defaults.pcm.card 0
defaults.pcm.device 0
defaults.pcm.subdevice -1
defaults.pcm.nonblock 1
defaults.pcm.compat 0
defaults.pcm.minperiodtime 5000		# in us
defaults.pcm.ipc_key 5678293
defaults.pcm.ipc_gid audio
[... 533 lines cut ...]
	@args.DEV {
		type integer
		default {
			@func refer
			name defaults.timer.device
		}
	}
	@args.SUBDEV {
		type integer
		default {
			@func refer
			name defaults.timer.subdevice
		}
	}
	type hw
	class $CLASS
	sclass $SCLASS
	card $CARD
	device $DEV
	subdevice $SUBDEV
	hint {
		description "Direct timer device"
		device $DEV
	}
}

shivasiddharth, 2020-03-07:
Can you just check the /etc/asound.conf file ?

mc9625, 2020-03-07:
Yes, sorry wrong file! :)

Here is the content of the etc/asound.conf

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2020-03-07:
Please do not hide parts of manual start command, its absolutely safe to share here. Also I need to know if you made any change to the service file ? Can you share the service file here ?

mc9625, 2020-03-07:
I haven't changed the service file. 
The manual command is 

/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id gassistpi2020-gassistpi

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'gassistpi2020' --device_model_id 'gassistpi2020-gassistpi'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

BTW I've noticed that occasionally the raspberry plays the audio effect that GassistPi plays when it starts. But only the chimes, not the phrase. This even without touching anything.

shivasiddharth, 2020-03-08:
open a terminal and run:

```
source env/bin/activate
pip install google-assistant-grpc --upgrade
pip install google-assistant-sdk --upgrade
pip install google-assistant-sdk[samples] --upgrade
```

See if it works

shivasiddharth, 2020-03-08:
Can you also give me the name of the Buster image that you are using (release date).

mc9625, 2020-03-08:
Unfortunately this solution didn't work. Same errors.

Actually I can't tell you the name of the Buster image. I've used the Raspberry Pi Imager, it took care of download the image and burn it. I've done it just a couple of days ago so I guess it should be the very last version.

shivasiddharth, 2020-03-08:
Is this error occurring occasionally or  repeatedly ? Its an issue with the AIY-HAT audio causing the errors. Is you AIY hat version 1 or the bonnet ?

mc9625, 2020-03-08:
This error occurs everytime. I mean, GassistPi doesn't start at boot (even if it keeps playing the chimes tone, not the welcome message, every once in a while) and stopping the service and try to start it manually results in the error I've posted few message ago.
I have a AIY-HAT version 1.

shivasiddharth, 2020-03-08:
Try removing the 
```
ttable {
0.0 30.0
1.1 30.0
}
```
in both .asoundrc and asound.conf

shivasiddharth, 2020-03-08:
Give it a restart after removing the table.

mc9625, 2020-03-08:
something has changed! Now I've got this error:
    logger.exception(error)
NameError: name 'logger' is not defined

I've searched other issues with the same errors. Someone claimed it has fixed the issue by reinstalling pulse audio, but I read that you said it was not needed.

shivasiddharth, 2020-03-09:
in the last line of main.py, change `logger` to `logging` as shown here

https://github.com/shivasiddharth/GassistPi/blob/b999c35e75da5cac055d822c5f3029ab5b45fff5/src/main.py#L1026

shivasiddharth, 2020-03-09:
to fix the rate issue, after this line https://github.com/shivasiddharth/GassistPi/blob/b999c35e75da5cac055d822c5f3029ab5b45fff5/audio-drivers/AIY-HAT/scripts/asound.conf#L11 
in both asound.aonf and .asoundrc, add 

`rate 16000`

reboot and give it a shot

mc9625, 2020-03-09:
Sorry the last one is not clear: do I need to modify the GassistPi/audio-drivers/AIY-HAT/scripts/asound.conf or the /etc/asound.conf and .asoundrc file?
Anyway I've done both and modified the main.py. Again nothing happened after the reboot. I've stopped the service and tried to start it manually but it doesn't start. This is the error:

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-09 09:32:15,178 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[00d71590] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[00d8d3e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_route.c:1314:(_snd_pcm_route_open) Unknown field rate
ALSA lib pcm_route.c:1314:(_snd_pcm_route_open) Unknown field rate
ALSA lib pcm_route.c:1314:(_snd_pcm_route_open) Unknown field rate
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
2020-03-09 09:32:17,024 : ERROR : [Errno -9997] Invalid sample rate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate

I've added the line rate 16000 in both /etc/asound.conf and .asoundrc but apparently without any luck.

shivasiddharth, 2020-03-09:
under pcm.micboost, can you change the type from route to plug in the files and remove the rate parameter that you added. ?

The new settings should look like:

```
pcm.softvol {
type softvol
slave.pcm dmix
control {
name Master
card 0
}
}

pcm.micboost {
type plug
slave.pcm dsnoop
}

pcm.!default {
type asym
playback.pcm "plug:softvol"
capture.pcm "plug:micboost"
}

ctl.!default {
type hw
card 0
}
```

shivasiddharth, 2020-03-09:
Change both asound.conf and .asoundrc

mc9625, 2020-03-09:
Still errors:

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-09 11:00:50,302 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[00df18a0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[00e055f8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x76fa6ad0 (most recent call first):
  File "/usr/lib/python3.7/subprocess.py", line 775 in __init__
  File "/usr/lib/python3.7/ctypes/util.py", line 291 in _findSoname_ldconfig
  File "/usr/lib/python3.7/ctypes/util.py", line 322 in find_library
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 317 in _load_lib
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 90 in __init__
  File "/home/pi/GassistPi/src/main.py", line 987 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

shivasiddharth, 2020-03-09:
Can you share your asound.conf and .asoundrc. Just want to make sure that changes were made properly.

mc9625, 2020-03-09:
pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type plug
    slave.pcm dsnoop
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}
ctl.!default {
    type hw
    card 0
}

shivasiddharth, 2020-03-09:
Can you try adding the rate after `type plug` add `rate 44100` This is an isolated incident with AIY HAT need to figure out as to what is going on.

mc9625, 2020-03-09:
This is the error now:

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-09 17:13:28,407 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01f38778] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01f53cf8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_plug.c:1286:(_snd_pcm_plug_open) Unknown field rate
ALSA lib pcm_plug.c:1286:(_snd_pcm_plug_open) Unknown field rate
ALSA lib pcm_plug.c:1286:(_snd_pcm_plug_open) Unknown field rate
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
2020-03-09 17:13:30,616 : ERROR : [Errno -9997] Invalid sample rate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate

shivasiddharth, 2020-03-10:
Try using the Custom-voice-hat drivers that i have given. That will work with AIY-HATS

mc9625, 2020-03-10:
Nope.

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-10 08:21:05,952 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[029d0018] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[029eada0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
CUSTOM-VOICE-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x76fbcad0 (most recent call first):
  File "/usr/lib/python3.7/subprocess.py", line 1453 in _execute_child
  File "/usr/lib/python3.7/subprocess.py", line 775 in __init__
  File "/usr/lib/python3.7/ctypes/util.py", line 291 in _findSoname_ldconfig
  File "/usr/lib/python3.7/ctypes/util.py", line 322 in find_library
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 317 in _load_lib
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 90 in __init__
  File "/home/pi/GassistPi/src/main.py", line 987 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

shivasiddharth, 2020-03-10:
can you give me the output of  `aplay --version`

mc9625, 2020-03-10:
aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

shivasiddharth, 2020-03-10:
Have you installed pulseaudio by any chance ?

mc9625, 2020-03-10:
not yet. I started from a clean buster lite image, directly downloaded by the PI imager and started your tutorial. Actually I was considering installing it. What do you suggest?

shivasiddharth, 2020-03-10:
Try this. Hopefully this should fix the error. Open a terminal and run one command after another.

```
wget ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.3.tar.bz2
tar xvjf alsa-lib-1.1.3.tar.bz2
cd alsa-lib-1.1.3
./configure
make && sudo make install
cd ..
sudo apt-get install libncursesw5-dev -y
wget ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.3.tar.bz2
tar xvjf alsa-utils-1.1.3.tar.bz2
cd alsa-utils-1.1.3
./configure --disable-alsaconf --disable-bat --disable-xmlto --with-curses=ncursesw
make && sudo make install

```
Do a restart after this. And then give it a shot.

mc9625, 2020-03-10:
Unfortunately it didn't work.

/home/pi/env/lib/python3.7/site-packages/past/builtins/misc.py:45: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  from imp import reload
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-10 17:07:25,619 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01f411e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01f56ff0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
CUSTOM-VOICE-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x76f1ead0 (most recent call first):
  File "/usr/lib/python3.7/subprocess.py", line 1453 in _execute_child
  File "/usr/lib/python3.7/subprocess.py", line 775 in __init__
  File "/usr/lib/python3.7/ctypes/util.py", line 291 in _findSoname_ldconfig
  File "/usr/lib/python3.7/ctypes/util.py", line 322 in find_library
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 322 in _load_lib
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 95 in __init__
  File "/home/pi/GassistPi/src/main.py", line 987 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

I wonder if I should try to start from scratch again, maybe from stretch.

shivasiddharth, 2020-03-10:
Stretch will work for sure. You can try using stretch.

mc9625, 2020-03-10:
I see. Thank you for your support! I feel I've already waste your time too much with this issue! Let's start from scratch... from stretch and see what happens!
Thanks
Massimo

shivasiddharth, 2020-03-11:
I will try and find a fix for this issue.

shivasiddharth, 2020-03-11:
Install pulseaudio. Seems like the only fix.

closed 2020-03-11 by shivasiddharth

normanord, 2020-03-11:
Had quite similar problem with ALSA on RPI4 with buster (mine assistant was starting, and recognising speach, but was restarting in random moments). Can confirm that installing pulsaudio solve the problem. Both with manual start and autostart.

mc9625, 2020-03-11:
It really seems that GassistPi doesn't work with my setup, at least starting from scratch. I even started from a blank stretch image but at the end I still get the same exact error. I still have an SD with GassistPi working. That SD was originally an AIY VOICE KIT image. I installed GassistPI on top of the AIY installation and it worked (at least running it manually, it still doesn't start automatically). I even updated the SO to buster without further issue. It's the only working solution for me.

shivasiddharth, 2020-03-12:
@mc9625 , AIY image has pulseaudio installed by default, thats why its working without any issues. Certain stretch images had Alsa version 1.1.3 which has no problem. From Alsa 1.1.4, the conflict with portaudio started if we used dsnoop plugin. Buster comes with Alsa 1.1.7 or 1.1.8 which would still have the same issues as 1.1.4. I tried to rollback Alsa on buster to 1.1.3 to refrain from installing pulseaudio, but it did not work. Only after multiple attempts that i have suggested to install pulseaudio.

mc9625, 2020-03-12:
Actually on the stretch image Pulseaudio is installed. 

pulseaudio is already the newest version (12.2-4+deb10u1)

but Alsa is actually the 1.1.8.

mc9625, 2020-03-12:
WOW NOW IT WORKS...

This is what I've done:

I've tried to install Alsa 1.1.3 by downloading it and then installing it 

This is the exact sequence of commands I've used:

wget ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.3.tar.bz2
tar xvjf alsa-lib-1.1.3.tar.bz2 
cd alsa-lib-1.1.3/
./configure 
make && sudo make install
cd
wget ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.3.tar.bz2
tar xvjf alsa-utils-1.1.3.tar.bz2 
cd alsa-utils-1.1.3/
sudo apt-get install libncursesw5-dev -y
./configure --disable-alsaconf --disable-bat --disable-xmlto --with-curses=ncursesw
make && sudo make install

The last command actually gave me an error:

make[1]: *** No rule to make target 'alsaucm.1', needed by 'all-am'.  Stop.
make[1]: Leaving directory '/home/pi/alsa-utils-1.1.3/alsaucm'
make: *** [Makefile:353: all-recursive] Error 1

I've still have Alsa 1.1.8 as default. In fact doing a aplay --version returns: 
aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

but I've tried to start GassistPi manually and... it works!

Now I will try to make it start automatically... we'll see! 

Hope this will help you trying to figure out what's the problema and find a solution!

Thanks

mc9625, 2020-03-12:
Just FYI: the auto start up still doesn't work. After the reboot it tooks a lot of minutes before playing the chimes but it still stops soon after, without even play the "hey" voice.
Anyway the manual startup works flawlessly.

tonijota, 2020-07-24:
> Just FYI: the auto start up still doesn't work. After the reboot it tooks a lot of minutes before playing the chimes but it still stops soon after, without even play the "hey" voice.
> Anyway the manual startup works flawlessly.
> BTW now I am on stretch.

did you get it to work?

i'm facing the same problem with no success :(

commit 86f1c6c2 (2020-03-11): Update GassistPi-system-requirements.txt

Fixes #993

Solves portaudio-alsa errors