shivasiddharth/GassistPi #995: Yoututbe API doesn't work

Ijdtm7, 2020-03-14:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000009a819b16

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Whenever i request for a youtube video to be played it says "ERROR: HTTP is not supported" 
This is new and it always used to work. It is still broken after i reinstalled GassistPi
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
Yes

### 2. Is the issue related to assistant installation ?
Yes/No
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
No

### 4. Is the assistant service restarting automatically?
Yes/No
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
It didnt crash, just it said "error please check the logs" after i said "play daft punk around the world from youtube"
```

### Paste the contents of the terminal below
```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-14 11:54:29,233 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
AIY-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: raspberry-pi-3-ijd
device_id: 3482591B99A23733AA3DB780D3F06E61

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Daft Punk Around The World from YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
daft punk around the world
ERROR: HTTP is not supported.
ERROR: HTTP is not supported.
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/4333054/GassistPi.log)

AGID-WL, 2020-03-14:
I have the same problem and was trying to fix it for 2 days. I even bought DNS and used letsencrypt for ssl and still having the same error

shivasiddharth, 2020-03-15:
Can you run the following and share the output?

```
source env/bin/activate
pip freeze
```

shivasiddharth, 2020-03-15:
Also, manually start the assistant, give the youtube command and share the terminal contents.

Ijdtm7, 2020-03-15:
> Also, manually start the assistant, give the youtube command and share the terminal contents.

Well I already did, that’s the terminal output I posted above in my issue

> Can you run the following and share the output?
> 
> ```
> source env/bin/activate
> pip freeze
> ```

Ok I will when possible

shivasiddharth, 2020-03-15:
https://github.com/shivasiddharth/GassistPi/blob/master/src/youtube_search_engine.py in this, uncomment lines 36, 65, 66, 67, 68.

After this line https://github.com/shivasiddharth/GassistPi/blob/86f1c6c2d705608d7bbed12c8a075b63780a23d0/src/actions.py#L1176 add 
`print(str(streamurl))`

Now manually start the assistant, give the command and share the terminal contents.

AGID-WL, 2020-03-15:
Here is the log from Youtube command:

`stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.

{'regionCode': 'EG', 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/gh-YXBIV2XeNpaIcIyu5vl6wDc0"', 'pageInfo': {'resultsPerPage': 5, 'totalResults': 1000000}, 'kind': 'youtube#searchListResponse', 'nextPageToken': 'CAUQAA', 'items': [{'id': {'videoId': 'eyzxu26-Wqk', 'kind': 'youtube#video'}, 'snippet': {'liveBroadcastContent': 'none', 'channelTitle': 'Sony Pictures Entertainment', 'channelId': 'UCz97F7dMxBNOfGYu3rx8aCw', 'description': 'The only thing more dangerous than the front lines was the fight to get there. Screenplay by Tom Hanks inspired by actual events, Greyhound comes to theaters ...', 'publishedAt': '2020-03-05T17:00:07.000Z', 'title': 'GREYHOUND - Official Trailer (HD)', 'thumbnails': {'default': {'width': 120, 'url': 'https://i.ytimg.com/vi/eyzxu26-Wqk/default.jpg', 'height': 90}, 'high': {'width': 480, 'url': 'https://i.ytimg.com/vi/eyzxu26-Wqk/hqdefault.jpg', 'height': 360}, 'medium': {'width': 320, 'url': 'https://i.ytimg.com/vi/eyzxu26-Wqk/mqdefault.jpg', 'height': 180}}}, 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/nTXDCUa-0YmFRwl6ButpYswl_bE"', 'kind': 'youtube#searchResult'}, {'id': {'videoId': 'LF--yzj4W1s', 'kind': 'youtube#video'}, 'snippet': {'liveBroadcastContent': 'none', 'channelTitle': 'The Late Show with Stephen Colbert', 'channelId': 'UCMtFAi84ehTSYSE9XoHefig', 'description': 'America\'s favorite astrophysicist, host of "Cosmos: Possible Worlds" on NatGeo, stirred up all kinds of trouble when he pointed out the Elsa from "Frozen" isn\'t ...', 'publishedAt': '2020-03-07T08:35:00.000Z', 'title': 'Neil deGrasse Tyson Stands By His Tweet About &quot;Frozen&quot;', 'thumbnails': {'default': {'width': 120, 'url': 'https://i.ytimg.com/vi/LF--yzj4W1s/default.jpg', 'height': 90}, 'high': {'width': 480, 'url': 'https://i.ytimg.com/vi/LF--yzj4W1s/hqdefault.jpg', 'height': 360}, 'medium': {'width': 320, 'url': 'https://i.ytimg.com/vi/LF--yzj4W1s/mqdefault.jpg', 'height': 180}}}, 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/tCToqFnaeacEOb9apZg0J13E_nk"', 'kind': 'youtube#searchResult'}, {'id': {'videoId': 'N3idyCo4hSw', 'kind': 'youtube#video'}, 'snippet': {'liveBroadcastContent': 'none', 'channelTitle': 'FBE', 'channelId': 'UC0v-tlzsn0QZwJnkiaUSJVQ', 'description': 'Have you subscribed to our new Try Not To Channel yet? Click the link to subscribe now and HIT the bell! https://fbereact.com/SubscribeTryNotTo Generations ...', 'publishedAt': '2020-03-06T22:00:02.000Z', 'title': '10 Tricks Advertisers Use To Make Food Look Delicious | Generations React', 'thumbnails': {'default': {'width': 120, 'url': 'https://i.ytimg.com/vi/N3idyCo4hSw/default.jpg', 'height': 90}, 'high': {'width': 480, 'url': 'https://i.ytimg.com/vi/N3idyCo4hSw/hqdefault.jpg', 'height': 360}, 'medium': {'width': 320, 'url': 'https://i.ytimg.com/vi/N3idyCo4hSw/mqdefault.jpg', 'height': 180}}}, 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/fJ7Lr-8XKio0iAhoOxquyraIR-I"', 'kind': 'youtube#searchResult'}, {'id': {'videoId': 'nc6dz18GPU0', 'kind': 'youtube#video'}, 'snippet': {'liveBroadcastContent': 'none', 'channelTitle': 'Fox News', 'channelId': 'UCXIJgqnII2ZOINSWNOGFThA', 'description': "Sen. Kelly Loeffler, one of the 14 senators co-sponsoring the resolution, calls for Sen. Schumer's apology on 'Fox & Friends.' #FoxNews FOX News operates the ...", 'publishedAt': '2020-03-06T14:25:00.000Z', 'title': 'Senate Republicans draft resolution to censure Schumer&#39;s SCOTUS &#39;threat&#39;', 'thumbnails': {'default': {'width': 120, 'url': 'https://i.ytimg.com/vi/nc6dz18GPU0/default.jpg', 'height': 90}, 'high': {'width': 480, 'url': 'https://i.ytimg.com/vi/nc6dz18GPU0/hqdefault.jpg', 'height': 360}, 'medium': {'width': 320, 'url': 'https://i.ytimg.com/vi/nc6dz18GPU0/mqdefault.jpg', 'height': 180}}}, 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/yKVswCg0vzJnYTpHZEg7OM8bntQ"', 'kind': 'youtube#searchResult'}, {'id': {'videoId': '5VsHnpU-Xjc', 'kind': 'youtube#video'}, 'snippet': {'liveBroadcastContent': 'none', 'channelTitle': 'MSNBC', 'channelId': 'UCaXkIU1QidjPwiAYu6GcHjg', 'description': 'A federal judge appointed by George W. Bush issues a scathing critique of Attorney General William Barr over his distortions of the findings of special counsel ...', 'publishedAt': '2020-03-06T23:36:08.000Z', 'title': 'Federal Judge Blasts Barr Over Mueller Report | Deadline | MSNBC', 'thumbnails': {'default': {'width': 120, 'url': 'https://i.ytimg.com/vi/5VsHnpU-Xjc/default.jpg', 'height': 90}, 'high': {'width': 480, 'url': 'https://i.ytimg.com/vi/5VsHnpU-Xjc/hqdefault.jpg', 'height': 360}, 'medium': {'width': 320, 'url': 'https://i.ytimg.com/vi/5VsHnpU-Xjc/mqdefault.jpg', 'height': 180}}}, 'etag': '"SJZWTG6xR0eGuCOh2bX6w3s4F94/yJGlfkqIwuu2bwUDkOaH9PbDDsE"', 'kind': 'youtube#searchResult'}]}
Videos
 GREYHOUND - Official Trailer (HD) (eyzxu26-Wqk)
Neil deGrasse Tyson Stands By His Tweet About &quot;Frozen&quot; (LF--yzj4W1s)
10 Tricks Advertisers Use To Make Food Look Delicious | Generations React (N3idyCo4hSw)
Senate Republicans draft resolution to censure Schumer&#39;s SCOTUS &#39;threat&#39; (nc6dz18GPU0)
[... cut ...]
ERROR: eyzxu26-Wqk: Youtube says: HTTP is not supported.
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
`
Hope it helps

shivasiddharth, 2020-03-16:
Can you add print(urllid) after this line https://github.com/shivasiddharth/GassistPi/blob/86f1c6c2d705608d7bbed12c8a075b63780a23d0/src/actions.py#L1172

Then again give the command and post the terminal contents. I also need to see the pip freeze command that i mentioned above https://github.com/shivasiddharth/GassistPi/issues/995#issuecomment-599158858

AGID-WL, 2020-03-17:
ok, I found that the problem is from pafy and/or youtubedl but after upgrading them both. it doesn't say "ERROR: HTTP is not supported." but some videos/urls coming from pafy works and others don't

Ijdtm7, 2020-03-17:
Here are both of the outputs
pip freeze output:
```
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip freeze
adafruit-io==2.1
aftership==0.2
appdirs==1.4.3
asn1crypto==0.24.0
beautifulsoup4==4.7.1
bs4==0.0.1
cachetools==3.1.1
casttube==0.2.0
certifi==2019.6.16
cffi==1.12.3
chardet==3.0.4
click==6.7
colorzero==1.1
cryptography==2.4.2
decorator==4.4.0
enum34==1.1.6
feedparser==5.2.1
future==0.17.1
futures==3.1.1
gmusicapi==12.1.0
google-api-core==1.13.0
google-api-python-client==1.7.9
google-assistant-grpc==0.2.1
google-assistant-library==1.1.0
google-assistant-sdk==0.6.0
google-auth==1.6.3
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.4.0
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.6.0
googletrans==2.3.0
gpiozero==1.5.1
gpsoauth==0.4.1
grpcio==1.22.0
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.13.0
idna==2.8
ifaddr==0.1.6
kodi-json==1.0.0
lxml==4.3.4
MechanicalSoup==0.11.0
mock==3.0.5
mutagen==1.42.0
numpy==1.16.4
oauth2client==4.1.3
oauthlib==3.0.2
pafy==0.5.4
paho-mqtt==1.4.0
pathlib2==2.3.4
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.9.0
psutil==5.6.3
pushbullet.py==0.11.0
pyasn1==0.4.5
pyasn1-modules==0.2.5
PyAudio==0.2.11
PyChromecast==3.2.2
pycparser==2.19
pycryptodomex==3.7.2
pyOpenSSL==19.0.0
python-dateutil==2.8.0
python-magic==0.4.15
[... 120 lines cut ...]
  

FGBhQbmPwH8
ERROR: HTTP is not supported.
ERROR: HTTP is not supported.
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
Kodi TV box not online
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Kodi TV box not online

```

shivasiddharth, 2020-03-17:
Your youtube-dl  and pafy api is too old try updating it.
```
source env/bin/activate
pip3 install youtube-dl
pip3 install pafy
```

Ijdtm7, 2020-03-17:
> ok, I found that the problem is from pafy and/or youtubedl but after upgrading them both. it doesn't say "ERROR: HTTP is not supported." but some videos/urls coming from pafy works and others don't which begins with (https://r1)

what version did you upgrade too? mine says it's at latest

Ijdtm7, 2020-03-17:
oh i need to do pip3, sorry didnt see your comment

Ijdtm7, 2020-03-17:
> Your youtube-dl and pafy api is too old try updating it.
> 
> ```
> source env/bin/activate
> pip3 install youtube-dl
> pip3 install pafy
> ```

uhh, it didnt work:
```(env) pi@raspberrypi:~ $ pip install pafy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pafy in ./env/lib/python3.5/site-packages (0.5.4)
(env) pi@raspberrypi:~ $ pip3 install youtube-dl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: youtube-dl in ./env/lib/python3.5/site-packages (2019.7.12)
(env) pi@raspberrypi:~ $ pip3 install pafy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pafy in ./env/lib/python3.5/site-packages (0.5.4)
```

Ijdtm7, 2020-03-17:
should i reinstall it?

shivasiddharth, 2020-03-17:
I missed `-U`  try 
```
source env/bin/activate
pip3 install -U youtube-dl
pip3 install -U pafy
```
or 
```
source env/bin/activate
pip3 uninstall youtube-dl
pip3 uninstall pafy

pip3 install youtube-dl
pip3 install pafy
```

Ijdtm7, 2020-03-17:
Thanks! it works. But one last thing, what does this mean at the startup of the service:
```ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
```

shivasiddharth, 2020-03-17:
Nothing. Just ALSA warnings. No cause for concern.

Ijdtm7, 2020-03-17:
ok, thanks!

closed 2020-03-17 by Ijdtm7