shivasiddharth/GassistPi #998: Unable to stream radio when URL is m3u format

acowx, 2020-03-22:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```

```
Paste your Board and OS details here.   
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000005beb5e73

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
[... 636 lines cut ...]
Streaming radio x in vlc
[01bdbba0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

NO AUDIO PRODUCED

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
Streaming radio x in vlc
[01bdbba0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/4365674/GassistPi.log)

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-03-23:
```
Code is :
xurl='http://media-ice.musicradio.com/RadioXUK.m3u'
vlcplayer.media_manager(xurl,'Radio')
vlcplayer.media_player(xurl)
```
1. What is this code that you have mentioned here ?
2. Is only .m3u formats not playing or none of the formats are working ?
3. Are the default radio links provided working ?
4. Did you check the media player volume ? (Set media player volume to 70, command syntax is given in README to change the media player volume).

acowx, 2020-03-23:
That code is a snippit from my custom action in actions.py

I am using a custom keyword, not the standard "Radio" wakeword, my "Radio" section is disabled

Other formats (mp3) do work by just changing xurl to a non-m3u file reference

Yes, media volume is good, when I did enable the standard "Radio" keyword I do get audio

shivasiddharth, 2020-03-23:
My radio implementation will work out of the box. Any reason for not using that ? 
If cvlc works for you then open the radio stream using `os.subprocess` .

acowx, 2020-03-23:
Yes, I started off using the out-of-the-box radio implementation.

It seemed to work on all, aside from m3u streams

shivasiddharth, 2020-03-23:
I would suggest you to use the provided platform for radio streaming. The streaming link that you are using has AAC codec instead i would suggest you to use this http://media-ice.musicradio.com/RadioXUKMP3.m3u which is in MP3 format.

shivasiddharth, 2020-03-23:
Or use this http://ice-sov.musicradio.com/RadioXUKMP3

shivasiddharth, 2020-04-24:
closing this for inactivity

closed 2020-04-24 by shivasiddharth