shivasiddharth/GassistPi #999: Cannot start GassistPi

JSa1987, 2020-03-28:
### Board and OS details:    
```
pi@Apollo:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000027ca57fb
Model		: Raspberry Pi 3 Model B Rev 1.2

pi@Apollo:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
I did a fresh install of Raspbian Buster Lite. The audio devices are working as I can record and play back. I successfully when through the installation process for GassistPi. When I rebooted it seems that the assistant keeps trying to start: the speakers make a sound every minute or so.
If I try to start the assistant manually I get the error reported below.

I have tried to to make some changes to the .asoundrc and the
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/4397926/GassistPi.log)
 /etc/asound.conf files, but this did not help.
```
[... 102 lines cut ...]
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paTimedOut' failed in 'src/os/unix/pa_unix_util.c', line: 387
Expression 'PaUnixThread_New( &stream->thread, &CallbackThreadFunc, stream, 1., stream->rtSched )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2998
Expression 'pthread_join( self->thread, &pret )' failed in 'src/os/unix/pa_unix_util.c', line: 441
Expression 'PaUnixThread_Terminate( &stream->thread, !abort, &threadRes )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3102
2020-03-28 21:11:37,787 : ERROR : [Errno -9987] Wait timed out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 447, in __init__
    pa.start_stream(self._stream)
OSError: [Errno -9987] Wait timed out
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-03-29:
1. There seems to be something wrong with the models. Did you change the models or the sensitivity ? If yes, then what did you change? 
2. Post the outputs of :
```
aplay -l
arecord -l
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
```

JSa1987, 2020-03-29:
Hi,

Thank you for your quick reply. I have an IQaudIO Pi-CodecZero DAC:
[http://iqaudio.co.uk/hats/101-pi-codeczero.html](url)

In the /boot/config.txt file I have disabled the onboard audio and enabled the DAC HAT (dtoverlay=iqaudio-codec).

As per the installation guidelines after instilling and testing the DAC HAT I edit the USB-DAC config file after changing the hardware ids. Attached are the config files I'm currently using.

`pi@Apollo:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: IQaudIOCODEC [IQaudIOCODEC], device 0: IQaudIO CODEC HiFi v1.2 da7213-hifi-0 [IQaudIO CODEC HiFi v1.2 da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

`pi@Apollo:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: IQaudIOCODEC [IQaudIOCODEC], device 0: IQaudIO CODEC HiFi v1.2 da7213-hifi-0 [IQaudIO CODEC HiFi v1.2 da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

I can record sound using:
`arecord --format=S16_LE --duration=5 --rate=16000 --file-type=raw out.raw`

And then play it back without issues using:
`aplay --format=S16_LE --rate=16000 out.raw`

[asound_config_files.zip](https://github.com/shivasiddharth/GassistPi/files/4398811/asound_config_files.zip)

Thank you!

shivasiddharth, 2020-03-29:
Did the audio hat company not provide any audio config file ? Some audio hats will have codec issues which will prevent the usage of both the mic and speaker simultaneously. Try this and revert back with results.

Open one terminal and run:
`arecord -f S16_LE -d 30 -r 16000 --channels 2 --device="hw:0,0" /tmp/test-mic.wav`

Open second terminal and run:
`speaker-test -t wav`

Recording and speaker test command should be run parallelly.

JSa1987, 2020-03-29:
Hi,

The Audio HAT company provides recommended ALSA settings, attached.
[IQaudIO_Codec_OnBoardMIC_record_and_HP_playback.state.zip](https://github.com/shivasiddharth/GassistPi/files/4399014/IQaudIO_Codec_OnBoardMIC_record_and_HP_playback.state.zip)

I added the following line to the /etc/rc.local file to load this at startup:
`sudo alsactl restore -f /home/pi/IQaudIO/IQaudIO_Codec_OnBoardMIC_record_and_HP_playback.state`

I just tried the test you suggested and it worked great: in the recording you can actually hear the speaker test...
[test-mic.wav.zip](https://github.com/shivasiddharth/GassistPi/files/4398997/test-mic.wav.zip)

shivasiddharth, 2020-03-29:
Glad that mic and speaker are working at same time.
1. Do you have pulseaudio installed ?
2. Can you confirm changes to the hotword model sensitivity ?

JSa1987, 2020-03-29:
Hi,

I do not have pulseaudio installed. Should this be installed?

I made no changes to the hotword model sensitivity. Is there anything I should check here?

shivasiddharth, 2020-03-29:
Install pulseaudio and try manually running the assistant.

JSa1987, 2020-03-29:
Hi,

I installed pulseaudio. Now when running play -L I get the following (theJack and Pulse sections were not there before):

> pi@Apollo:/var/lib/alsa $ aplay -L
> null
>     Discard all samples (playback) or generate zero samples (capture)
> jack
>     JACK Audio Connection Kit
> pulse
>     PulseAudio Sound Server
> dsnooper
> dmixer
> default
>     Playback/recording through the PulseAudio sound server
> sysdefault:CARD=IQaudIOCODEC
>     IQaudIOCODEC, IQaudIO CODEC HiFi v1.2 da7213-hifi-0
>     Default Audio Device
> dmix:CARD=IQaudIOCODEC,DEV=0
>     IQaudIOCODEC, IQaudIO CODEC HiFi v1.2 da7213-hifi-0
>     Direct sample mixing device
> dsnoop:CARD=IQaudIOCODEC,DEV=0
>     IQaudIOCODEC, IQaudIO CODEC HiFi v1.2 da7213-hifi-0
>     Direct sample snooping device
> hw:CARD=IQaudIOCODEC,DEV=0
>     IQaudIOCODEC, IQaudIO CODEC HiFi v1.2 da7213-hifi-0
>     Direct hardware device without any conversions
> plughw:CARD=IQaudIOCODEC,DEV=0
>     IQaudIOCODEC, IQaudIO CODEC HiFi v1.2 da7213-hifi-0
>     Hardware device with all software conversions
> usbstream:CARD=IQaudIOCODEC
>     IQaudIOCODEC
>     USB Stream Output

However I still unable to start the assistant manually:
> pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id project-1087197276759
> /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
>   from collections import Mapping, Container
> /home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
>   if not isinstance(key, collections.Hashable):
> INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
> INFO:gmusicapi.Mobileclient1:version: 13.0.0
> INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
> INFO:gmusicapi.Mobileclient1:initialized
> INFO:gmusicapi.Mobileclient1:logged out
> /home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
>   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
> 2020-03-29 16:13:25,372 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
> WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
> INFO:gmusicapi.Mobileclient1:failed to authenticate
> Initializing GPIOs 5 and 6 for assistant activity indication
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... 20 lines cut ...]
> ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
> ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
> ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> Expression 'paTimedOut' failed in 'src/os/unix/pa_unix_util.c', line: 387
> Expression 'PaUnixThread_New( &stream->thread, &CallbackThreadFunc, stream, 1., stream->rtSched )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2998
> Expression 'pthread_join( self->thread, &pret )' failed in 'src/os/unix/pa_unix_util.c', line: 441
> Expression 'PaUnixThread_Terminate( &stream->thread, !abort, &threadRes )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3102
> 2020-03-29 16:13:44,649 : ERROR : [Errno -9987] Wait timed out
> Traceback (most recent call last):
>   File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
>     Myassistant().main()
[... cut ...] _init__
>     self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
>   File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
>     stream_callback=audio_callback)
>   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
>     stream = Stream(self, *args, **kwargs)
>   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 447, in __init__
>     pa.start_stream(self._stream)
> OSError: [Errno -9987] Wait timed out

shivasiddharth, 2020-03-29:
Do you have any other DAC or audio cards that you can test with ? Seems like a conflict between alsa and pyaudio.

shivasiddharth, 2020-03-29:
Remove this DAC card's alsa state when you try some other DAC.

JSa1987, 2020-03-29:
Hi Shivasiddharth,

I made some progress.

I disabled the DAC HAT in the /boot/config.txt and enabled the on board DAC of the Pi3B+. I then disabled the state file for the DAC HAT, plugged in a USB webcam (that has a mic) and rebooted.

I then setup the .asoundrc and the /etc/asound.conf files to use the webcam mic and the Pi3B+ onboard DAC. I was then able to start the assistant manually without issues.

Next I enabled the DAC HAT in the /boot/config.txt file and rebooted. With the  .asoundrc and the /etc/asound.conf files still set up to use the webcam mic and the Pi3B+ onboard DAC the assistant would start manually as expected. 
At this point I tried to edit the .asoundrc and the /etc/asound.conf files to use the DAC HAT for the speakers, but the assistant would still use the Pi3B+ onboard DAC.

I so edit the /boot/config.txt to disable the Pi3B+ onboard DAC, and rebooted. I adjusted the .asoundrc and the /etc/asound.conf files to use the USB webcam as mic and the DAC HAT for the speakers. With this setup I was able to start the assistant manually. 
I tried again to edit the .asoundrc and the /etc/asound.conf files to use the DAC HAT onboard mic, but by doing to the assistant failed to start.

I have now have the .asoundrc and the /etc/asound.conf to use the USB webcam as mic and the DAC HAT for the speakers. Everything works as expected and the assistant automatically start at boot.

So the current status is that we have narrowed down this issue as some sort of conflict when using the DAC HAT mic. Could pyaudio have issues using the same card/device both for the speakers and for the mic?
Is there anything I could try to find the route cause of this issue?

Thank you!

shivasiddharth, 2020-03-30:
Try if this script is working https://github.com/shivasiddharth/GassistPi/blob/e9896a757cfe64e0e9bc3502084ca4567ea9a0c5/src/audiorecorder.py

JSa1987, 2020-03-30:
Hi shivasiddharth,

I added the following lines to the bottom of the scrip you linked:
```
if __name__ == "__main__":
         record_to_file(sys.argv[1])
```

I first run this with the USB webcam plugged in. It recorded as expected.

I then physically unplugged the USB webcam, adjusted the .asoundrc file to use the HAT build in microphone and run the script again. Also this time it worked as expected!

The following warnings came out:
```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/audiorecorder.py '/home/pi/test.wav'
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5014:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
```

At this point, without changing anything in the .asoundrc file, I tried to manually start the assistant again but I still get the same error as before:
```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id raspberry-pi-6b03d --device-model-id raspberry-pi-6b03d-josto-amplifier-jwov8b
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
[... 48 lines cut ...]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paTimedOut' failed in 'src/os/unix/pa_unix_util.c', line: 387
Expression 'PaUnixThread_New( &stream->thread, &CallbackThreadFunc, stream, 1., stream->rtSched )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2998
Expression 'pthread_join( self->thread, &pret )' failed in 'src/os/unix/pa_unix_util.c', line: 441
[... cut ...]
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 447, in __init__
    pa.start_stream(self._stream)
OSError: [Errno -9987] Wait timed out
```

So in conclusion seems that pyaudio works well with the HAT's mic, but the assistant doesn't like it...

shivasiddharth, 2020-03-30:
Your command that you have posted for manually starting is a bit wrong it should be --device_model_id and not --device-model-id 

See if it works. You should always include both project id and model id

shivasiddharth, 2020-03-30:
Also post the outputs of:
pip freeze
and
source env/bin/activate
pip3 freeze

JSa1987, 2020-03-30:
Hi shivasiddharth,

I tired manually starting the assistant with --device_model_id and I get the same error:
```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id raspberry-pi-6b03d --device_model_id raspberry-pi-6b03d-josto-amplifier-jwov8b
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-03-30 17:45:41,945 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5014:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dmix.c:1043:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paTimedOut' failed in 'src/os/unix/pa_unix_util.c', line: 387
Expression 'PaUnixThread_New( &stream->thread, &CallbackThreadFunc, stream, 1., stream->rtSched )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2998
Expression 'pthread_join( self->thread, &pret )' failed in 'src/os/unix/pa_unix_util.c', line: 441
Expression 'PaUnixThread_Terminate( &stream->thread, !abort, &threadRes )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3102
2020-03-30 17:45:48,827 : ERROR : [Errno -9987] Wait timed out
Traceback (most recent call last):
[... 96 lines cut ...]
pycparser==2.20
pycryptodomex==3.7.2
pyOpenSSL==19.1.0
python-dateutil==2.8.1
python-magic==0.4.15
python-vlc==3.0.7110
pytz==2019.3
PyYAML==3.13
requests==2.23.0
requests-oauthlib==1.3.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.14.0
sounddevice==0.3.15
soupsieve==2.0
spidev==3.4
spotipy==2.10.0
tenacity==4.12.0
uritemplate==3.0.1
urllib3==1.25.8
validictory==1.1.2
websocket-client==0.57.0
youtube-dl==2020.3.24
zeroconf==0.24.5
```

shivasiddharth, 2020-03-30:
I did some digging and I think you should get in touch with that HAT manufacturer for the error. Its an pyaudio-alsa error raised by the HAT.

JSa1987, 2020-03-30:
Maybe pyaudio cannot handle using it as speaker and mic at the same time?

shivasiddharth, 2020-03-30:
Do you have the pinout diagram for that HAT ? If not not try and get the pinout ? I will see what best i can do

JSa1987, 2020-03-30:
Shivasiddharth,

Thank you for helping out with this, your help is much appreciated!

The documentation for the HAT is available at:
[http://www.iqaudio.com/downloads/IQaudIO.pdf](url) 
There is not much in this document, but there are some basic info. The type I have is the Pi-CodecZero.

According to the documentation (page 40) it uses the following pins of the GPIO:
- Pin 1   3.3v
- Pin 2  5v
- Pin 3  GPIO2 (I2C SDA1)
- Pin 4  5v
- Pin 5  GPIO3 (I2C SCL1)
- Pin 6  Ground
- Pin 9  Ground
- Pin 12 GPIO18 (I2S CLK) 
- Pin 13 GPIO27 (GEN2) => Hat button
- Pin 14  Ground
- Pin 16 GPIO23 (GEN4) => HAT LED 1
- Pin 17  3.3v
- Pin 18 GPIO24 (GEN5) => HAT LED 2
- Pin 20  Ground
- Pin 25  Ground
- Pin 30  Ground
- Pin 34  Ground
- Pin 35 GPIO19 (I2S)
- Pin 38 GPIO20 (I2S DIN)
- Pin 39  Ground
- Pin 40 GPIO21 (I2S DOUT) 

[image]

shivasiddharth, 2020-03-31:
I have some GPIOs declared in the config.yaml file. Please see if any of those GPIOs are coinciding with the GPIOs used by the HAT.  Replace the existing main.py with this https://github.com/shivasiddharth/GassistPi/blob/Just-Google-Assistant/src/main.py and see if this one works.

JSa1987, 2020-03-31:
Hi shivasiddharth,

I reviewed the config.yaml there were some GPIO declared there that are used in other IQaudIO HATs. According to the documentation these shouldn't be used on my HAT, but to be safe I changed these in the config.yaml file.

I edit the .asoundrc file to use the HAT's mic, disconnected the USB webcam and tried to start the assistant manually. Still not luck.

I then replaced the main.py file with the one that you provided and it worked! The assistant started up with now errors, it does the start up sound and says "Hi". At that point if I say "Hi google, what time is it?" from the terminal output it seems that the assistant recognize the voice... but there is no reply.... 

```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id raspberry-pi-6b03d --device_model_id raspberry-pi-6b03d-josto-amplifier-jwov8b
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: raspberry-pi-6b03d-josto-amplifier-jwov8b
device_id: 0030E48CD4FBFB0CE394528563CAB97F

ON_MUTED_CHANGED:
  {"is_muted": false}
Mic mute is set to: False
ON_START_FINISHED
ON_MEDIA_STATE_IDLE
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
{'is_error_response': False}
```

shivasiddharth, 2020-04-01:
Assistant freezing at 
```
ON_RESPONDING_STARTED:
  {"is_error_response": false}
{'is_error_response': False}
```
Is an issue with your DAC. The last main.py I gave you has no third party libraries. You need to discuss this issue further with your HAT manufacturer/seller. In one of the message here https://github.com/shivasiddharth/GassistPi/issues/999#issuecomment-605641179 you mentioned that Mic and Speaker both worked well. In that test what was your Audio out if it was pi's audio, then the assistant getting stuck is completely understandable.

shivasiddharth, 2020-04-01:
Even one of the  Seeed studio's HAT has this issue. I have opened an issue there on behalf of my users here. I suggest you similarly do that with your HAT manufacturer.

Floper, 2020-04-03:
Hi there,

I have the same problem. My assistant is stoping after:
```
ON_RESPONDING_STARTED:
  {"is_error_response": false}
```
like what you described above. Only one "{"is_error_response": false}".
I have a ReSpeaker 2Mic HAT from seed studio.

Speaker test and recoding test works well separately. But you are correct they cannot work at the same time as you reported here: https://github.com/respeaker/seeed-voicecard/issues/219

I've investigating that issue quite a lot for the past days and couldn't figure it out. On my  Google Activity I can see the Assistant is getting my request and his answering. Under the Google Cloud console I see that all my request are handled with a "google.internal.speech.S3OnePlatform.Assistant" error.

What I don't understand is that I used to have GassitPi working with this HAT before. What changed ? The Assistant SDK ?
Is there any workaround ? Coming back to a previous version of the repo for the HAT ?

Let me know if I should open another issue.

Thanks for your work and active support.

JSa1987, 2020-04-03:
Hi there,

I think Floper is right, something must have changed.

My Raspberry Pi was originally running Raspbian Stretch and I originally installed GassistPi around April or May 2019. At that time I was using an USB mic and the onboard DAC.
About a month ago I added the IQaudIO Pi-CodecZero HAT to this installation and the assistant was working fine using the HAT's mic and DAC.

Last week I made a fresh installation of Raspbian Buster and installed the latest version of GassistPi, and this is when I started having these issues...

Floper, 2020-04-04:
Good to know I am not the only one JSa1987.

I also did a fresh installation of Raspbian and GassitPi to see if that
could solve the problem. No luck.

I did try to run the main.py shared by Shivasiddharth above but same
problem.

I also tried to run the Google Assistant main from Google but same problem.
I therefore conclude that it's either coming from Google (and might be
linked to the fact that the SDK is depreciated) or to the HAT. But JSa1987
you are using another HAT so it seems to be related to Google assistant SDK.

I also made a brand new project under the Google Cloud Console as I read in
another git issue that it could be related to a too old project (
https://github.com/google/aiyprojects-raspbian/issues/448). But no luck
either.

JSa1987 do you get the same
"google.internal.speech.S3OnePlatform.Assistant" error on the Google Cloud
sonsole?

So the 2 reasons I can come up with at the moment :
- Google is not sending the answers to the SDK anymore
- Google assistant SDK changed and it is now needed that both microphone
and speaker works at the same time and our HATs cannot do it.

I am pretty sure Shivasiddharth knows.

shivasiddharth, 2020-04-04:
Just a shot in the dark, can you guys try fresh using Stretch and not buster.

Floper, 2020-04-04:
I am actually using Strech.

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

JSa1987, 2020-04-04:
Floper,

I don't think the issue is that:

> Google assistant SDK changed and it is now needed that both microphone and speaker works at the same time and our HATs cannot do it.

On my setup I tried to open two terminal run speaker-test from one and record from the mic on the other one. It all worked well, the mic was actually picking up the speaker-test. So, at least for my HAT, the mic and the speaker can be accessed at the same time.

shivasiddharth, 2020-04-04:
1. The issue is not Google Assistant SDK's. I have been working with Amazon Alexa as well The same issue is present in those project as well. 

2. The problem is with Alsa. It was all fine till Alsa version 1.1.3. Revision to Alsa let loose a cat among pigeons causing a number of audio related issues (Google up issues related to Alsa 1.1.4 and up).

3. The reason I asked you guys to try in an old version of Stretch is because, earlier releases of the OS came with non troublesome Alsa 1.1.3.  But some or the other program will eventually run a sudo apt-get update or sudo apt-get upgrade bumping the Alsa version.

4. I have worked with a number of Audio HATs and DACs. This particular issue is present mostly in HATs using AGC (Automatic Gain Control) unfortunately both of you HAT has that.

Only HAT manufacturers have to step in and update their drivers.

Floper, 2020-04-04:
It seems that I have 1.1.3

~ $ aplay --version
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>

So in theory I should be fine?

shivasiddharth, 2020-04-04:
@Floper  As you are in Alsa 1.1.3, This should work.

1. Uninstall the existing Respeaker installation.
2. Delete the seeed-voicecard folder in /home/pi/
3. Reboot.
4. Use the Respeaker installation process that i have given in the README (using my repo not seeed studio's).
5. Open a terminal and run: 
```
sudo apt-get purge pulseaudio
sudo rm /etc/asound.conf
sudo rm ./.asoundrc
```
6. Reboot
7. Now speaker should work if mic is open.

shivasiddharth, 2020-04-04:
@JSa1987  Try going back to Stretch if you can.

Floper, 2020-04-04:
Thanks @shivasiddharth I just tried but I am getting a problem in step 1.
I tried to uninstall seed-voicecard:
`$ sudo ./uninstall.sh`
But it get stuck at the first operation:
`remove dtbos`

I looked for a related issue on the seed-voicecard repo didn't find anything for the moment.

JSa1987, 2020-04-04:
Hi,

I have just did a fresh installation of Stretch and then a fresh installation of GassistPi.
At the end of the installation process I got the following error:

`ERROR: pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible.`

Now if I try to manually start the assistant if hear the boot up sound, it says "Hi". However the assistant does not work and on the terminal I get the following:

```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'raspberry-pi-6b03d'  --device-model-id 'raspberry-pi-6b03d-josto-amplifier-jwov8b'
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-04-04 18:09:10,437 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc

```

The Stopped - stopping vlc repeats forever

Floper, 2020-04-04:
Sorry JSa1987 I have no ideas how to help you here.

I solved my uninstall problem by uncommenting the line dtparam=audio=on in /boot/config.txt (I did it based on that issue : https://github.com/shivasiddharth/GassistPi/issues/89

I followed the steps you indicated, and you are correct mic and speaker works simultaneously now.

And I just tried Google Assistant and it works ! Thanks a lot.

JSa1987, 2020-04-04:
Hi,

I realized that I was pushbutton.py instead of main.py.

Now when I run main.py it doesn't crash... however it gets stuck (I can't even terminate it with Control+C):

```
pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id raspberry-pi-6b03d --device_model_id raspberry-pi-6b03d-josto-amplifier-jwov8b
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-04-04 20:49:58,670 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.IQaudIOCODEC.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
[... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paTimedOut' failed in 'src/os/unix/pa_unix_util.c', line: 387
Expression 'PaUnixThread_New( &stream->thread, &CallbackThreadFunc, stream, 1., stream->rtSched )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2998
Expression 'pthread_join( self->thread, &pret )' failed in 'src/os/unix/pa_unix_util.c', line: 441
```

Now this happens even if I have the USB mic connected and  .asoundrc set up to use that....

shivasiddharth, 2020-04-05:
@JSa1987  post the version of alsa using `aplay --version`

JSa1987, 2020-04-05:
That is version 1.1.3:
```
pi@Apollo:~ $ aplay --version
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>
```

shivasiddharth, 2020-04-05:
Try:
```
sudo apt-get purge pulseaudio
sudo reboot
```

Let me know what happens. Anyway you need to get the asound.conf for your DAC from your manufacturer.

JSa1987, 2020-04-05:
Thank you Shivasiddharth! This did the trick!
I now have the assistant up an running using the mic and the DAC of the IQaudIO Pi-CodecZero HAT.

As you mentioned there must be a compatibility issue between this HAT and the current ALSA version in Buster. But it works just fine with ALSA 1.1.3 included in Stretch.

shivasiddharth, 2020-04-05:
Closing this issue then.

closed 2020-04-05 by shivasiddharth