shivasiddharth/GassistPi #1000: Installation issue

Shidiqadm, 2020-04-07:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@Spotify:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000108bcdb2
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@Spotify:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@Spotify:~ $ 

### Describe the bug:
```
during install it shows a error "sed: -e expression #1, char 19: unterminated `s' command "
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_779423133407-j7drclqpc9l0kpeooek0b1mn7fke7iai.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: pi-assistant-4b739

Enter the modelid that was generated in the actions console: Pi Assistant

Hit:1 http://archive.raspberrypi.org/debian buster InRelease                   
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                
Hit:3 https://dtcooper.github.io/raspotify raspotify InRelease              
Reading package lists... Done                         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.16.1-4).
bison is already the newest version (2:3.3.2.dfsg-1).
build-essential is already the newest version (12.6).
gcc is already the newest version (4:8.3.0-1+rpi2).
git is already the newest version (1:2.20.1-2+deb10u1).
libatlas-base-dev is already the newest version (3.10.3-8+rpi1).
libffi-dev is already the newest version (3.2.1-9).
libjack-jackd2-dev is already the newest version (1.9.12~dfsg-2).
libmpg123-dev is already the newest version (1.25.10-2).
libpcre3 is already the newest version (2:8.39-12).
libpcre3-dev is already the newest version (2:8.39-12).
libsox-fmt-mp3 is already the newest version (14.4.2+git20190427-1).
libxml2-dev is already the newest version (2.9.4+dfsg1-7+b3).
libxslt1-dev is already the newest version (1.1.32-2.2~deb10u1).
mpg123 is already the newest version (1.25.10-2).
mplayer is already the newest version (2:1.3.0-8+b5).
mpv is already the newest version (0.29.1-1).
nmap is already the newest version (7.70+dfsg1-6).
portaudio19-dev is already the newest version (19.6.0-1).
pulseaudio is already the newest version (12.2-4+deb10u1).
python is already the newest version (2.7.16-1).
python-dev is already the newest version (2.7.16-1).
python-setuptools is already the newest version (40.8.0-1).
python3 is already the newest version (3.7.3-1).
python3-dev is already the newest version (3.7.3-1).
python3-lxml is already the newest version (4.3.2-1).
python3-setuptools is already the newest version (40.8.0-1).
python3-venv is already the newest version (3.7.3-1).
socat is already the newest version (1.7.3.2-2).
sox is already the newest version (14.4.2+git20190427-1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u2+rpt1).
python-pip is already the newest version (18.1-5+rpt1).
python3-pip is already the newest version (18.1-5+rpt1).
vlc is already the newest version (3.0.8-0+deb10u1+rpt7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages (0.2.11)

Checking OS Compatability

You are running the installer on Buster

Your board supports Ok-Google Hotword. You can also trigger the assistant using custom-wakeword

GPIO pins can be used with the assistant

Copying Snowboy files to GassistPi directory

Changing particulars in service files

Changing particulars in service files for Ok-Google hotword
sed: -e expression #1, char 19: unterminated `s' command

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-04-07:
1. Did you run the installer multiple times ?
2. Did you modify anything (installer script or service file) ?

closed 2020-04-07 by shivasiddharth

Shidiqadm, 2020-04-07:
no I haven't change anything

Shidiqadm, 2020-04-07:
but due to network failure I run the installer again

Shidiqadm, 2020-04-07:
also I already have a project in my cloud console but I cannot view the model id in actions on google

Shidiqadm, 2020-04-07:
when I try to create new project it dosent shows me device registration option also dosent shows any developer options other than overview.

commit b323258b (2020-04-07): Update gassist-installer.sh

Fixes #1000

Fix for installer crashing if run multiple times.