shivasiddharth/GassistPi #1001: Problem after upgrade from a 2018 version

VITOAMOROSO82, 2020-04-09:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 0000000012e56a29

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
[... 521 lines cut ...]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 987, in main
    with Assistant(credentials, device_model_id) as assistant:
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assist                                                                                                             ant.py", line 106, in __init__
    credentials, self._set_credentials)
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/auth_h                                                                                                             elpers.py", line 40, in __init__
    self._credentials.refresh(http_request)
  File "/home/pi/env/lib/python3.5/site-packages/google/oauth2/credentials.py",                                                                                                              line 183, in refresh
    self._scopes,
  File "/home/pi/env/lib/python3.5/site-packages/google/oauth2/_client.py", line                                                                                                              248, in refresh_grant
    response_data = _token_endpoint_request(request, token_uri, body)
  File "/home/pi/env/lib/python3.5/site-packages/google/oauth2/_client.py", line                                                                                                              105, in _token_endpoint_request
    response = request(method="POST", url=token_uri, headers=headers, body=body)
  File "/home/pi/env/lib/python3.5/site-packages/google/auth/transport/requests.                                                                                                             py", line 186, in __call__
    six.raise_from(new_exc, caught_exc)
  File "<string>", line 3, in raise_from
google.auth.exceptions.TransportError: HTTPSConnectionPool(host='accounts.google                                                                                                             .com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by SSLE                                                                                                             rror(SSLError("bad handshake: Error([('', 'osrandom_rand_bytes', 'getrandom() in                                                                                                             itialization failed.')],)",),))
2020-04-09 21:41:27,264 : DEBUG : Making request: POST https://accounts.google.c                                                                                                             om/o/oauth2/token
2020-04-09 21:41:27,282 : DEBUG : Starting new HTTPS connection (1): accounts.go                                                                                                             ogle.com:443
2020-04-09 21:41:27,539 : DEBUG : https://accounts.google.com:443 "POST /o/oauth                                                                                                             2/token HTTP/1.1" 200 None
2020-04-09 22:00:27,711 : DEBUG : Making request: POST https://accounts.google.c                                                                                                             om/o/oauth2/token
2020-04-09 22:00:27,730 : DEBUG : Starting new HTTPS connection (1): accounts.go                                                                                                             ogle.com:443
2020-04-09 22:00:27,987 : DEBUG : https://accounts.google.com:443 "POST /o/oauth                                                                                                             2/token HTTP/1.1" 200 None

shivasiddharth, 2020-04-10:
I am kind of lost here. 

Is the Mic not working if you actually boot from USB or is the Mic not working just with the addition of `program_usb_boot_mode=1`. 

Anyways, can you share the output of  `aplay -l` and `arecord -l` from your mic non-working image ?

VITOAMOROSO82, 2020-04-10:
> I am kind of lost here.
> 
> Is the Mic not working if you actually boot from USB or is the Mic not working just with the addition of `program_usb_boot_mode=1`.
> 
> Anyways, can you share the output of `aplay -l` and `arecord -l` from your mic non-working image ?

Hi, It's not working in any case, I meant that I found this issue only AFTER I added the program_usb_boot_mode=1, but I cannot be sure at 100% that the issue happened for this parameter. I think this because If I roll-back to 2018 SD card image everything works fine, even if, as you know, the _program_usb_boot_mode_ parameter is eternal :-)

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 6/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**** List of CAPTURE Hardware Devices ****
card 1: U0x46d0x81b [USB Device 0x46d:0x81b], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2020-04-10:
So correct me if I am wrong. 

You have this image of 2018 which boots from USB and Mic works well. And now you have this new installation image booted from USB and Mic does not work ? 

Is the new fresh installation also on stretch or buster ?

VITOAMOROSO82, 2020-04-10:
> So correct me if I am wrong.
> 
> You have this image of 2018 which boots from USB and Mic works well. And now you have this new installation image booted from USB and Mic does not work ?
> 
> Is the new fresh installation also on stretch or buster ?

Even if I boot from USB or SD card, the 2018 image works correctly. In the new installation, that is THE SAME 2018 "stretch" image in which I uninstalled everything, updated packages without kernel due to LIRC issues, my MIC does not works even If I boot via USB or SD card

VITOAMOROSO82, 2020-04-10:
Now I'm trying to install from the 2018 image in which I only disabled the old "gassist-ok-google.service", deleted Gassist folder and no "sudo apt-get upgrade" command executed, I want to understand If there are problems with some packages. 

As you can understand, because from Dec. 2018 I didn't upgrade packages, there are a lot of updates to be done, when I upgraded them 3 days ago I blocked the following:

raspberrypi-kernel
raspberrypi-sys-mods
raspberrypi-ui-mods
raspi-gpio
systemd-sysv
wiringpi

If you ask me "why" I choose to not update them because I understood they are related in deep with the kernel and gpio, and in fact after I installed everything, reinstalled the last version of Gassist everything works fine with my MIC. I think something happened after the USB boot mod...

shivasiddharth, 2020-04-10:
The project has undergone a lot of changes since 2018. Both with respect to packages and files structure. If you cloned the git again and ran the provided installer things should work fine. Deleting the GassistPi folder has very little effect, the packages are there in the env

VITOAMOROSO82, 2020-04-10:
Ok now It works!!! So there's a problem in one of this packages If I upgrade:

apt/oldstable 1.4.9 armhf [aggiornabile da: 1.4.8]
apt-transport-https/oldstable 1.4.9 armhf [aggiornabile da: 1.4.8]
apt-utils/oldstable 1.4.9 armhf [aggiornabile da: 1.4.8]
base-files/oldstable 9.9+rpi1+deb9u11 armhf [aggiornabile da: 9.9+rpi1+deb9u4]
bluez-firmware/stable 1.2-3+rpt7 all [aggiornabile da: 1.2-3+rpt6]
chromium-browser/stable 72.0.3626.121-0+rpt4 armhf [aggiornabile da: 65.0.3325.181-0+rpt4]
chromium-browser-l10n/stable 72.0.3626.121-0+rpt4 all [aggiornabile da: 65.0.3325.181-0+rpt4]
chromium-codecs-ffmpeg-extra/stable 72.0.3626.121-0+rpt4 armhf [aggiornabile da: 65.0.3325.181-0+rpt4]
curl/oldstable 7.52.1-5+deb9u10 armhf [aggiornabile da: 7.52.1-5+deb9u8]
dbus/oldstable 1.10.28-0+deb9u1 armhf [aggiornabile da: 1.10.26-0+deb9u1]
dbus-user-session/oldstable 1.10.28-0+deb9u1 all [aggiornabile da: 1.10.26-0+deb9u1]
dbus-x11/oldstable 1.10.28-0+deb9u1 armhf [aggiornabile da: 1.10.26-0+deb9u1]
e2fslibs/oldstable 1.43.4-2+deb9u1 armhf [aggiornabile da: 1.43.4-2]
e2fsprogs/oldstable 1.43.4-2+deb9u1 armhf [aggiornabile da: 1.43.4-2]
erlang-base/oldstable 1:19.2.1+dfsg-2+deb9u2 armhf [aggiornabile da: 1:19.2.1+dfsg-2+deb9u1]
erlang-crypto/oldstable 1:19.2.1+dfsg-2+deb9u2 armhf [aggiornabile da: 1:19.2.1+dfsg-2+deb9u1]
erlang-syntax-tools/oldstable 1:19.2.1+dfsg-2+deb9u2 armhf [aggiornabile da: 1:19.2.1+dfsg-2+deb9u1]
ffmpeg/stable 7:3.2.14-1~deb9u1+rpt1 armhf [aggiornabile da: 7:3.2.10-1~deb9u1+rpt2]
file/oldstable 1:5.30-1+deb9u3 armhf [aggiornabile da: 1:5.30-1+deb9u2]
fonts-opensymbol/oldstable 2:102.7+LibO5.2.7-1+rpi1+deb9u9 all [aggiornabile da: 2:102.7+LibO5.2.7-1+rpi1+deb9u3]
fuse/oldstable 2.9.7-1+deb9u2 armhf [aggiornabile da: 2.9.7-1+deb9u1]
geany/stable 1.29-1+rpt2 armhf [aggiornabile da: 1.29-1+rpt1]
geany-common/stable 1.29-1+rpt2 all [aggiornabile da: 1.29-1+rpt1]
gettext/oldstable 0.19.8.1-2+deb9u1 armhf [aggiornabile da: 0.19.8.1-2]
gettext-base/oldstable 0.19.8.1-2+deb9u1 armhf [aggiornabile da: 0.19.8.1-2]
git-man/oldstable 1:2.11.0-3+deb9u5 all [aggiornabile da: 1:2.11.0-3+deb9u4]
gnupg/oldstable 2.1.18-8~deb9u4 armhf [aggiornabile da: 2.1.18-8~deb9u2]
gnupg-agent/oldstable 2.1.18-8~deb9u4 armhf [aggiornabile da: 2.1.18-8~deb9u2]
gpgv/oldstable 2.1.18-8~deb9u4 armhf [aggiornabile da: 2.1.18-8~deb9u2]
gstreamer1.0-alsa/oldstable 1.10.4-1+deb9u1 armhf [aggiornabile da: 1.10.4-1]
gstreamer1.0-plugins-base/oldstable 1.10.4-1+deb9u1 armhf [aggiornabile da: 1.10.4-1]
gstreamer1.0-x/oldstable 1.10.4-1+deb9u1 armhf [aggiornabile da: 1.10.4-1]
gtk2-engines-clearlookspix/stable 1:2.20.7 armhf [aggiornabile da: 1:2.20.5]
icu-devtools/oldstable 57.1-6+deb9u4 armhf [aggiornabile da: 57.1-6+deb9u2]
java-common/oldstable 0.58+deb9u1 all [aggiornabile da: 0.58]
libapt-inst2.0/oldstable 1.4.9 armhf [aggiornabile da: 1.4.8]
libapt-pkg5.0/oldstable 1.4.9 armhf [aggiornabile da: 1.4.8]
libarchive13/oldstable 3.2.2-2+deb9u2 armhf [aggiornabile da: 3.2.2-2]
libasound2/stable 1.1.3-5+rpt4 armhf [aggiornabile da: 1.1.3-5+rpi3]
libasound2-data/stable 1.1.3-5+rpt4 all [aggiornabile da: 1.1.3-5+rpi3]
libasound2-dev/stable 1.1.3-5+rpt4 armhf [aggiornabile da: 1.1.3-5+rpi3]
libaudiofile1/oldstable 0.3.6-4+deb9u1 armhf [aggiornabile da: 0.3.6-4]
libav-tools/stable 7:3.2.14-1~deb9u1+rpt1 all [aggiornabile da: 7:3.2.10-1~deb9u1+rpt2]
libbasicusageenvironment1/oldstable 2016.11.28-1+deb9u2 armhf [aggiornabile da: 2016.11.28-1]
libc-bin/oldstable 2.24-11+deb9u4 armhf [aggiornabile da: 2.24-11+deb9u3]
libc-dev-bin/oldstable 2.24-11+deb9u4 armhf [aggiornabile da: 2.24-11+deb9u3]
libc-l10n/oldstable 2.24-11+deb9u4 all [aggiornabile da: 2.24-11+deb9u3]
libc6/oldstable 2.24-11+deb9u4 armhf [aggiornabile da: 2.24-11+deb9u3]
libc6-dbg/oldstable 2.24-11+deb9u4 armhf [aggiornabile da: 2.24-11+deb9u3]
libc6-dev/oldstable 2.24-11+deb9u4 armhf [aggiornabile da: 2.24-11+deb9u3]
libcaca0/oldstable 0.99.beta19-2.1~deb9u1 armhf [aggiornabile da: 0.99.beta19-2]
libcomerr2/oldstable 1.43.4-2+deb9u1 armhf [aggiornabile da: 1.43.4-2]
libcups2/oldstable 2.2.1-8+deb9u5 armhf [aggiornabile da: 2.2.1-8+deb9u2]
libcupsfilters1/stable 1.11.6-3+rpt2+deb9u1 armhf [aggiornabile da: 1.11.6-3+rpi1]
libcupsimage2/oldstable 2.2.1-8+deb9u5 armhf [aggiornabile da: 2.2.1-8+deb9u2]
libcurl3/oldstable 7.52.1-5+deb9u10 armhf [aggiornabile da: 7.52.1-5+deb9u8]
libcurl3-gnutls/oldstable 7.52.1-5+deb9u10 armhf [aggiornabile da: 7.52.1-5+deb9u8]
libdbus-1-3/oldstable 1.10.28-0+deb9u1 armhf [aggiornabile da: 1.10.26-0+deb9u1]
libebml4v5/oldstable 1.3.4-1+deb9u1 armhf [aggiornabile da: 1.3.4-1]
libexif12/oldstable 0.6.21-2+deb9u1 armhf [aggiornabile da: 0.6.21-2]
libexpat1/oldstable 2.2.0-2+deb9u3 armhf [aggiornabile da: 2.2.0-2+deb9u1]
libexpat1-dev/oldstable 2.2.0-2+deb9u3 armhf [aggiornabile da: 2.2.0-2+deb9u1]
libfaad2/oldstable 2.8.0~cvs20161113-1+deb9u2 armhf [aggiornabile da: 2.8.0~cvs20161113-1+deb9u1]
libfreetype6/oldstable 2.6.3-3.2+deb9u1 armhf [aggiornabile da: 2.6.3-3.2]
libfreetype6-dev/oldstable 2.6.3-3.2+deb9u1 armhf [aggiornabile da: 2.6.3-3.2]
libfribidi0/oldstable 0.19.7-1+deb9u1 armhf [aggiornabile da: 0.19.7-1]
libfuse2/oldstable 2.9.7-1+deb9u2 armhf [aggiornabile da: 2.9.7-1+deb9u1]
libgd3/oldstable 2.2.4-2+deb9u5 armhf [aggiornabile da: 2.2.4-2+deb9u2]
[... 170 lines cut ...]
samba-libs/oldstable 2:4.5.16+dfsg-1+deb9u2 armhf [aggiornabile da: 2:4.5.12+dfsg-2+deb9u3]
scratch2/stable 0.23 armhf [aggiornabile da: 0.21]
[... cut ...]
wpasupplicant/stable 2:2.6-21~bpo9~rpt1 armhf [aggiornabile da: 2:2.4-1+deb9u1]
xxd/oldstable 2:8.0.0197-4+deb9u3 armhf [aggiornabile da: 2:8.0.0197-4+deb9u1]
zenity/stable 3.28.1-1.1~bpo9+rpt1 armhf [aggiornabile da: 3.28.1-1.1~bpo9+1]
zenity-common/stable 3.28.1-1.1~bpo9+rpt1 all [aggiornabile da: 3.28.1-1.1~bpo9+1]

Do you think you can find some of them ( obviously the kernel I cannot upgrade ) that I need to skip? Or because now It works you think "leave everything this way and fuck you"!!! :-)

shivasiddharth, 2020-04-10:
Skip ALSA upgrade for sake. Check your Alsa version using `aplay --version` if it is 1.1.3 or earlier, skip Alsa upgrade. Also, if Alsa version is 1.1.3 or earlier, uninstall pulseaudio that gets installed whilst installing GA.

VITOAMOROSO82, 2020-04-10:
> Skip ALSA upgrade for God sake. Check your Alsa version using `aplay --version` if it is 1.1.3 or earlier, skip Alsa upgrade. Also, if Alsa version is 1.1.3 or earlier, uninstall pulseaudio that gets installed whilst installing GA.

Ok, before this I create a new image for rollback :-) I'll keep you update on this...

For the moment just another question...When I switch to it-IT language, my customized command used for LIRC are answered by a let me say "transexual voice" ahahahah...It's a little slowly and strange, how can I fix It?

shivasiddharth, 2020-04-10:
Will fix that.

VITOAMOROSO82, 2020-04-10:
Regarding the USB boot, I can confirm that It works also with the new version, but I have to disconnect the USB device because It doesn't boot, and after I connect the USB mic and the service is working well...Anyway no problem, the USB key needs only for backup/restore of the MicroSD using rpi-clone.

Regarding the list of packages, I confirm that I have alsa 1.1.3 but I cannot see any Alsa upgrade, so some other packages must be the problem...I think I will update only things like samba, chromium, apt and leave everything untouched, doesn't matter. If you want that I do other test for you to understand whick package could be the issue, I can help without problems.

Thank you very much for your work and your help, very appreciated :-)

closed 2020-04-11 by VITOAMOROSO82