shivasiddharth/GassistPi #1002: youtube restarted service and not work youtube api

koncza, 2020-04-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```rasperry pi3
```
Paste your Board and OS details here.   
```
Raspbian
### Describe the bug:
```
if i give the command to play ....... on youtube, the service will restart.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/no

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes
[
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/4511703/GassistPi.log)
]([GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/4511691/GassistPi.log))

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-04-22:
You have configured the YouTube API incorrectly. You have set a limit in the project console and that is blocking your request. You need to fix that.

koncza, 2020-04-22:
thanks for your help! where can i find help setting up youtube api?

shivasiddharth, 2020-04-22:
I have provided the steps that you need to follow in the README

shivasiddharth, 2020-07-28:
Closing this issue for inactivity

closed 2020-07-28 by shivasiddharth