shivasiddharth/GassistPi #1004: Importerror

justinw78, 2020-04-26:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 2a02082
Serial		: 00000000b1a45bd1
Model		: Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
at starting manually i got:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 37, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 15, in <module>
    import _snowboydetect
ImportError: dynamic module does not define init function (init_snowboydetect)
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
/No

### 4. Is the assistant service restarting automatically?
/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-04-26:
Copy files from src/resources/Buster-wrapper/ into the src directory and then try again.

justinw78, 2020-04-26:
Thnx for this idea, but unfortunately it didnt work:(

shivasiddharth, 2020-04-26:
Post the command that you used to manually start the assistant along with the entire terminal contents.

shivasiddharth, 2020-04-26:
Also I would like to know if you have made any changes anywhere to the scripts.

justinw78, 2020-04-27:
This morning i did a clean start. i got the following errors.

Building wheels for collected packages: pyaudio, spotipy, googletrans
  Running setup.py bdist_wheel for pyaudio ... error
  Complete output from command /home/pi/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-icklbzhe/pyaudio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-n267bi32 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for pyaudio
  Running setup.py clean for pyaudio
  Running setup.py bdist_wheel for spotipy ... error
  Complete output from command /home/pi/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-mrhdncxq/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-zcgbs0oa --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for spotipy
  Running setup.py clean for spotipy
  Running setup.py bdist_wheel for googletrans ... error
  Complete output from command /home/pi/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-wszoz9h0/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-7fhksk7j --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for googletrans
  Running setup.py clean for googletrans
Failed to build pyaudio spotipy googletrans
Installing collected packages: pyaudio, chardet, urllib3, idna, certifi, requests, aftership, feedparser, kodi-json, pyasn1, rsa, httplib2, six, pyasn1-modules, oauth2client, pycryptodomex, gpsoauth, python-dateutil, decorator, validictory, mutagen, proboscis, lxml, soupsieve, beautifulsoup4, MechanicalSoup, protobuf, appdirs, gmusicapi, click, gTTS-token, bs4, gTTS, youtube-dl, ifaddr, zeroconf, casttube, PyChromecast, pafy, psutil, cachetools, google-auth, pytz, googleapis-common-protos, google-api-core, google-auth-httplib2, uritemplate, google-api-python-client, python-magic, websocket-client, pushbullet.py, python-vlc, pyyaml, spidev, colorzero, gpiozero, numpy, paho-mqtt, google-cloud-texttospeech, google-cloud-speech, adafruit-io, pycparser, cffi, asn1crypto, cryptography, mock, spotipy, googletrans
  Running setup.py install for pyaudio ... done

--------------------------------------------

pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible.

------------------------------------------

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id hassio-home-xxx  --device_model_id hassio-home-xxx-hassio-googlehome-xxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 18, in <module>
    import faulthandler
ImportError: No module named faulthandler   (solved with pip install faulthandler)
-----------------
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id hassio-home-xxx  --device_model_id hassio-home-xxxx-hassio-googlehome-xxxxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ImportError: No module named kodijson    (solved with pip install kodi-json)

------------------------
 /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id hassio-home-xxxxxx  --device_model_id hassio-home-xxxxxxhassio-googlehome-xxxxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 29, in <module>
    import pathlib2 as pathlib
ImportError: No module named pathlib2 (solved with pip install pathlib2)
----------------------------------

/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id hassio-home-xxxxxx  --device_model_id hassio-home-xxxxxxassio-googlehome-xxxxxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 33, in <module>
    import psutil
ImportError: No module named psutil (solved with pip install psutil)
------------------------------------------
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id hassio-home-xxxx --device_model_id hassio-home-xxxxassio-googlehome-xxxxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 37, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 15, in <module>
    import _snowboydetect
ImportError: dynamic module does not define init function (init_snowboydetect)

i hope it will help to find a solution
And thnx for your time:)

justin

shivasiddharth, 2020-04-27:
Is this a Hassio image ?

justinw78, 2020-04-27:
No its Raspbian Buster.  hassio is just in the name

shivasiddharth, 2020-04-27:
What you posted is so fragmented. 

> pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible.

is part of the installation and right after that, you have posted the output of the manual start. I need to look at the entire installation part. Coz, so many pip packages have been skipped.

shivasiddharth, 2020-04-27:
Is it Raspbian Lite or Full Desktop image ?

justinw78, 2020-04-27:
full desktop image

justinw78, 2020-04-27:
I just had the errors posted.  the other parts were installed corrected

shivasiddharth, 2020-04-27:
Also post the outputs of:
`pip freeze`

and

```
source env/bin/activate
pip3 freeze
```

justinw78, 2020-04-27:
pip freeze:
arandr==0.1.9
asn1crypto==0.24.0
automationhat==0.2.0
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2018.8.24
chardet==3.0.4
Click==7.0
colorama==0.3.7
colorzero==1.1
configparser==3.5.0b2
cookies==2.2.1
cryptography==2.6.1
drumhat==0.1.0
entrypoints==0.3
enum34==1.1.6
envirophat==1.0.0
ExplorerHAT==0.4.2
faulthandler==3.2
Flask==1.0.2
fourletterphat==0.1.0
funcsigs==1.0.2
gpiozero==1.5.1
idna==2.6
ipaddress==1.0.17
itsdangerous==0.24
Jinja2==2.10
keyring==17.1.1
keyrings.alt==3.1.1
kodi-json==1.0.0
MarkupSafe==1.1.0
microdotphat==0.2.1
mock==2.0.0
mote==0.0.4
motephat==0.0.3
numpy==1.16.2
oauthlib==2.1.0
olefile==0.46
pantilthat==0.0.7
pathlib2==2.3.5
pbr==4.2.0
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
piglow==1.2.5
pigpio==1.44
Pillow==5.4.1
psutil==5.7.0
PyAudio==0.2.11
pycairo==1.16.2
pycrypto==2.6.1
pygame==1.9.4.post1
PyGObject==3.30.4
pyinotify==0.9.6
PyJWT==1.7.0
pyOpenSSL==19.0.0
pyserial==3.4
pyxdg==0.25
rainbowhat==0.1.0
requests==2.21.0
requests-oauthlib==1.0.0
responses==0.9.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
scandir==1.10.0
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-hat==2.2.0
simplejson==3.16.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.4
touchphat==0.0.1
twython==3.7.0
unicornhathd==0.0.4
urllib3==1.24.1
virtualenv==15.1.0
Werkzeug==0.14.1

justinw78, 2020-04-27:
(env ) pip3 freeze:

adafruit-io==2.1
aftership==1.0.post2
appdirs==1.4.3
asn1crypto==1.3.0
beautifulsoup4==4.9.0
bs4==0.0.1
cachetools==4.1.0
casttube==0.2.1
certifi==2020.4.5.1
cffi==1.14.0
chardet==3.0.4
click==6.7
colorzero==1.1
cryptography==2.4.2
decorator==4.4.2
enum34==1.1.10
feedparser==5.2.1
futures==3.1.1
gmusicapi==13.0.0
google-api-core==1.17.0
google-api-python-client==1.8.2
google-assistant-grpc==0.2.1
google-assistant-library==1.1.0
google-assistant-sdk==0.6.0
google-auth==1.14.1
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.4.1
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.51.0
googletrans==2.3.0
gpiozero==1.5.1
gpsoauth==0.4.1
grpcio==1.28.1
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.17.3
idna==2.9
ifaddr==0.1.6
kodi-json==1.0.0
lxml==4.5.0
MechanicalSoup==0.12.0
mock==3.0.5
mutagen==1.44.0
numpy==1.18.3
oauth2client==4.1.3
oauthlib==3.1.0
pafy==0.5.5
paho-mqtt==1.4.0
pathlib2==2.3.5
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.11.3
psutil==5.7.0
pushbullet.py==0.11.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
PyAudio==0.2.11
PyChromecast==5.0.0
pycparser==2.20
pycryptodomex==3.7.2
pyOpenSSL==19.1.0
python-dateutil==2.8.1
python-magic==0.4.15
python-vlc==3.0.7110
pytz==2019.3
PyYAML==3.13
requests==2.23.0
requests-oauthlib==1.3.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.14.0
sounddevice==0.3.15
soupsieve==2.0
spidev==3.4
spotipy==2.12.0
tenacity==4.12.0
uritemplate==3.0.1
urllib3==1.25.9
validictory==1.1.2
websocket-client==0.57.0
youtube-dl==2020.3.24
zeroconf==0.26.0

shivasiddharth, 2020-04-27:
Try this: 
```
source env/bin/activate
python /home/pi/GassistPi/src/main.py --project_id "YOUR PROJECT ID" --device_model_id "YOUR MODELID"
```

shivasiddharth, 2020-04-27:
Did you do a `git clone` or `sudo git clone` ?

justinw78, 2020-04-27:
(env) pi@raspberrypi:~ $  source env/bin/activate
(env) pi@raspberrypi:~ $ python /home/pi/GassistPi/src/main.py --project_id hassio-home-xxx--device_model_id hassio-home-xxxhassio-googlehome-xxx
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 37, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 15, in <module>
    import _snowboydetect
ImportError: dynamic module does not define init function (init_snowboydetect)

used: git clone https://github.com/shivasiddharth/GassistPi

shivasiddharth, 2020-04-27:
Open a new terminal and run the following code on after another.

```
sudo mkdir -p programs/libraries/ && cd programs/libraries
sudo git clone https://github.com/swig/swig.git
cd /home/pi/programs/libraries/swig/
sudo ./autogen.sh
sudo ./configure
sudo make
sudo make install
cd ~/programs
sudo git clone https://github.com/Kitt-AI/snowboy.git
cd /home/pi/programs/snowboy/swig/Python3
sudo make
sudo \cp -f ./_snowboydetect.so /home/pi/GassistPi/src/_snowboydetect.so
sudo \cp -f ./snowboydetect.py /home/pi/GassistPi/src/snowboydetect.py
cd /home/pi/
```

After this try running the assistant manually.

shivasiddharth, 2020-04-27:
What is you audio setup ?

justinw78, 2020-04-27:
output from raspberry pi jack plug 
input from respeaker 4 seedstudio

shivasiddharth, 2020-04-27:
Just as i suspected. Did you use Seedstudio's Respeaker installation steps ?

justinw78, 2020-04-27:
Yes i did. http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

And tested it and all works

shivasiddharth, 2020-04-27:
Thats the issue here. Their snowboy wrappers are old. **Do not use their official setup process.** That is why I have provided my own instructions in the README.

justinw78, 2020-04-27:
Thnx. first i tried your setup got the same errors. Then i had a clean install from raspian stretch and got all the same errors. Now i have raspbian stretch and first installed google home with your link and then the microphon and jack output. Now it worked. 

Thnx for your help:)

greetings

closed 2020-04-27 by justinw78