shivasiddharth/GassistPi #1005: Play YouTube video command HTTP 403 error

SethThMc, 2020-04-28:
IMPORTANT NOTICE
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED

Board and OS details:
Open a terminal and run

cat /proc/cpuinfo  
cat /etc/os-release  

Paste your Board and OS details here.

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 10000000c06ca088
Model		: Raspberry Pi 4 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspbian
[... 183 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/__init__.py", line 42, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2020-04-28 12:02:06,834 : INFO : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyCffx73HtNbdkouBbRGjFNfbXLosXj6gXc
2020-04-28 12:02:07,198 : INFO : URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=&part=id%2Csnippet&key=AIzaSyCffx73HtNbdkouBbRGjFNfbXLosXj6gXc&alt=json

shivasiddharth, 2020-04-28:
1. Youtube feature does not play video its for audio only. 
2.  The links are valid only for short time, they automatically expire after sometime and hence you will get this 403 error.
3. Your syntax for youtube command is wrong.

SethThMc, 2020-04-28:
1. Yeah, sorry, I was trying to say "play the audio from the youtube video", I'm aware it doesn't play video.

2. When I say, " Play ... from YouTube", it immediately results in the 403 error. Just 2 days ago, this command was working, so I don't understand why the links would start expiring faster than they were two days ago.

3. I tried saying "Play ... from YouTube" like it says in the README file, but it comes up with the same error.

Is there any other information I can give you to help me? I appreciate all your help on this project.

SethThMc, 2020-04-28:
Well it seems like I can play videos I haven't previously played with no issue, but if I've played the song in the past, I seem to have this error 9/10 times, with some exceptions. I can't find what the problem is, and it is so inconsistent, so I will close this issue.

SethThMc, 2020-04-28:
Welp, I'm still having issues. I can barely play anything from YouTube now, except for 2 songs. Any help or insights would be appreciated.

shivasiddharth, 2020-04-29:
Its an upstream issue with youtube-dl https://github.com/ytdl-org/youtube-dl/issues/23638 Even they do not have a solution yet. Some videos get played and some dont. I have no control over it.

SethThMc, 2020-04-29:
Okay thanks for your help.

closed 2020-04-29 by SethThMc