shivasiddharth/GassistPi #1007: unrecognized arguments:

joebiwankenobi, 2020-05-12:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 00000000317e1490
Model           : Raspberry Pi 3 Model B Rev 1.2
  

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
[... 259 lines cut ...]
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
usage: main.py [-h] [--device-model-id DEVICE_MODEL_ID]
               [--project-id PROJECT_ID] [--nickname NICKNAME]
               [--device-config DEVICE_CONFIG_FILE]
               [--credentials OAUTH2_CREDENTIALS_FILE] [--query QUERY] [-v]
main.py: error: unrecognized arguments: pidashboard-277015-pidashboard-1re6zm

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
no file found

shivasiddharth, 2020-05-12:
Did you edit or modify the service file ? Your manual start command is wrong. There should be a space between 'pidashboard-277015' and --device_model_id.

joebiwankenobi, 2020-05-12:
Good Evening

Many thanks for getting back to me so quickly earlier..

I have corrected  the typo in the command and re-ran ...  no idea how I missed that, I checked over and over ☹

I have Hit another error unfortunately..  Just in the process of googling it LOL
Forgive my low knowledge on this.. i’m learning, I have a degree of Linux Skills and Wintel etc.. but this sort of coding is new to me.

pi@PiDashboard:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'pidashboard-277015' --device_model_id 'pidashboard-277015-pidashboard-1re6zm'
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-05-12 21:05:59,141 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
[... cut ...]
  File "/home/pi/env/lib/python3.7/site-packages/google/oauth2/_client.py", line 124, in _token_endpoint_request
    _handle_error_response(response_body)
  File "/home/pi/env/lib/python3.7/site-packages/google/oauth2/_client.py", line 60, in _handle_error_response
    raise exceptions.RefreshError(error_details, response_body)
google.auth.exceptions.RefreshError: ('deleted_client: The OAuth client was deleted.', '{\n  "error": "deleted_client",\n  "error_description": "The OAuth client was deleted."\n}')

Many Thanks Again for your time

Kind Regards

Joe

shivasiddharth, 2020-05-13:
Looks like the client that you created in Google Actions Console is no longer there or not working. Create a new device in the actions console and get a new model id. After that download the new credentials file and re-authenticate using:

```
source env/bin/activate
google-oauthlib-tool --scope https://www.googleapis.com/auth/assistant-sdk-prototype \
	    --scope https://www.googleapis.com/auth/gcm \
	    --save --headless --client-secrets "path to your credentials file"
```

joebiwankenobi, 2020-05-13:
Good Morning

I have re-created the model registration as described and I can now confirm that the assistant is responding to voice and giving feedback – YAY!

Just one last niggle now – running through the Headless start – I’m getting this error:

pi@PiDashboard:~ $ sudo systemctl start gassistpi.service
Failed to start gassistpi.service: Unit gassistpi.service has a bad unit file setting.
See system logs and 'systemctl status gassistpi.service' for details.

Obviously I have ran this too many time over the last couple of days trying to make this work.

Just need to figure out how to clean that up now

Many thanks for your Help

Kind Regards Joe

joebiwankenobi, 2020-05-13:
Hi again shivasiddharth<mailto:notifications@github.com>

Sorry to be a pain  but I could use your advice here.

I’m still getting errors when trying to get this to auto start...  I think it may be best if I |Format this SD card and start from scratch.. if I do this.. will it just be a case of running through the install as per your guide or will this do something to the Tokens and cause me more issues?

Also – I’m looking to get this integrated visually into MagicMiror2.. would you have any recommendations in order to do this or should I use “MMM-GoogleAssistant”?

Many thanks in advance

Joe

shivasiddharth, 2020-05-13:
I dont have visual integration setup yet. Feel free to try this https://github.com/bugsounet/MMM-AssistantMk2 . Let me know if are proceeding with this project or giving up. If you are proceeding further we can start troubleshooting the autostart issue.

joebiwankenobi, 2020-05-13:
I’m not giving up
Just wondered if starting from scratch would have fixed the AutoStart issue.

I’m wanting to get thie “MagicMirror” on the wall in the kitchen as I have old bits lying around that I can use.

Ideally visual animations for Gassistpi on MM would be fantastic but for the time being i’d just like to get the assistant to AutoStart along with MM & then I can backup the SD card and tinker with the integration at a later date

I will have a look at the suggested module this evening once I have finished work

Regards

Joe

shivasiddharth, 2020-05-13:
Can you post the output of 
`sudo systemctl status gassistpi.service`
Can you also attach the gassistpi.service file that you have in the /home/pi/GassistPi/systemd folder?

joebiwankenobi, 2020-05-13:
pi@PiDashboard:~ $ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Wed 2020-05-13 13:09:09 BST; 2h 6min ago
  Process: 1799 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id PiDashAssi
Main PID: 1799 (code=killed, signal=TERM)

May 13 13:09:00 PiDashboard systemd[1]: Started Google Assistant-Voice.
May 13 13:09:07 PiDashboard python[1799]: /home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255:
May 13 13:09:07 PiDashboard python[1799]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
May 13 13:09:07 PiDashboard python[1799]: 2020-05-13 13:09:07,889 - gmusicapi.Mobileclient1 (shared:176) [WARNING
May 13 13:09:09 PiDashboard python[1799]: [011c2048] vlcpulse audio output error: PulseAudio server connection fa
May 13 13:09:09 PiDashboard python[1799]: [01217960] vlcpulse audio output error: PulseAudio server connection fa
May 13 13:09:09 PiDashboard systemd[1]: Stopping Google Assistant-Voice...
May 13 13:09:09 PiDashboard systemd[1]: gassistpi.service: Main process exited, code=killed, status=15/TERM
May 13 13:09:09 PiDashboard systemd[1]: gassistpi.service: Succeeded.
May 13 13:09:09 PiDashboard systemd[1]: Stopped Google Assistant-Voice.

& the .service:

[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/__USER__/env
Environment=PATH=/home/__USER__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
#ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --project_id 'created-project-i$
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/pushbutton.py --project-id 'PiDashAssist$

WorkingDirectory=/home/__USER__/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=__USER__

[Install]
WantedBy=multi-user.target

It does start – you can hear the chime then “Hi”.. but then a few seconds later it starts again and same happens
Stuck in a boot loop like scenario

I had a quick look earlier on my lunch break at some of the previous reported issues that similar – hence the “#” in from the Main.py line

shivasiddharth, 2020-05-13:
1. We are using main.py for Pi 3 and not the pushbutton file, so you have commented wrong ExecStart.
2. The modifications to the service file will happen automatically if the default installer is used. I have never had a case where the automod did not happen. Just curious if the installer was customized in anyway or if the modified service file was replaced with a new one.

The details of the service file that you have posted is truncated. Can you attach the file here instead of pasting ? If you have issues, the "How to" for attaching is given right here in the issue window.

joebiwankenobi, 2020-05-13:
Attached is the output of the “systemctl status” & the .service file itself – I have now commented out the PushButton and restored the main.py as explained.

Then > sudo systemctl disable gassistpi.service

Then ran through the other commands on your guide:

  1.  Open the service files in the /GassistPi/systemd/ directory and verify your project and model ids and save the file.
  2.  Change directory

cd /home/${USER}

  1.  Make the service installer executable

sudo chmod +x ./GassistPi/scripts/service-installer.sh

  1.  Run the service installer

sudo ./GassistPi/scripts/service-installer.sh

  1.  Enable the service

sudo systemctl enable gassistpi.service

  1.  Start the service

sudo systemctl start gassistpi.service

service starts but as before – you get the Chime followed by “hi” ... OK_Google voice does not work at this point and then we hear the chime followed by “Hi” again

I haven’t modified or changed any of your scripts .. i’m just wondering if these last few days of tinkering has broken something somewhere maybe

Many thanks for your patience with me here – as previously mentioned... i’m new to this and learning

joebiwankenobi, 2020-05-13:
Rebooted the Pi and the service now auto-starts

You can activate google and ask questions but it appears that the service is in some form of boot loop...  every 20 seconds or so you get the Chime followed by “Hi”... service works for a bit then it restarts as above

shivasiddharth, 2020-05-13:
I dont see the service file. Unfortunately it has not been attached.

shivasiddharth, 2020-05-13:
What is your audio setup ?

joebiwankenobi, 2020-05-13:
how about now?

   Active: active (running) since Wed 2020-05-13 15:39:12 BST; 24s ago
 Main PID: 4023 (python)
    Tasks: 24 (limit: 2077)
   Memory: 67.5M
   CGroup: /system.slice/gassistpi.service
           └─4023 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id PiDashAssistant --device

May 13 15:39:21 PiDashboard python[4023]: jack server is not running or cannot be started
May 13 15:39:21 PiDashboard python[4023]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipp
May 13 15:39:21 PiDashboard python[4023]: /home/pi/env/lib/python3.7/site-packages/google/assistant/library/assis
May 13 15:39:21 PiDashboard python[4023]:   warnings.warn('Google Assistant Library for Python is deprecated', De
May 13 15:39:22 PiDashboard python[4023]: device_model_id: pidashassistant-pidashassistant-pphzni
May 13 15:39:22 PiDashboard python[4023]: device_id: E768B685B59D2C8F9241DF672835EF05
May 13 15:39:22 PiDashboard python[4023]: ON_MUTED_CHANGED:
May 13 15:39:22 PiDashboard python[4023]:   {"is_muted": false}
May 13 15:39:31 PiDashboard python[4023]: ON_MEDIA_STATE_IDLE
May 13 15:39:31 PiDashboard python[4023]: ON_START_FINISHED

pi@PiDashboard:~ $ sudo systemctl stop gassistpi.service

pi@PiDashboard:~/GassistPi/systemd $ cat gassistpi.service

ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --project_id 'PiDashAssistant' --device_model_id 'pidashassistant-pidashassistant-pphzni'
#ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/pushbutton.py --project-id 'PiDashAssistant'  --device-model-id 'pidashassistant-pidashassistant-pphzni'

joebiwankenobi, 2020-05-13:
Audio setup is:

PS3 USB Camera for MIC
3.5mm Jack for audio output.

I did run the audio setup script and I also have this file in ~/home/pi/.asoundrc

pi@PiDashboard:~ $ cat .asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

On the back of this.. I did have a lot of Audio issues to start with, did some googling etc...
In the end it turned out that I was using a mini USB Mic I bought a couple of years back and it was no longer working – then I found the PS3 cam was a great alternative... sound started to work after that

shivasiddharth, 2020-05-13:
Everything is fine. I would suggest you to stop the service, manually start the assistant and play around with it and once it crashes, post the terminal content here. That will give me some idea.

joebiwankenobi, 2020-05-13:
Ok will do

I have already stopped the service and manually started it... so far it works like a charm this way

I will perform some stress tests and see if I can get it to spit out some errors.

Just curious though – if I did format the SD and start again... would that affect the tokens, ID’s etc on my google Action / Cloud and render those profiles useless or would it work using the same .json file etc.. on a fresh install ?
Just sometimes it’s easier to flatten and rebuild than investigate

shivasiddharth, 2020-05-13:
Re-installing will have no impact on the credentials. You can use the same credentials files. But, you will have to re-authenticate that will happen as a part of installation.

joebiwankenobi, 2020-05-13:
Ok thank you

That wont be an issue to re-authorise... I was just wondering if it will break the project on the cloud as the details have already been downloaded and used..

I think i’ll do this tomorrow... format and start again, just to make sure none of the other things I have been playing with are affecting this...  apart from the autostart being in a loop.. the manual process works perfectly... it’s been running for a while now and I have been getting it to play youtube and answer questions without fault.

joebiwankenobi, 2020-05-13:
Ok.. so.

I erased the SD Card and started again, it went very smoothly this time until after I tried to start it.

pi@PiDashboard:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'pidashassistant' --device_model_id 'pidashassistant-pidashassistant-pphzni'
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-05-13 18:58:17,359 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[022b4c10] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[022c4248] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
[... cut ...]
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/queue.py", line 179 in get
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
 File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
  File "/home/pi/GassistPi/src/main.py", line 1013 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted
pi@PiDashboard:~ $

Damn it!

I’m searching google now

shivasiddharth, 2020-05-13:
Have you changed the OS ? Is it the same Buster or Stretch ? Can you check the if the audio config files are same. Remove pulseaudio using `sudo apt-get purge pulseaudio` and then try again.

joebiwankenobi, 2020-05-13:
Same OS Image from before.. BUSTER

.asoundrc looks just like the one I sent earlier

Removed puleaudio with suggested command

Same issue on manual restart

shivasiddharth, 2020-05-14:
1. You need to restart after removing pulseaudio.
2. Try following this:
   ```
  sudo apt-get purge portaudio19-dev
  sudo reboot
  wget http://www.portaudio.com/archives/pa_stable_v190600_20161030.tgz
  tar zxf pa_stable_v190600_20161030.tgz
  pushd portaudio
  ./configure --without-jack
  make
  sudo make install
  sudo reboot
```

NOTE: see if the **make** command spits out errors, if it does, then use `sudo make`

This should fix the error.

joebiwankenobi, 2020-05-14:
Good morning to you Sir

Thank you – I have ran the suggested commands, the service works perfectly under the manual start command.

Autostart does work now but seems to restart itself every 30 seconds or so..
I asked what the weather was and got a response but half way through google telling what the weather is doing.. it just went quiet then a few seconds later the start-up chime was heard followed by Hi.
It has done this a few times in the last 5 minutes ... just seems to keep restarting the service.

But Manual start does what it is supposed to ?

shivasiddharth, 2020-05-14:
Is the restart due to specific commands ? Example, does it restart every time when you ask for weather info or sometimes it restarts and sometimes it does not ? Try setting custom wakeword to "Disabled" in the config file to see if it has any role to play.

joebiwankenobi, 2020-05-14:
Just seems to be Random really.. when it starts initially it seems to start about 3 times then it settles for a minute or so.. but it does restart when you ask it commands too.. sometimes it finishes the voice feedback – others it just restarts again half way through.

I have disabled the custom word in the config file... no change though.

Manual command still works perfectly

shivasiddharth, 2020-05-14:
Unfortunately, there is nothing much i can do without knowing as to what is making it to restart. All that i can suggest you to do is try switching to the last stable version of Stretch. Stretch does not have audio issues and it will work out of the box. If you are not bound to Buster, try using Stretch. Just remember to sudo apt-get purge pulseaudio after the installation completes. Here is the link to the downloads section https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/

joebiwankenobi, 2020-05-14:
Understood... thank you for your help anyway.. it has been very much appreciated.
I will give the Stretch a go at a later date I think.. for now I will just manually start it and leave the cmd terminal open in the background

Interestingly though – after removing  pulseaudio on Buster... even the manual start fails, re-installing pulseaudio makes it work again.

Kind Regards

shivasiddharth, 2020-05-14:
Next time the service file crashes, use sudo journalctl -u gassistpi.service and post the output here.

shivasiddharth, 2020-05-26:
Solutions are being discussed in #1014 . Closing this. Please follow #1014

closed 2020-05-26 by shivasiddharth