shivasiddharth/GassistPi #1008: After doing installing the audio config files and rebooting my audio greyed out and I have no audio

alborzy81, 2020-05-15:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
``` Raspberry pi 4
Rasbian buster

### Describe the bug:
``` So since I am using a USB mic and headphones I did the USB MIC and Jack audio file commands
sudo chmod +x ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
speaker-test  
after that I restarted the Pi and my audio greyed out meaning I had no audio

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
yes
### 2. Is the issue related to assistant installation ?
Yes/No
no
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
yes
### 4. Is the assistant service restarting automatically?
Yes/No
no
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
``` EVERYTHING WORKS BUT WHEN I REBOOT I DONT HAVE AUDIO

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

alborzy81, 2020-05-15:
Once I reboot the pi i have no audio

alborzy81, 2020-05-15:
someone please help

shivasiddharth, 2020-05-15:
Just because the icon is greyed out, it does not mean that you don't have audio. Try the speaker-test it will work.

alborzy81, 2020-05-15:
I did and there was no audio

alborzy81, 2020-05-15:
please help there is no audio I can’t even control audio I did the speaker test I heard nothing I tried to listen to a mp3 file nothing I switched headphones and it still won’t work

shivasiddharth, 2020-05-15:
Stop panicking. Open a terminal and share the outputs of the following.

```
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
```

alborzy81, 2020-05-15:
I will try thanks sid I will tell you if it works

alborzy81, 2020-05-16:
I formated the sd and started from scratch and the icon greyed out I did the speaker test and it worked thank you

alborzy81, 2020-05-19:
asoundrc 
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
asound.conf
ppcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug

alborzy81, 2020-05-19:
now what do I do also so I did the speaker test and it worked but after a few days it didnt

alborzy81, 2020-05-19:
i fixed it

closed 2020-05-19 by alborzy81