shivasiddharth/GassistPi #1009: gassistpi not working after adding 3 custom wakewords

alborzy81, 2020-05-20:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

alborzy81, 2020-05-20:
after adding 2 more custom wakewords to the config.yaml file gassistpi is not making the boot up sound and is not responding to any wakeword even ok google and ok google is enabled I have no idea how this happened maybe I accidently deleted something in the main.py IDK

shivasiddharth, 2020-05-20:
1.  You have not provided any detail at all.
2.  You should have provided the modified config.yaml file showing what you modified.
3.  The issue template is asking you to manually start the assistant and post the contents. You should have provided atleast that.

Issue template has been put in place for speeding up the troubleshooting process.

Closing this issue for not filling in the issue template. Feel free to reopen after you edit and provide all the required details.

closed 2020-05-20 by shivasiddharth

alborzy81, 2020-05-20:
Ok