shivasiddharth/GassistPi #1010: google assistant not working after adding custom wakeword

alborzy81, 2020-05-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
RPI 4 4gb
rasbian buster

### Describe the bug:
```After adding a custom wakeword and adding Jarvis as a wakeword the assistant no longer responds nor makes the booting music
maybe I deleted something in my main.py or my config.yaml

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
yes
### 2. Is the issue related to assistant installation ?
Yes/No
no
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
it did
### 4. Is the assistant service restarting automatically?
Yes/No
no
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'replace this with the project id '--device_model_id 'replace this with the model id'

```

### Paste the contents of the terminal below
```/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-05-21 00:27:34,904 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 960, in <module>
    main()
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 895, in invoke
[... 517 lines cut ...]
stopping vlc
^CStopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

alborzy81, 2020-05-21:
also here is my main.py

alborzy81, 2020-05-21:
and here is my config.yaml
#Config file for declaring custom actions'parameters. The parameters have been listed here in alphabetical order (A-Z).

#Adafruit_IO. Sending commands from Genuine Google Home/Assistant Devices to control Domoticz Tasmota etc.
ADAFRUIT_IO:
  ADAFRUIT_IO_CONTROL: 'Disabled'
  ADAFRUIT_IO_USERNAME: 'ENTER YOUR ADAFRUIT_IO USERNAME HERE'
  ADAFRUIT_IO_KEY: 'ENTER YOUR ADAFRUIT_IO KEY HERE'
  FEEDNAME: 'ENTER YOUR ADAFRUIT_IO FEEDNAME THAT YOU WANT TO MONITOR HERE'

#Custom conversation. Add your queries and expected assistant response. Do not add symbols or special charecters.
Conversation:
  Conversation_Control: 'Enabled'
  question:
    1:
      - 'Who are you'
    2:
      - 'Do you have a family'
  answer:
    1:
      - 'I am not the one you expect to be'
      - 'Who do you think I am'
      - 'What bothers you'
    2:
      - 'Nope i am forever alone'
      - 'The whole world is my family'

#Send sms with clickatell
Clickatell:
  Clickatell_Control: 'Disabled'
  Clickatell_API: 'ENTER_YOUR_CLICKATELL_API'
  Name:
    - 'Bob'
    - 'Adam'
  Reciever:
    - '+48123456789'
    - '+46123456789'

# Deezer user credentials
Deezer:
  Deezer_Control: 'Enabled'
  User_id: '2247762504'

# DIY-HUE declarations
DIYHUE:
  DIYHUE_Control: 'Enabled'

# Domoticz declarations
Domoticz:
  # Set this to 'Enabled' to control devices registered to Domoticz server.
  Domoticz_Control: 'Disabled'
  Server_IP:
    # If you are running the server on the Pi itself, then change the ip to 'localhost'
    - '192.168.0.10'
  Server_port:
    - '443'
  Devices:
    #The number of names and number of ids should be equal
    #List the device names that you want to control
    Name:
      - 'Lamp1'
      - 'Lamp2'
      - 'Lamp3'
    #Assign the device ids in the same order as the devices listed above
    Id:
      - '1'
      - '2'
      - '3'

#ESP control declarations. Follow a similar pattern to add your own list.
[... 180 lines cut ...]

#Text to speech conversation
TextToSpeech:
  # Set the choice to:
  # 1. "gTTS" for quotaless text to speech conversion
  # 2. "Google Cloud" for Google Cloud text to sppech conversion. Free limit 4 Million Characters/month
  Choice: 'gTTS'
  Google_Cloud_TTS_Credentials_Path: "ENTER THE PATH TO YOUR TTS CREDENTIALS FILE HERE"
  #Voice for the "SAY" function or text to speech. Set it to 'Female' or 'Male' depending on your preference.
  #Note: This does not change Google's voice. this is only for the speech generated within custom actions.
  Voice_Gender: 'Female'

#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'
    - '/home/pi/GassistPi/src/resources/friday.pmdl'
    - '/home/pi/GassistPi/src/resources/jarvis.umdl'

YouTube:
  YouTube_Control: 'Enabled'

alborzy81, 2020-05-21:
btw I made pmdl called friday and added it to resources

shivasiddharth, 2020-05-21:
Remove the jarvis.pmdl model it has an issue. It is read as two hotwords. Otherwise change it to https://github.com/shivasiddharth/GassistPi/blob/b323258b5daaf578f25db2d9929c0f4b26e8be23/src/main.py#L190 
to
`self.sensitivity = [0.5]*5`

Whenever you change the the hotword models you need to alter this line as well.

shivasiddharth, 2020-05-21:
Check the way you have wired the pushbutton if you have added that.

shivasiddharth, 2020-05-21:
Duplicate of #455

closed 2020-05-21 by shivasiddharth

alborzy81, 2020-05-21:
what do you mean by dublicate of 455 do I have to duplicate anything

alborzy81, 2020-05-21:
never mind thank you

alborzy81, 2020-05-21:
ok after doing what you said I manually started the assistant and I got this
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-05-21 01:33:13,447 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
2020-05-21 01:33:14,053 : ERROR : number of hotwords in decoder_model (2) and sensitivity (5) does not match
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 100, in __init__
    "(%d) does not match" % (self.num_hotwords, len(sensitivity))
AssertionError: number of hotwords in decoder_model (2) and sensitivity (5) does not match

alborzy81, 2020-05-21:
here is my new main.py
#!/usr/bin/env python

# Copyright (C) 2017 Google Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from __future__ import print_function
import faulthandler
faulthandler.enable()
from kodijson import Kodi, PLAYER_VIDEO
try:
    import RPi.GPIO as GPIO
except Exception as e:
    if str(e) == 'No module named \'RPi\'':
        GPIO = None
import argparse
import json
import os.path
import pathlib2 as pathlib
import os
import subprocess
import re
import psutil
import logging
import time
import random
import snowboydecoder
import sys
import signal
import requests
import io
import google.oauth2.credentials
from google.assistant.library import Assistant
from google.assistant.library.event import EventType
from google.assistant.library.file_helpers import existing_file
from google.assistant.library.device_helpers import register_device
from google.cloud import speech
from google.cloud.speech import enums
from google.cloud.speech import types
import paho.mqtt.client as mqtt
from actions import say
from actions import trans
from actions import Action
from actions import YouTube_No_Autoplay
from actions import YouTube_Autoplay
from actions import stop
from actions import radio
from actions import ESP
from actions import track
from actions import feed
from actions import kodiactions
from actions import mutevolstatus
from actions import gmusicselect
from actions import refreshlists
from actions import chromecast_play_video
from actions import chromecast_control
from actions import kickstarter_tracker
from actions import getrecipe
from actions import hue_control
[... 922 lines cut ...]
                    register_device(args.project_id, credentials,
                                    device_model_id, device_id, args.nickname)
                    pathlib.Path(os.path.dirname(args.device_config)).mkdir(
                        exist_ok=True)
                    with open(args.device_config, 'w') as f:
                        json.dump({
                            'last_device_id': device_id,
                            'model_id': device_model_id,
                        }, f)
                else:
                    print(WARNING_NOT_REGISTERED)

            for event in events:
                if event.type == EventType.ON_START_FINISHED and args.query:
                    assistant.send_text_query(args.query)
                self.process_event(event)

        if custom_wakeword:
            self.detector.terminate()

if __name__ == '__main__':
    try:
        Myassistant().main()
    except Exception as error:
        logging.exception(error)

alborzy81, 2020-05-21:
the wakeword I want to use is friday and now ok google wont work

shivasiddharth, 2020-05-21:
Please go through my comments carefully:

> Whenever you change the the hotword models you need to alter this line as well.

I had mentioned that you need to modify the the number of sensitivities depending on the number of models. Previously you had 4 models including jarvis which is taken as 2 mdodels hence you had to set it to 5. Now you have 2 models so the sentivities should be 2 and not 5.

shivasiddharth, 2020-05-21:
If you are not using jarvis, just leave it as it was there originally https://github.com/shivasiddharth/GassistPi/blob/b323258b5daaf578f25db2d9929c0f4b26e8be23/src/main.py#L190