shivasiddharth/GassistPi #1012: Servo not moving with assistant

alborzy81, 2020-05-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```RPI 4
     BUSTER

### Describe the bug:
``` After connecting a servo to pin 27 and the other wires to power and ground and saying ok google trigger servo 90. the assistant says moving servo to 90 degrees but the servo doesn’t work

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
yes
### 2. Is the issue related to assistant installation ?
Yes/No
no
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
yes
### 4. Is the assistant service restarting automatically?
Yes/No
no
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2020-05-22:
There is hardly any detail in the issue again. Picture of the wiring ? Model of the servo ?

alborzy81, 2020-05-22:
The sensor pin is on 27 the power pin is on the 5v pin and the ground pin
is on ground

shivasiddharth, 2020-05-22:
Have you connected it to Board pin 27 or BCM pin 27 ? Did you use a logic level shifter ? Does the connection work normally using some other script ?

alborzy81, 2020-05-22:
to RPI board pin 27

alborzy81, 2020-05-22:
what is the difference between board and bcm

alborzy81, 2020-05-22:
it is on id/sd pin on the RPI

alborzy81, 2020-05-22:
and no logic level shifter

shivasiddharth, 2020-05-22:
https://iot4beginners.com/difference-between-bcm-and-board-pin-numbering-in-raspberry-pi/  

I have clearly mentioned that the pin numbers used are BCM pin numbers https://github.com/shivasiddharth/GassistPi#list-of-gpios-used-in-raspberry-pi-boards

alborzy81, 2020-05-22:
I am using pin 27 in this diagram

[image]

alborzy81, 2020-05-22:
for some reason its not working

shivasiddharth, 2020-05-22:
You dont seem to pay heed to what i am saying. Board Pin 27 is not the same as BCM 27.  I have clearly mentioned in the README the pins used and have clearly mentioned that it is GPIO 27. Check for GPIO 27 in the pinout that you have posted.

alborzy81, 2020-05-22:
OHHH okay thank I will try it

alborzy81, 2020-05-22:
THANK YOU IT WORKED BLESS YOU

alborzy81, 2020-05-23:
how can I change the praise Trigger Servo 90 to for example trigger faceplate open
and trigger servo 180 to trigger faceplate closed

shivasiddharth, 2020-05-23:
Your issue has been resolved. What you have posted now is query/help request. I will help you here this time, but you are requested to use gitter for any such posts in the future (I have mentioned that clearly in README).

Here in this part https://github.com/shivasiddharth/GassistPi/blob/b323258b5daaf578f25db2d9929c0f4b26e8be23/src/actions.py#L1731

Change
```
if 'servo' in phrase:
        for s in re.findall(r'\b\d+\b', phrase):
            SetAngle(int(s))
        if 'zero' in phrase:
            SetAngle(0) 
```
to 
```
if 'servo' in phrase:
        for s in re.findall(r'\b\d+\b', phrase):
            SetAngle(int(s))
        if 'zero' in phrase:
            SetAngle(0)
if 'faceplate open' in phrase:
       SetAngle(90)
if 'faceplate close' in phrase:
       SetAngle(180)
```

shivasiddharth, 2020-05-23:
Manually make the changes in the script. Copy pasting will raise indentation issues.

closed 2020-05-23 by shivasiddharth

alborzy81, 2020-05-23:
tgabks

alborzy81, 2020-05-23:
Bad news after I did this the google assistant isn't working anymore

alborzy81, 2020-05-23:
here is the actions.py
#!/usr/bin/env python

#This is different from AIY Kit's actions
#Copying and Pasting AIY Kit's actions commands will not work

from kodijson import Kodi, PLAYER_VIDEO
from googleapiclient.discovery import build
from googleapiclient.errors import HttpError
from oauth2client.tools import argparser
from spotipy.oauth2 import SpotifyClientCredentials
import spotipy.util as util
import spotipy.oauth2 as oauth2
from pushbullet import Pushbullet
from mediaplayer import api
from youtube_search_engine import google_cloud_api_key
from googletrans import Translator
from youtube_search_engine import youtube_search
from youtube_search_engine import youtube_stream_link
from google.cloud import texttospeech
from gtts import gTTS
import requests
import mediaplayer
import os
import os.path
try:
    import RPi.GPIO as GPIO
except Exception as e:
    if str(e) == 'No module named \'RPi\'':
        GPIO = None
import time
import re
import subprocess
import aftership
import feedparser
import json
import urllib.request
import pafy
import pychromecast
import spotipy
import pprint
import yaml

ROOT_PATH = os.path.realpath(os.path.join(__file__, '..', '..'))
USER_PATH = os.path.realpath(os.path.join(__file__, '..', '..','..'))

with open('{}/src/config.yaml'.format(ROOT_PATH),'r', encoding='utf8') as conf:
    configuration = yaml.load(conf)

with open('{}/src/lang.yaml'.format(ROOT_PATH),'r', encoding='utf8') as lang:
    langlist = yaml.load(lang)

TTSChoice=''
if configuration['TextToSpeech']['Choice']=="Google Cloud":
    if not os.environ.get("GOOGLE_APPLICATION_CREDENTIALS", ""):
        if configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']!="ENTER THE PATH TO YOUR TTS CREDENTIALS FILE HERE":
            os.environ['GOOGLE_APPLICATION_CREDENTIALS'] = configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']
            TTSChoice='GoogleCloud'
            # Instantiates a client
            client = texttospeech.TextToSpeechClient()
        else:
            print("Set the path to your Google cloud text to speech credentials in the config.yaml file. Using gTTS for now.....")
            TTSChoice='GTTS'
    else:
        TTSChoice='GoogleCloud'
        # Instantiates a client
        client = texttospeech.TextToSpeechClient()
else:
    TTSChoice='GTTS'

[... 1617 lines cut ...]
        os.system("sudo shutdown -h now")
        #subprocess.call(["shutdown -h now"], shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    if 'servo' in phrase:
    for s in re.findall(r'\b\d+\b', phrase):
          SetAngle(int(s))
    if 'zero' in phrase:
        SetAngle(0)
if 'faceplate open' in phrase:
       SetAngle(90)
if 'faceplate close' in phrase:
       SetAngle(180)
       
    else:
        if GPIOcontrol:
            for num, name in enumerate(var):
                if name.lower() in phrase:
                    pinout=gpio[num]
                    if custom_action_keyword['Dict']['On'] in phrase:
                        GPIO.output(pinout, 1)
                        say("Turning On " + name)
                    elif custom_action_keyword['Dict']['Off'] in phrase:
                        GPIO.output(pinout, 0)
                        say("Turning Off " + name)
        else:
            say("GPIO controls, is not supported for your device.")

shivasiddharth, 2020-05-23:
Manually start the assistant that will tell you what the fault is. Most likely, the lines that you entered would not have been formatted properly. 
Do you see how the formatting has been messed  up when codes are pasted here? You should attach the codes and not paste them.

alborzy81, 2020-05-23:
how do I attach them

shivasiddharth, 2020-05-23:
As you are typing the comments do you see "Attach files by dragging & dropping..." below in the box ?

alborzy81, 2020-05-23:
yes

alborzy81, 2020-05-23:
btw here is the terminal
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 1732
    for s in re.findall(r'\b\d+\b', phrase):
      ^
IndentationError: expected an indented block

alborzy81, 2020-05-23:
it wont let me attach it it is not supported

alborzy81, 2020-05-23:
what do I do now

shivasiddharth, 2020-05-23:
Check your gitter msg