shivasiddharth/GassistPi #1014: Frequent crash/restarting with:  Assertion `ret == self->nfds' failed

aarmstrong78, 2020-05-25:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    

Hardware	: BCM2835
Revision	: c03111
Serial		: 100000003e6677d9
Model		: Raspberry Pi 4 Model B Rev 1.1

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

### Describe the bug:
```
The assistant runs in service mode but frequently and randomly restarts.

Restarts sometimes occur without any interaction from the user, but more frequently when playing an audio stream or just asking the assistant a question 

The logs (from: journalctl -u gassistpi.service -b) show the following error at every crash:

python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
Nothing in the terminal, restarting as a service.
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

I haven't been able to get any audio from the assistant when starting it manually, so the following is before and after the crash from the logs in:  journalctl -u gassistpi.service -b

### Paste the command below, that crashed the assistant
```
May 25 07:14:37 raspberrypi python[596]: https://r4---sn-8vq54voxo1-53az.googlevideo.com/videoplayback?expire=1590407936&ei=oF7LXt6eBMSC8QOIkIuYAw&ip=2407%3A7000%3A880a%3Add00%3A4c92%3A1441%3Ac6b6%3Ab0ed&id=o-ANIux6k4FmIsHU6oqdmiPnrrAFGKxqnrpdrHA
May 25 07:14:37 raspberrypi python[596]: [a669e1a8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused

```

### Paste the contents of the terminal below
```
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x677ff460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

aarmstrong78, 2020-05-25:
Posting this here to help others who might have the same error.
Solution provided from @shivasiddharth via gitter (https://gitter.im/GassistPi/Lobby/):

Run in terminal:

sudo apt-get install pulseaudio
sudo apt-get purge portaudio19-dev
 sudo reboot
 wget http://www.portaudio.com/archives/pa_stable_v190600_20161030.tgz
 tar zxf pa_stable_v190600_20161030.tgz
 pushd portaudio
 ./configure --without-jack
 make
 sudo make install
 sudo reboot

aarmstrong78, 2020-05-25:
Spoke too soon.  The error is still there.  Most recent crash log from `journalctl -u gassistpi.service -b`

`May 25 23:20:13 raspberrypi python[4373]: ON_RECOGNIZING_SPEECH_FINISHED:
May 25 23:20:13 raspberrypi python[4373]:   {"text": "what is the time"}
May 25 23:20:13 raspberrypi python[4373]: ON_RESPONDING_STARTED:
May 25 23:20:13 raspberrypi python[4373]:   {"is_error_response": false}
May 25 23:20:15 raspberrypi python[4373]: ON_RESPONDING_FINISHED
May 25 23:20:15 raspberrypi python[4373]: ON_CONVERSATION_TURN_FINISHED:
May 25 23:20:15 raspberrypi python[4373]:   {"with_follow_on_turn": true}
May 25 23:20:15 raspberrypi python[4373]: ON_CONVERSATION_TURN_STARTED
May 25 23:20:22 raspberrypi python[4373]: ON_END_OF_UTTERANCE
May 25 23:20:22 raspberrypi python[4373]: ON_END_OF_UTTERANCE
May 25 23:20:22 raspberrypi python[4373]: ON_RECOGNIZING_SPEECH_FINISHED:
May 25 23:20:22 raspberrypi python[4373]:   {"text": ""}
May 25 23:20:22 raspberrypi python[4373]: ON_NO_RESPONSE
May 25 23:20:22 raspberrypi python[4373]: ON_CONVERSATION_TURN_FINISHED:
May 25 23:20:22 raspberrypi python[4373]:   {"with_follow_on_turn": false}
May 25 23:21:48 raspberrypi python[4373]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
May 25 23:21:48 raspberrypi python[4373]: Fatal Python error: Aborted
May 25 23:21:48 raspberrypi python[4373]: Thread 0x9e1ff460 (most recent call first):
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 25 23:21:48 raspberrypi python[4373]: Thread 0x9edff460 (most recent call first):
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 25 23:21:48 raspberrypi python[4373]: Thread 0xa78ff460 (most recent call first):
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 552 in wait
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 1164 in run
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 25 23:21:48 raspberrypi python[4373]: Thread 0xb6feead0 (most recent call first):
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 25 23:21:48 raspberrypi python[4373]:   File "/usr/lib/python3.7/queue.py", line 179 in get
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/GassistPi/src/main.py", line 1013 in main
May 25 23:21:48 raspberrypi python[4373]:   File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
May 25 23:21:49 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
May 25 23:21:49 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
`

aarmstrong78, 2020-05-25:
I'm now trying to implement this suggested fix: https://lists.columbia.edu/pipermail/portaudio/2018-October/001556.html

This requires editing src/hostapi/alsa/pa_linux_alsa.c from pa_stable_v190600_20161030, then rebuilding using:

```
pushd portaudio
./configure --without-jack --libdir=/lib/arm-linux-gnueabihf --includedir=/usr/include
make
sudo make install
sudo reboot
```
Note that   --libdir=/lib/arm-linux-gnueabihf --includedir=/usr/include  is required to overwrite the appropriate files.

shivasiddharth, 2020-05-26:
The problem is with alsa. We need to downgrade the alsa to 1.1.3 Please follow the steps below.

```
wget "ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.3.tar.bz2"
tar -xf alsa-lib-1.1.3.tar.bz2
cd alsa-lib-1.1.3
./configure && make
sudo make install
sudo apt-get update
sudo apt-get update --fix-missing
sudo apt-get install libncursesw5-dev -y
wget ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.3.tar.bz2
tar xvjf alsa-utils-1.1.3.tar.bz2
cd alsa-utils-1.1.3
./configure --disable-alsaconf --disable-bat --disable-xmlto --disable-rst2man --with-curses=ncursesw
make clean
sudo make install
```

aarmstrong78, 2020-05-26:
OK - I need to do a clean OS install again first.  Do you think I should downgrade alsa before running the GassastPi install script, or after?

aarmstrong78, 2020-05-26:
Is it possible to install a package instead?  Like libasound2_1.1.3-5_armhf.deb from http://snapshot.debian.org/package/alsa-lib/1.1.3-5/  

Or will that not work.

shivasiddharth, 2020-05-26:
That might or may not work. Better to be safe, we should compile the library by ourselves. You can downgrade the alsa at any point. Try doing that first on your existing installation to save some time. For all i know, it might work.

aarmstrong78, 2020-05-26:
Nope, crashed 3 times just starting up:

May 26 08:29:49 raspberrypi python[851]: ON_MEDIA_STATE_IDLE
May 26 08:29:53 raspberrypi python[851]: Assistant is listening....
May 26 08:29:53 raspberrypi python[851]: ON_CONVERSATION_TURN_STARTED
May 26 08:29:54 raspberrypi python[851]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
May 26 08:29:54 raspberrypi python[851]: Fatal Python error: Aborted
May 26 08:29:54 raspberrypi python[851]: Thread 0xa00fa460 (most recent call first):
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 08:29:54 raspberrypi python[851]: Thread 0xa28ff460 (most recent call first):
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 08:29:54 raspberrypi python[851]: Thread 0xa77ff460 (most recent call first):
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 552 in wait
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 1164 in run
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 08:29:54 raspberrypi python[851]: Thread 0xb6f6fad0 (most recent call first):
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 26 08:29:54 raspberrypi python[851]:   File "/usr/lib/python3.7/queue.py", line 179 in get
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/GassistPi/src/main.py", line 1013 in main
May 26 08:29:54 raspberrypi python[851]:   File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
May 26 08:29:54 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT

shivasiddharth, 2020-05-26:
Is this starting from scratch with Alsa 1.1.3 or Alsa 1.1.3 with your existing image ?

Can you post the output of `aplay --version`

aarmstrong78, 2020-05-26:
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>

From a fresh install of Rasbian Lite.  Downgraded alsa first, then followed the install process from the readme.

shivasiddharth, 2020-05-26:
Did you compile the portaudio also ? Just like how its mentioned here https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633440404

aarmstrong78, 2020-05-26:
Nope, I'll try that now.

aarmstrong78, 2020-05-26:
Have done that, testing now.
This was the config summary:

Configuration summary:

  Target ...................... armv7l-unknown-linux-gnueabihf
  C++ bindings ................ no
  Debug output ................ no

  ALSA ........................ yes
  ASIHPI ...................... no

  OSS ......................... yes
  JACK ........................ no

I'm guessing that because c++ bindings was 'no' it didn't replace libportaudiocpp.so.0.0.12 ?

```
-rw-r--r-- 1 pi   pi   819738 May 26 08:59 /usr/lib/arm-linux-gnueabihf/libportaudio.a.bad
lrwxrwxrwx 1 root root     25 Dec 25  2016 /usr/lib/arm-linux-gnueabihf/libportaudiocpp.so.0 -> libportaudiocpp.so.0.0.12
-rw-r--r-- 1 root root  38344 Dec 25  2016 /usr/lib/arm-linux-gnueabihf/libportaudiocpp.so.0.0.12
lrwxrwxrwx 1 root root     21 Dec 25  2016 /usr/lib/arm-linux-gnueabihf/libportaudio.so.2 -> libportaudio.so.2.0.0
-rw-r--r-- 1 root root 166052 Dec 25  2016 /usr/lib/arm-linux-gnueabihf/libportaudio.so.2.0.0
-rw-r--r-- 1 pi   pi   627364 May 26 08:59 /usr/lib/arm-linux-gnueabihf/libportaudio.so.2.0.0.bad

```

Also, after install it says:
On some systems (e.g. Linux) you should run 'ldconfig' now
to make the shared object available.  You may also need to
modify your LD_LIBRARY_PATH environment variable to include
the directory /lib/arm-linux-gnueabihf

aarmstrong78, 2020-05-26:
Just crashed:

May 26 09:56:20 raspberrypi python[586]: Playing MPEG stream 1 of 1: female-say.mp3 ...
May 26 09:56:20 raspberrypi python[586]: MPEG 2.0 L III cbr32 24000 mono
May 26 09:56:22 raspberrypi python[586]: [0:02] Decoding of female-say.mp3 finished.
May 26 09:56:23 raspberrypi sudo[774]:       pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/bin/rm /home/pi/.player.json
May 26 09:56:23 raspberrypi sudo[774]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 26 09:56:23 raspberrypi sudo[774]: pam_unix(sudo:session): session closed for user root
May 26 09:56:23 raspberrypi sudo[784]:       pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/bin/rm /home/pi/.trackqueue.json
May 26 09:56:23 raspberrypi sudo[784]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 26 09:56:23 raspberrypi sudo[784]: pam_unix(sudo:session): session closed for user root
May 26 09:56:23 raspberrypi python[586]: [01de0750] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
May 26 09:56:23 raspberrypi python[586]: /home/pi/env/lib/python3.7/site-packages/vlc.py:1825: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0
May 26 09:56:23 raspberrypi python[586]:   if not any(getargspec(callback)[:2]):  # list(...)
May 26 09:56:23 raspberrypi python[586]: ON_CONVERSATION_TURN_FINISHED:
May 26 09:56:23 raspberrypi python[586]:   {"with_follow_on_turn": false}
May 26 09:56:28 raspberrypi python[586]: [9ed4ef70] prefetch stream error: unimplemented query (264) in control
May 26 10:01:08 raspberrypi python[586]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
May 26 10:01:08 raspberrypi python[586]: Fatal Python error: Aborted
May 26 10:01:08 raspberrypi python[586]: Thread 0x9dcfd460 (most recent call first):
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 10:01:08 raspberrypi python[586]: Thread 0xa2b92460 (most recent call first):
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 865 in run
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 10:01:08 raspberrypi python[586]: Thread 0xa796f460 (most recent call first):
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 552 in wait
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 1164 in run
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 26 10:01:08 raspberrypi python[586]: Thread 0xb6f00ad0 (most recent call first):
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 26 10:01:08 raspberrypi python[586]:   File "/usr/lib/python3.7/queue.py", line 179 in get
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/GassistPi/src/main.py", line 1013 in main
May 26 10:01:08 raspberrypi python[586]:   File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
May 26 10:01:08 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT

shivasiddharth, 2020-05-26:
Can you try changing your /etc/asound.conf and ./.asoundrc to this

```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:0,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

shivasiddharth, 2020-05-26:
Is the USB Mic plugged into USB 3.0 or USB 2.0 port ? See if switching the ports make any difference. It could be an issue with WiFi interference sometimes.

aarmstrong78, 2020-05-26:
Before doing the above, the logs are getting filled up with this now:

May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:52:41 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa

and:

May 26 10:52:45 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa
May 26 10:53:18 raspberrypi systemd-journald[122]: Suppressed 385326 messages from gassistpi.service
May 26 10:53:18 raspberrypi python[586]: Expression 'self->capture.ready || self->playback.ready' failed in 'src/hostapi/alsa

shivasiddharth, 2020-05-26:
Did you follow my instructions or the stackoverflow for portaudio installation ?

aarmstrong78, 2020-05-26:
Yours.  I'll try again from a clean install tomorrow, can't get rid of the steam of error messages at the moment :|

shivasiddharth, 2020-05-26:
Try the desktop version.

aarmstrong78, 2020-05-27:
OK so this time I tried (headless, using SSH):
1. Clean Raspbian Lite install

2. Downgrade Alsa (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633795486)

3. sudo apt-get install pulseaudio
4. sudo apt-get purge portaudio19-dev (not required? Not installed so not removed)
5. Build portaudio (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633440404)
6. sudo ldconfig

7. Check mic is plugged into USB2.0

8. Clone project
9. nano GassistPi-system-requirements.txt.  Remove portaudio19-dev and pulseaudio.
10. Follow 'Install audio config files' instructions.  (speaker-test(s) work)
11. Change sound configs (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633925388)
12. Reboot, then try speaker-tests.  Both work

13. Copy client secret over.
14. Run install script and authorise.
15. install service, start, reboot.

Crashed within a few seconds, not getting any chimes/sound.

aarmstrong78, 2020-05-27:
Alsa problems:

May 27 03:52:00 raspberrypi python[807]: ON_RECOGNIZING_SPEECH_FINISHED:
May 27 03:52:00 raspberrypi python[807]:   {"text": "tell me a tell me the weather"}
May 27 03:52:00 raspberrypi python[807]: ON_RESPONDING_STARTED:
May 27 03:52:00 raspberrypi python[807]:   {"is_error_response": false}
May 27 03:52:00 raspberrypi python[807]: ALSA lib pcm_dmix.c:1165:(snd_pcm_dmix_open) unable to open slave
May 27 03:52:00 raspberrypi python[807]: [ERROR:audio_output_stream.cc(72)] Device could not be opened: default
May 27 03:52:00 raspberrypi python[807]: ON_RESPONDING_FINISHED
May 27 03:52:00 raspberrypi python[807]: ON_CONVERSATION_TURN_FINISHED:

aarmstrong78, 2020-05-27:
Reverted alsa config files back to the GassistPi defaults and at least got sound back.
But immediately crashing still:
May 27 04:15:14 raspberrypi python[1777]: Cannot connect to server socket err = No such file or directory
May 27 04:15:14 raspberrypi python[1777]: Cannot connect to server request channel
May 27 04:15:14 raspberrypi python[1777]: jack server is not running or cannot be started
May 27 04:15:14 raspberrypi python[1777]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
May 27 04:15:14 raspberrypi python[1777]: /home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assi
May 27 04:15:14 raspberrypi python[1777]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
May 27 04:15:14 raspberrypi python[1777]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' f
May 27 04:15:14 raspberrypi python[1777]: Fatal Python error: Aborted
May 27 04:15:14 raspberrypi python[1777]: Thread 0xa7a5c460 (most recent call first):
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/threading.py", line 552 in wait
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/threading.py", line 1164 in run
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
May 27 04:15:14 raspberrypi python[1777]: Thread 0xb6fc1ad0 (most recent call first):
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/subprocess.py", line 1476 in _execute_child
May 27 04:15:14 raspberrypi python[1777]:   File "/usr/lib/python3.7/subprocess.py", line 775 in __init__
May 27 04:15:14 raspberrypi python[1777]:   File "/home/pi/GassistPi/src/main.py", line 992 in main
May 27 04:15:14 raspberrypi python[1777]:   File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
May 27 04:15:14 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
May 27 04:15:15 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
M

aarmstrong78, 2020-05-27:
Might try desktop version (just saw note above), or another image like LibreElec.

aarmstrong78, 2020-05-27:
Just noticed that mpg123 has libportaudio2 as a dependency, so when the GassistPi install script ran it installed the package version of portaudio in addition to the one I had built manually.

aarmstrong78, 2020-05-27:
So, currently testing after doing this:

pi@raspberrypi:~ $ sudo apt-get purge libportaudio2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libportaudio2* mpg123*
0 upgraded, 0 newly installed, 2 to remove and 52 not upgraded.
After this operation, 574 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 59772 files and directories currently installed.)
Removing mpg123 (1.25.10-2) ...
Removing libportaudio2:armhf (19.6.0-1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for mime-support (3.62) ...

aarmstrong78, 2020-05-27:
Radio stream still playing without any crashes so far. If it hasn't crashed by morning then I will try:
apt-get install mpg123 libportaudio2-

to get mpg123 back safely.

aarmstrong78, 2020-05-28:
Agent now seems to be running smoothly *fingers crossed*.  Modified install process as follows:

1. Clean Raspbian Lite install

2. Downgrade Alsa (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633795486)

3. sudo apt-get install pulseaudio
4. sudo apt-get purge portaudio19-dev (not required? Not installed so not removed)
5. Build portaudio (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633440404)
6. sudo ldconfig (not sure if required)
7. Build mpg123 (http://www.linuxfromscratch.org/blfs/view/svn/multimedia/mpg123.html) 

8. Check mic is plugged into USB2.0

9. Clone project
10. nano GassistPi-system-requirements.txt.  Remove portaudio19-dev, pulseaudio and mpg123.
11.  nano GassistPi/systemd/gassistpi.service.  Edit [Service] section to include MPG123_MODDIR environment variable:

    [Service]
    Environment=VIRTUAL_ENV=/home/pi/env
    Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
    Environment=MPG123_MODDIR=/home/pi/mpg123-1.25.13/src/libout123/modules/.libs

12. Follow 'Install audio config files' instructions.  (speaker-test(s) work)
13. Change sound configs (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-633925388)
14. Reboot, then try speaker-tests. 

15. Copy client secret over.
16. Run install script and authorise.
17. install service, start, reboot.

closed 2020-05-28 by aarmstrong78

shivasiddharth, 2020-05-28:
Cool. Thats well done. I shall make modifications to the master to make everything automated.

shivasiddharth, 2020-05-28:
Can you get back tomorrow on the status of crashes. If all is well, then I shall translate the fix to the master.

aarmstrong78, 2020-05-28:
Yep, I'll leave it running overnight again.  Currently looks OK with youtube playlists minor fix (see below) and google music get_top_songs() implemented.

I'll also move the mic back to usb 3.0 and activate wifi (currently just ethernet to save setup time) to make sure nothing else basic triggers a crash.

youtube playlists fix (maxResults must be passed to get required max number in playlist):
```
        list_response = youtube.playlistItems().list(
            playlistId=urlid,
            part='contentDetails',
            maxResults=maximum
        ).execute()

```

aarmstrong78, 2020-05-30:
Everything seems fine.  I tried moving mic to usb3.0 and had issues so I moved it back, but not sure if it was something fixable.

JSa1987, 2020-12-26:
Hi all,

I did a clean install yesterday, starting from Raspberry Pi OS Lite 2nd Dec 2020. I'm now having the same issue mentioned in this thread....

Is the fix mentioned above still applicable or was this automatically implemented as mentioned by shivasiddharth (https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-635100194)?

shivasiddharth, 2020-12-26:
@JSa1987 , everything is working fine now out of the box. You are recommended to use the desktop version of the OS. Also, you should not upgrade the system using the initial dialog box or the upgrade command. Everything has been mentioned in the documents.