shivasiddharth/GassistPi #1019: ERROR: pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible

tomscott50, 2020-06-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor : 0
model name : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS : 697.95
Features : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xb76
CPU revision : 7

Hardware : BCM2835
Revision : 9000c1
Serial : 000000009e44d3c1
Model : Raspberry Pi Zero W Rev 1.1

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Been trying to install on a pi zero w and after lots of failed attempts (due to poor wifi, much better after i moved it to right by the router) i've hit another issue and i can't seem to see any other issues that have covered this - ERROR: pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible is the message that I get, have tried twice now and it takes a long time to get to that point each time. Have updated and upgraded and tried rebooting between tries.
Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
Yes
### 2. Is the issue related to assistant installation ?
Yes/No
Yes
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
Enter the your full credential file name including the path and .json extension: /home/mrdj/h6d3ouh9i5sa6n2202idl515uknro0hp.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: mr-dj-278519

Enter the modelid that was generated in the actions console: mr-dj-278519-mr-dj-36wnp7

Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Fetched 15.0 kB in 3s (5,860 B/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.16.1-4).
[... 257 lines cut ...]
### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
File is empty when I use Nano to view contents

shivasiddharth, 2020-06-08:
This has nothing to do with openssl or cryptography. Your path to the credentials is incorrect. Double check the filename and path that you are entering.  You need not use quotes while entering the path. And I am not sure as to why you have a "cd" at the end of the path

tomscott50, 2020-06-08:
the 'cd' at the end of the path is a copy/paste error from when I copied the log out and isn't in the path at the start of the above log. It's not in quotes either?
to check the path of the .json files i typed the following commands to get to the directory and then typed ls to confirm that the file was there, hopefully this is right and will give me the correct path?

 mrdj@MrDJ:/tmp $ cd /home/mrdj
mrdj@MrDJ:~ $ dir
client_secret_560722794581-h6d3ouh9i5sa6n2202idl515uknro0hp.apps.googleusercontent.com.json  GassistPi
env                                                                                          seeed-voicecard
mrdj@MrDJ:~ $ ls
client_secret_560722794581-h6d3ouh9i5sa6n2202idl515uknro0hp.apps.googleusercontent.com.json  GassistPi
env                                                                                          seeed-voicecard
mrdj@MrDJ:~ $

meaning the correct path for installation is /home/mrdj/client_secret...........json?

tomscott50, 2020-06-08:
Ah, I just spotted that somehow I must have missed the start of the .json file from the original start of the script. I'm just restarting it to try again now... But I still had the opensll error, won't that cause a problem?
Thanks for your help and I'll let you know how it goes.

tomscott50, 2020-06-08:
Got it to install at last, but now running into the limitations of the Pi zero and struggling to get the hotword working. I created a snowboy file, scp'd it to the right folder on the zero and it doesn't respond to my voice. It does make a noise though, like a chime as if it has heard and is waiting for a response but doesn't respond.

To be fair I would probably close this, as I think I'm going to buy a normal Pi to run the assistant on, I would have preferred the Zero as it's much smaller but I can make a normal Pi fit I think. If I get any problems with that I'll let you know, thanks again for the help and for this project, just hope I can get it working and fitted into my robot, it should be pretty cool if I can.

shivasiddharth, 2020-06-08:
Pi Zero W has audio rate issues. Can you manually run the assistant and post the terminal contents. Let mw see as to whats going wrong.

Also post outputs of:
```
aplay -l
arecord -l
```

tomscott50, 2020-06-08:
mrdj@MrDJ:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'mr-dj-278519'  --d
evice-model-id 'mr-dj-278519-mr-dj-36wnp7'
/home/mrdj/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/mrdj/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-06-08 15:53:13,384 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/mrdj/.local/share/gmusicapi/mobileclient.cred''
Initializing GPIOs 5 and 6 for assistant activity indication
ERROR (Input():snowboy-io.cc:315) Fail to open input file "/home/pi/GassistPi/src/resources/alexa.umdl"
terminate called after throwing an instance of 'std::runtime_error'
  what():  ERROR (Input():snowboy-io.cc:315) Fail to open input file "/home/pi/GassistPi/src/resources/alexa.umdl"

[stack trace: ]
Backtrace has not been implemented yet.

Aborted

mrdj@MrDJ:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

mrdj@MrDJ:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
mrdj@MrDJ:~ $

shivasiddharth, 2020-06-08:
In the config.yaml, i have set the default paths for "pi" user. As you are not operating it as "pi" you need to change those paths.

How did you install seeed studio's hat my installer or seeed studio's ?

tomscott50, 2020-06-08:
have changed the default paths and found that the file I downloaded from Snowboy ends in .pmdl and your script was asking for .umdl? I've changed the config.yaml to .pmdl and am just trying it now.

Can't honestly remember, but I think I might have done it through seeed studios... I've been playing with it on multiple pis and installed it quite a few times over the last few days, so might well have been seeed studios. Can I just follow your script to reinstall it correctly or is there another way?

shivasiddharth, 2020-06-08:
You need to uninstall the existing hat installation also delete the voicecard folder and then follow the process from README

tomscott50, 2020-06-08:
OK, I'll have a look at that in a bit, I'm assuming its hung at the moment though - been out for an hour and its not changed from where it got to when I left - see below

mrdj@MrDJ:~/GassistPi/src $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'mr-dj-278519'  --device-model-id 'mr-dj-278519-mr-dj-36wnp7'
/home/mrdj/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/mrdj/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-06-08 16:09:09,465 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/mrdj/.local/share/gmusicapi/mobileclient.cred''
Initializing GPIOs 5 and 6 for assistant activity indication

Been sat there for a good hour now...

shivasiddharth, 2020-06-08:
Have you selected pushbutton trigger or custom wakeword trigger ? can you share the output of again.
```
aplay -l
arecord -l
```

tomscott50, 2020-06-08:
Custom wakeword as other than the button on respeaker hat I don't have any buttons to use and i don't know how to set that one up to be fair. I've not uninstalled the seeed studio hat install yet, going to do that now if i can figure out how to. 
I'm afraid this is all pretty new to me as you might have guessed!

shivasiddharth, 2020-06-08:
It can be uninstalled using:
cd ./seeed-voicecard
sudo ./uninstall.sh

shivasiddharth, 2020-06-08:
The voicecard folder cna be removed using:

```
cd /home/${USER}/
sudo rm -rf seeed-voicecard
```

After this perform the installation as per the README guidelines. Reboot the Pi and check the devices 

```
aplay -l 
arecord -l
```

The issue is for some weird reason, the onboard pi audio is also appearing along with the 2 mic HAT which should not happen.

tomscott50, 2020-06-08:
ah ok, so you want me to keep going? no worries if you do, but now realising the limitations of the zero w i'm tempted to install on my other RPI 4b that i've got here for now, then buy a new 4b for this project.

but either way, how long should the uninstaller take? it was running for 25 mins, stuck on 'remove dtbos'. I've just logged out of mrdj user and logged in as root and tried again - still stuck on 'remove dtbos'?

shivasiddharth, 2020-06-08:
It will take quite a while due to the limitations of Pi Zero. If you have a Pi 4 I would suggest you to ditch Pi Zero. It's not worth the hassles. Pi Zero is not for Voice Assistant applications. Cheers.

tomscott50, 2020-06-08:
OK will do, thanks for your help 👍 
Unless I missed it can I suggest that you add a note to your readme as I didn't find it clear that Pi Zero would struggle?

tomscott50, 2020-06-08:
To close this for sure now, I've just installed on my Pi4b and it's working. Was so much quicker and easier!

I've got some other questions on using voice to make a motor run, shall I ask them in the gitter rather than here?

closed 2020-07-28 by shivasiddharth