shivasiddharth/GassistPi #1020: Youtube Music Streaming: No video formats found

SethThMc, 2020-06-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 10000000c06ca088
Model		: Raspberry Pi 4 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[... 75 lines cut ...]
2020-06-08 13:09:15,154 : DEBUG : lang_check: True
2020-06-08 13:09:15,155 : DEBUG : pre_processor_funcs: [<function tone_marks at 0xaf7fca98>, <function end_of_line at 0xaf7fccd8>, <function abbreviations at 0xaf7fcdf8>, <function word_sub at 0xaf80d228>]
2020-06-08 13:09:15,155 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |\\\n|、|¿|：|\\(|;|，|。|:|\\]|¡|،|—|\\)|\\[|…|‥', re.IGNORECASE) from: [<function tone_marks at 0xaf80d348>, <function period_comma at 0xaf80d390>, <function other_punctuation at 0xaf80d3d8>]>
2020-06-08 13:09:15,159 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2020-06-08 13:09:15,194 : DEBUG : http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2020-06-08 13:09:15,201 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-06-08 13:09:15,349 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2020-06-08 13:09:15,431 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2020-06-08 13:09:15,456 : DEBUG : http://translate.google.com:80 "GET //translate/releases/twsfe_w_20200601_RC00/r/js/translate_m.js HTTP/1.1" 200 569074
2020-06-08 13:09:15,570 : DEBUG : langs: {'af': 'Afrikaans', 'sq': 'Albanian', 'ar': 'Arabic', 'hy': 'Armenian', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'hr': 'Croatian', 'cs': 'Czech', 'da': 'Danish', 'nl': 'Dutch', 'en': 'English', 'eo': 'Esperanto', 'et': 'Estonian', 'tl': 'Filipino', 'fi': 'Finnish', 'fr': 'French', 'de': 'German', 'el': 'Greek', 'gu': 'Gujarati', 'hi': 'Hindi', 'hu': 'Hungarian', 'is': 'Icelandic', 'id': 'Indonesian', 'it': 'Italian', 'ja': 'Japanese', 'jw': 'Javanese', 'kn': 'Kannada', 'km': 'Khmer', 'ko': 'Korean', 'la': 'Latin', 'lv': 'Latvian', 'mk': 'Macedonian', 'ml': 'Malayalam', 'mr': 'Marathi', 'my': 'Myanmar (Burmese)', 'ne': 'Nepali', 'no': 'Norwegian', 'pl': 'Polish', 'pt': 'Portuguese', 'ro': 'Romanian', 'ru': 'Russian', 'sr': 'Serbian', 'si': 'Sinhala', 'sk': 'Slovak', 'es': 'Spanish', 'su': 'Sundanese', 'sw': 'Swahili', 'sv': 'Swedish', 'ta': 'Tamil', 'te': 'Telugu', 'th': 'Thai', 'tr': 'Turkish', 'uk': 'Ukrainian', 'ur': 'Urdu', 'vi': 'Vietnamese', 'cy': 'Welsh', 'zh-cn': 'Chinese (Mandarin/China)', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'en-us': 'English (US)', 'en-ca': 'English (Canada)', 'en-uk': 'English (UK)', 'en-gb': 'English (UK)', 'en-au': 'English (Australia)', 'en-gh': 'English (Ghana)', 'en-in': 'English (India)', 'en-ie': 'English (Ireland)', 'en-nz': 'English (New Zealand)', 'en-ng': 'English (Nigeria)', 'en-ph': 'English (Philippines)', 'en-za': 'English (South Africa)', 'en-tz': 'English (Tanzania)', 'fr-ca': 'French (Canada)', 'fr-fr': 'French (France)', 'pt-br': 'Portuguese (Brazil)', 'pt-pt': 'Portuguese (Portugal)', 'es-es': 'Spanish (Spain)', 'es-us': 'Spanish (United States)'}
2020-06-08 13:09:15,571 : DEBUG : pre-processing: <function tone_marks at 0xaf7fca98>
2020-06-08 13:09:15,572 : DEBUG : pre-processing: <function end_of_line at 0xaf7fccd8>
2020-06-08 13:09:15,572 : DEBUG : pre-processing: <function abbreviations at 0xaf7fcdf8>
2020-06-08 13:09:15,574 : DEBUG : pre-processing: <function word_sub at 0xaf80d228>
2020-06-08 13:09:15,575 : DEBUG : text_parts: 1
2020-06-08 13:09:15,578 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-06-08 13:09:15,718 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2020-06-08 13:09:15,744 : DEBUG : payload-0: {'ie': 'UTF-8', 'q': 'Encountered an exception please check the logs.', 'tl': 'en', 'ttsspeed': 1, 'total': 1, 'idx': 0, 'client': 'tw-ob', 'textlen': 47, 'tk': '791696.699289'}
2020-06-08 13:09:15,748 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-06-08 13:09:15,818 : DEBUG : https://translate.google.com:443 "GET /translate_tts?ie=UTF-8&q=Encountered+an+exception+please+check+the+logs.&tl=en&ttsspeed=1&total=1&idx=0&client=tw-ob&textlen=47&tk=791696.699289 HTTP/1.1" 200 13152
2020-06-08 13:09:15,822 : DEBUG : headers-0: {'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Accept-Encoding': 'gzip, deflate', 'Accept': '*/*', 'Connection': 'keep-alive', 'Referer': 'http://translate.google.com/'}
2020-06-08 13:09:15,822 : DEBUG : url-0: https://translate.google.com/translate_tts?ie=UTF-8&q=Encountered+an+exception+please+check+the+logs.&tl=en&ttsspeed=1&total=1&idx=0&client=tw-ob&textlen=47&tk=791696.699289
2020-06-08 13:09:15,822 : DEBUG : status-0: 200
2020-06-08 13:09:15,823 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2020-06-08 13:09:15,823 : DEBUG : Saved to /tmp/female-say.mp3

shivasiddharth, 2020-06-08:
share output of 
```
source env/bin/activate 
pip freeze
```

SethThMc, 2020-06-08:
pi@raspberrypi:~ $ source env/bin/activate 
(env) pi@raspberrypi:~ $ pip freeze
adafruit-io==2.1
aftership==1.0.post2
appdirs==1.4.3
asn1crypto==1.3.0
beautifulsoup4==4.9.0
bs4==0.0.1
cachetools==4.1.0
casttube==0.2.1
certifi==2020.4.5.1
cffi==1.14.0
chardet==3.0.4
click==6.7
colorzero==1.1
cryptography==2.4.2
decorator==4.4.2
enum34==1.1.10
feedparser==5.2.1
futures==3.1.1
gmusicapi==13.0.0
google-api-core==1.17.0
google-api-python-client==1.8.2
google-assistant-grpc==0.2.1
google-assistant-library==1.1.0
google-assistant-sdk==0.6.0
google-auth==1.14.1
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.4.1
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.51.0
googletrans==2.3.0
gpiozero==1.5.1
gpsoauth==0.4.1
grpcio==1.28.1
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.17.3
idna==2.9
ifaddr==0.1.6
kodi-json==1.0.0
lxml==4.5.0
MechanicalSoup==0.12.0
mock==3.0.5
mutagen==1.44.0
numpy==1.18.3
oauth2client==4.1.3
oauthlib==3.1.0
pafy==0.5.5
paho-mqtt==1.4.0
pathlib2==2.3.5
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.11.3
psutil==5.7.0
pushbullet.py==0.11.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
PyAudio==0.2.11
PyChromecast==5.0.0
pycparser==2.20
pycryptodomex==3.7.2
pyOpenSSL==19.1.0
python-dateutil==2.8.1
python-magic==0.4.15
python-vlc==3.0.7110
pytz==2019.3
PyYAML==3.13
requests==2.23.0
requests-oauthlib==1.3.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.14.0
sounddevice==0.3.15
soupsieve==2.0
spidev==3.4
spotipy==2.12.0
tenacity==4.12.0
uritemplate==3.0.1
urllib3==1.25.9
validictory==1.1.2
websocket-client==0.57.0
youtube-dl==2020.3.24
zeroconf==0.26.0

That's interesting, youtube-dl is listed as outdated. Do you know how to fix that?

shivasiddharth, 2020-06-08:
You should be within the environment when you are trying to update packages. Run
source env/bin/activate
pip3 install --upgrade youtube-dl

SethThMc, 2020-06-08:
Thank you so much!

closed 2020-06-08 by SethThMc