shivasiddharth/GassistPi #1027: NameError: name 'GPIO' is not defined

nkchokshi, 2020-07-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt         vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03112
Serial          : 10000000a982f6c6
Model           : Raspberry Pi 4 Model B Rev 1.2

cat /etc/os-release  

```
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
[... 242 lines cut ...]
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
serialized_end=727,
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
options=None, file=DESCRIPTOR),
/home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
from collections import Mapping, Container
/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
if not isinstance(key, collections.Hashable):
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 13.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-07-26 17:19:14,082 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 51, in
from actions import say
File "/home/pi/GassistPi/src/actions.py", line 149, in
if GPIO!=None:
NameError: name 'GPIO' is not defined
```

shivasiddharth, 2020-07-28:
I need to know the branch you are using. Did you make any changes to the project files ?

nkchokshi, 2020-07-28:
> I need to know the branch you are using. Did you make any changes to the project files ?

I am using this same branch GassistPi. which you showed on youtube as well. 
I havent made any changes. I have now installed MagicMirror, but this one i have tried before installing magicmirror. 

Please help. thank you for replay.

shivasiddharth, 2020-07-28:
run the following commands and paste the output here.

```
source env/bin/activate
pip3 freeze
```

Reni0013, 2020-07-28:
Me too facing same amounts of errors while turning on service manually. Automatically service is still not being start

Reni0013, 2020-07-28:
I think i found the cause of all that errors. i manually go into the /home/pi/gassistpi/requriments path and ran ./Gassistpi-pip-req.txt and found that many of pip files were no installed. I'll try to install them manually one by one with the exact same version.

nkchokshi, 2020-07-28:
> run the following commands and paste the output here.
> 
> ```
> source env/bin/activate
> pip3 freeze
> ```

pi@raspberrypi:~ $ pip3 freeze
appdirs==1.4.3
asn1crypto==0.24.0
astroid==2.1.0
asttokens==1.1.13
automationhat==0.2.0
beautifulsoup4==4.7.1
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2018.8.24
chardet==3.0.4
Click==7.0
colorama==0.3.7
colorzero==1.1
cookies==2.2.1
cryptography==2.6.1
cycler==0.10.0
decorator==4.3.0
devscripts==2.19.5+deb10u1
docutils==0.14
drumhat==0.1.0
entrypoints==0.3
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.0.2
fourletterphat==0.1.0
gpg==1.12.0
gpiozero==1.5.1
guizero==0.6.0
html5lib==1.0.1
idna==2.6
ipykernel==4.9.0
ipython==5.8.0
ipython-genutils==0.2.0
isort==4.3.4
itsdangerous==0.24
jedi==0.13.2
Jinja2==2.10
jupyter-client==5.2.3
jupyter-core==4.4.0
keyring==17.1.1
keyrings.alt==3.1.1
kiwisolver==1.0.1
lazy-object-proxy==1.3.1
logilab-common==1.4.2
lxml==4.3.2
MarkupSafe==1.1.0
matplotlib==3.0.2
mccabe==0.6.1
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.3
mypy==0.670
mypy-extensions==0.4.1
nudatus==0.0.4
numpy==1.16.2
oauthlib==2.1.0
olefile==0.46
pantilthat==0.0.7
parso==0.3.1
pexpect==4.6.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
pickleshare==0.7.5
picraft==1.0
piglow==1.2.5
pigpio==1.44
Pillow==5.4.1
prompt-toolkit==1.0.15
psutil==5.5.1
PyAudio==0.2.11
pycairo==1.16.2
pycodestyle==2.4.0
pycrypto==2.6.1
pyflakes==2.0.0
pygame==1.9.4.post1
Pygments==2.3.1
PyGObject==3.30.4
pyinotify==0.9.6
PyJWT==1.7.0
pylint==2.2.2
pyOpenSSL==19.0.0
pyparsing==2.2.0
pyserial==3.4
python-apt==1.8.4.1
python-dateutil==2.7.3
python-debian==0.1.35
python-magic==0.4.16
pyxdg==0.25
pyzmq==17.1.2
qtconsole==4.3.1
rainbowhat==0.1.0
requests==2.21.0
requests-oauthlib==1.0.0
responses==0.9.0
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
semver==2.0.1
Send2Trash==1.5.0
sense-emu==1.1
sense-hat==2.2.0
simplegeneric==0.8.1
simplejson==3.16.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
soupsieve==1.8
spidev==3.4
ssh-import-id==5.7
thonny==3.2.6
tornado==5.1.1
touchphat==0.0.1
traitlets==4.3.2
twython==3.7.0
typed-ast==1.3.1
uflash==1.2.4
unicornhathd==0.0.4
unidiff==0.5.4
urllib3==1.24.1
wcwidth==0.1.7
webencodings==0.5.1
Werkzeug==0.14.1
wrapt==1.10.11
pi@raspberrypi:~ $

shivasiddharth, 2020-07-28:
Can you please follow the instructions. ?
you have not run the first command.

shivasiddharth, 2020-07-28:
> I think i found the cause of all that errors. i manually go into the /home/pi/gassistpi/requriments path and ran ./Gassistpi-pip-req.txt and found that many of pip files were no installed. I'll try to install them manually one by one with the exact same version.

RPi.GPIO is not a part of the requirements file. Its installed separately.  Installation issues are to be reported with the terminal contents. Even after emphasizing so many times the necessity to stick to issue template, nobody bothers to listen.

nkchokshi, 2020-07-28:
> RPi.GPIO is not a part of the requirements file. Its installed separately. Installation issues are to be reported with the terminal contents. Even after emphasizing so many times the necessity to [...]

Hi I am sorry i am not giving you much info as i am not into much coding, so i have followed exact steps as you showed on youtube. I have not defined, or customized any install options to either enable GPIO or not. or have installed separately.  

(env) pi@raspberrypi:~/GassistPi $ source env/bin/activate
(env) pi@raspberrypi:~/GassistPi $ pip3 freeze
pkg-resources==0.0.0
(env) pi@raspberrypi:~/GassistPi $ 

this is what i have received from the command you asked. 
I have formatted Raspberry and installed again, but no luck, same error appearing.

shivasiddharth, 2020-07-28:
pip3 freeze has yeilded nothing. There is no assistant installed either. Did you get an authentication link ? Did you authenticate ? Golden rule of thumb is to follow the README document.

nkchokshi, 2020-07-28:
Sid i have tried and installed it 3 times. i am ending up with same error. I did get auth. code and and after manually adding command to run script i am getting error of GPIO. i have done same way man. If you want i can post a screen shot now on for your reference.

shivasiddharth, 2020-07-28:
RUn the following commands and post the output:

```
cd /home/pi/
ls
cd /home/pi/env/
ls
cd /home/pi/env/lib/python3.7/
ls
```

I really need to see the terminal contents from your installation step.

Are you using the 64 bit version of OS or the usual 32 bit version ?

nkchokshi, 2020-07-29:
> RUn the following commands and post the output:
> 
> ```
> cd /home/pi/
> ls
> cd /home/pi/env/
> ls
> cd /home/pi/env/lib/python3.7/
> ls
> ```
> 
> I really need to see the terminal contents from your installation step.
> 
> Are you using the 64 bit version of OS or the usual 32 bit version ?

pi@raspberrypi:~ $ cd /home/pi/

pi@raspberrypi:~ $ ls
Bookshelf                                                                                    MagicMirror
client_secret_562010175920-96vpo5fmeoa9rhg5meo54unnuormejk7.apps.googleusercontent.com.json  modelid.txt
Desktop                                                                                      Music
Documents                                                                                    Pictures
Downloads                                                                                    Public
env                                                                                          Templates
GassistPi                                                                                    Videos
install.log

pi@raspberrypi:~ $ cd /home/pi/env/
pi@raspberrypi:~/env $ ls
bin  etc  include  lib  LICENSE  pyvenv.cfg  share

pi@raspberrypi:~/env $ cd /home/pi/env/lib/python3.7/
pi@raspberrypi:~/env/lib/python3.7 $ ls
site-packages

pi@raspberrypi:~/env/lib/python3.7 $

I am using Raspberry pi 32 bit normal version. 

https://imgur.com/S9RUiNT
https://imgur.com/YsysunV

nkchokshi, 2020-07-29:
> give me the output of:
> 
> ```
> cd /home/pi/env/lib/python3.7/site-packages/
> ls
> ```

 cd /home/pi/env/lib/python3.7/site-packages/
pi@raspberrypi:~/env/lib/python3.7/site-packages $ ls
Adafruit_IO
adafruit_io-2.1.dist-info
aftership
aftership-1.0.post2.dist-info
apiclient
appdirs-1.4.4.dist-info
appdirs.py
argparse-1.4.0.dist-info
argparse.py
asn1crypto
asn1crypto-1.4.0.dist-info
beautifulsoup4-4.9.1.dist-info
bs4
bs4-0.0.1.dist-info
cachetools
cachetools-4.1.1.dist-info
casttube
casttube-0.2.1.dist-info
certifi
certifi-2020.6.20.dist-info
cffi
cffi-1.14.1.dist-info
_cffi_backend.cpython-37m-arm-linux-gnueabihf.so
chardet
chardet-3.0.4.dist-info
click
click-6.7.dist-info
colorzero
colorzero-1.1.dist-info
Cryptodome
cryptography
cryptography-2.4.2.dist-info
dateutil
decorator-4.4.2.dist-info
decorator.py
easy_install.py
enum
enum34-1.1.10.dist-info
ez_setup.py
feedparser-5.2.1.dist-info
feedparser.py
futures-3.1.1.dist-info
gmusicapi
gmusicapi-13.0.0.dist-info
google
googleapiclient
google_api_core-1.22.0.dist-info
google_api_core-1.22.0-py3.8-nspkg.pth
google_api_python_client-1.10.0.dist-info
googleapis_common_protos-1.52.0.dist-info
googleapis_common_protos-1.52.0-py3.8-nspkg.pth
google_assistant_grpc-0.2.1.dist-info
google_assistant_grpc-0.2.1-py2.7-nspkg.pth
google_assistant_library-1.1.0.dist-info
google_assistant_library-1.1.0-py2.7-nspkg.pth
google_assistant_sdk-0.6.0.dist-info
google_assistant_sdk-0.6.0-py3.5-nspkg.pth
google_auth-1.20.0.dist-info
google_auth-1.20.0-py3.8-nspkg.pth
google_auth_httplib2-0.0.4.dist-info
google_auth_httplib2.py
google_auth_oauthlib
google_auth_oauthlib-0.4.1.dist-info
google_cloud_speech-0.36.0.dist-info
google_cloud_speech-0.36.0-py3.6-nspkg.pth
google_cloud_texttospeech-0.3.0.dist-info
google_cloud_texttospeech-0.3.0-py3.6-nspkg.pth
googlesamples
googletrans
googletrans-2.3.0.dist-info
gpiozero
gpiozero-1.5.1.dist-info
gpiozerocli
gpsoauth
gpsoauth-0.4.1.dist-info
grpc
grpcio-1.30.0.dist-info
gtts
gTTS-2.0.2.dist-info
gtts_token
gTTS_token-1.1.3.dist-info
httplib2
httplib2-0.18.1.dist-info
idna
idna-2.10.dist-info
ifaddr
ifaddr-0.1.7.dist-info
kodijson
kodi_json-1.0.0.dist-info
lxml
lxml-4.5.2.dist-info
magic.py
mechanicalsoup
MechanicalSoup-0.12.0.dist-info
mock
mock-3.0.5.dist-info
mutagen
mutagen-1.45.0.dist-info
numpy
numpy-1.19.1.dist-info
oauth2client
oauth2client-4.1.3.dist-info
oauthlib
oauthlib-3.1.0.dist-info
OpenSSL
pafy
pafy-0.5.5.dist-info
paho
paho_mqtt-1.4.0.dist-info
pathlib2
pathlib2-2.3.5.dist-info
pip
pip-20.2.dist-info
pkg_resources
pkg_resources-0.0.0.dist-info
_portaudio.cpython-37m-arm-linux-gnueabihf.so
proboscis
proboscis-1.2.6.0.dist-info
protobuf-3.12.2.dist-info
protobuf-3.12.2-nspkg.pth
psutil
psutil-5.7.2.dist-info
pushbullet
pushbullet.py-0.11.0.dist-info
pyasn1
pyasn1-0.4.8.dist-info
pyasn1_modules
pyasn1_modules-0.2.8.dist-info
PyAudio-0.2.11.dist-info
pyaudio.py
__pycache__
pychromecast
PyChromecast-7.1.2.dist-info
pycparser
pycparser-2.20.dist-info
pycryptodomex-3.7.2.dist-info
pyOpenSSL-19.1.0.dist-info
python_dateutil-2.8.1.dist-info
python_magic-0.4.18.dist-info
python_vlc-3.0.11115.dist-info
pytz
pytz-2020.1.dist-info
PyYAML-3.13.dist-info
requests
requests-2.24.0.dist-info
requests_oauthlib
requests_oauthlib-1.3.0.dist-info
RPi
RPi.GPIO-0.6.3.dist-info
rsa
rsa-4.6.dist-info
setuptools
setuptools-49.2.0.dist-info
six-1.15.0.dist-info
six.py
sounddevice-0.3.15.dist-info
_sounddevice.py
sounddevice.py
soupsieve
soupsieve-2.0.1.dist-info
spidev-3.5.dist-info
spidev.cpython-37m-arm-linux-gnueabihf.so
spotipy
spotipy-2.13.0.dist-info
tenacity
tenacity-4.12.0.dist-info
uritemplate
uritemplate-3.0.1.dist-info
urllib3
urllib3-1.25.10.dist-info
validictory
validictory-1.1.2.dist-info
vlc.py
websocket
websocket_client-0.57.0.dist-info
wheel
wheel-0.34.2.dist-info
yaml
youtube_dl
youtube_dl-2020.7.28.dist-info
zeroconf
zeroconf-0.28.0.dist-info
pi@raspberrypi:~/env/lib/python3.7/site-packages $

shivasiddharth, 2020-07-29:
Here in this line https://github.com/shivasiddharth/GassistPi/blob/1b9ef1b5dae7ae3b2e3e19ad473de0f625c348e1/src/actions.py#L46  

add the following

`print(GPIO)`

now run the assistant and share the output

nkchokshi, 2020-07-29:
> Here in this line
> 
> https://github.com/shivasiddharth/GassistPi/blob/1b9ef1b5dae7ae3b2e3e19ad473de0f625c348e1/src/actions.py#L46
> 
> add the following
> 
> `print(GPIO)`
> 
> now run the assistant and share the output

INFO:gmusicapi.Mobileclient1:failed to authenticate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 46, in <module>
    print(GPIO)
NameError: name 'GPIO' is not defined

shivasiddharth, 2020-07-29:
Post the contents of your actions.py script.

nkchokshi, 2020-07-29:
#!/usr/bin/env python

#This is different from AIY Kit's actions
#Copying and Pasting AIY Kit's actions commands will not work

from kodijson import Kodi, PLAYER_VIDEO
from googleapiclient.discovery import build
from googleapiclient.errors import HttpError
from oauth2client.tools import argparser
from spotipy.oauth2 import SpotifyClientCredentials
import spotipy.util as util
import spotipy.oauth2 as oauth2
from pushbullet import Pushbullet
from mediaplayer import api
from youtube_search_engine import google_cloud_api_key
from googletrans import Translator
from youtube_search_engine import youtube_search
from youtube_search_engine import youtube_stream_link
from google.cloud import texttospeech
from gtts import gTTS
import requests
import mediaplayer
import os
import os.path
try:
    import RPi.GPIO as GPIO
except Exception as e:
    if str(e) == 'No module named \'RPi\'':
        GPIO = None
import time
import re
import subprocess
import aftership
import feedparser
import json
import urllib.request
import pafy
import pychromecast
import spotipy
import pprint
import yaml

ROOT_PATH = os.path.realpath(os.path.join(__file__, '..', '..'))
USER_PATH = os.path.realpath(os.path.join(__file__, '..', '..','..'))

print(GPIO)
with open('{}/src/config.yaml'.format(ROOT_PATH),'r', encoding='utf8') as conf:
    configuration = yaml.load(conf)

with open('{}/src/lang.yaml'.format(ROOT_PATH),'r', encoding='utf8') as lang:
    langlist = yaml.load(lang)

TTSChoice=''
if configuration['TextToSpeech']['Choice']=="Google Cloud":
    if not os.environ.get("GOOGLE_APPLICATION_CREDENTIALS", ""):
        if configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']!="ENTER THE PATH TO YOUR TTS CREDENTIALS FILE HERE":
            os.environ['GOOGLE_APPLICATION_CREDENTIALS'] = configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']
            TTSChoice='GoogleCloud'
            # Instantiates a client
            client = texttospeech.TextToSpeechClient()
        else:
            print("Set the path to your Google cloud text to speech credentials in the config.yaml file. Using gTTS for now.....")
            TTSChoice='GTTS'
    else:
        TTSChoice='GoogleCloud'
        # Instantiates a client
        client = texttospeech.TextToSpeechClient()
else:
    TTSChoice='GTTS'

[... 1612 lines cut ...]
#GPIO Device Control
def Action(phrase):
    if 'shutdown' in phrase:
        say('Shutting down Raspberry Pi')
        time.sleep(10)
        os.system("sudo shutdown -h now")
        #subprocess.call(["shutdown -h now"], shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    if 'servo' in phrase:
        for s in re.findall(r'\b\d+\b', phrase):
            SetAngle(int(s))
        if 'zero' in phrase:
            SetAngle(0)
    else:
        if GPIOcontrol:
            for num, name in enumerate(var):
                if name.lower() in phrase:
                    pinout=gpio[num]
                    if custom_action_keyword['Dict']['On'] in phrase:
                        GPIO.output(pinout, 1)
                        say("Turning On " + name)
                    elif custom_action_keyword['Dict']['Off'] in phrase:
                        GPIO.output(pinout, 0)
                        say("Turning Off " + name)
        else:
            say("GPIO controls, is not supported for your device.")

shivasiddharth, 2020-07-29:
do:
```
source env/bin/activate
pip3 install RPi.GPIO==0.6.3
```
I dont think I can do anything more.

I asked you for the entire installation terminal contents already a number of times and all you have given me is an image that hardly helps. If you need your issue to be fixed, then you need to provide the information requested.

nkchokshi, 2020-07-29:
Sir i understand you are helping me out, and i really appreciate it. I have re-installed it again. here is the full output of terminal. 
and that pip install RPi.GPIO==0.6.3 already done yesterday, but didnt worked, tried today too but still same error. 

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_562010175920-96vpo5fmeoa9rhg5meo54unnuormejk7.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: assitancemirror

Enter the modelid that was generated in the actions console: assitancemirror-mirrorassistance-d68vsy

Hit:1 http://archive.raspberrypi.org/debian buster InRelease                                       
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.16.1-4).
bison is already the newest version (2:3.3.2.dfsg-1).
build-essential is already the newest version (12.6).
gcc is already the newest version (4:8.3.0-1+rpi2).
git is already the newest version (1:2.20.1-2+deb10u3).
libatlas-base-dev is already the newest version (3.10.3-8+rpi1).
libffi-dev is already the newest version (3.2.1-9).
libjack-jackd2-dev is already the newest version (1.9.12~dfsg-2).
libmpg123-dev is already the newest version (1.25.10-2).
libpcre3 is already the newest version (2:8.39-12).
libpcre3-dev is already the newest version (2:8.39-12).
libsox-fmt-mp3 is already the newest version (14.4.2+git20190427-1).
libxml2-dev is already the newest version (2.9.4+dfsg1-7+b3).
libxslt1-dev is already the newest version (1.1.32-2.2~deb10u1).
mpg123 is already the newest version (1.25.10-2).
mplayer is already the newest version (2:1.3.0-8+b5).
mpv is already the newest version (0.29.1-1).
nmap is already the newest version (7.70+dfsg1-6).
portaudio19-dev is already the newest version (19.6.0-1).
python is already the newest version (2.7.16-1).
python-dev is already the newest version (2.7.16-1).
python-setuptools is already the newest version (40.8.0-1).
python3 is already the newest version (3.7.3-1).
python3-dev is already the newest version (3.7.3-1).
python3-lxml is already the newest version (4.3.2-1).
python3-setuptools is already the newest version (40.8.0-1).
python3-venv is already the newest version (3.7.3-1).
socat is already the newest version (1.7.3.2-2).
sox is already the newest version (14.4.2+git20190427-1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u3+rpt1).
python-pip is already the newest version (18.1-5+rpt1).
python3-pip is already the newest version (18.1-5+rpt1).
vlc is already the newest version (3.0.11-0+deb10u1+rpt1).
Suggested packages:
  pavumeter paman paprefs
The following NEW packages will be installed:
  pulseaudio
0 upgraded, 1 newly installed, 0 to remove and 8 not upgraded.
Need to get 1,036 kB of archives.
After this operation, 5,851 kB of additional disk space will be used.
Get:1 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf pulseaudio armhf 12.2-4+deb10u1 [1,036 kB]
Fetched 1,036 kB in 9s (119 kB/s)    
Selecting previously unselected package pulseaudio.
(Reading database ... 160302 files and directories currently installed.)
Preparing to unpack .../pulseaudio_12.2-4+deb10u1_armhf.deb ...
Unpacking pulseaudio (12.2-4+deb10u1) ...
Setting up pulseaudio (12.2-4+deb10u1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for dbus (1.12.16-1) ...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages (0.2.11)

[... 173 lines cut ...]
Requirement already satisfied: CFFI>=1.0 in ./env/lib/python3.7/site-packages (from sounddevice<0.4,>=0.3.7; extra == "samples"->google-assistant-sdk[samples]==0.6.0) (1.14.1)
Requirement already satisfied: requests>=2.0.0 in ./env/lib/python3.7/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (2.24.0)
Requirement already satisfied: oauthlib>=3.0.0 in ./env/lib/python3.7/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (3.1.0)
Requirement already satisfied: cachetools<5.0,>=2.0.0 in ./env/lib/python3.7/site-packages (from google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (4.1.1)
Requirement already satisfied: rsa<5,>=3.1.4; python_version >= "3.5" in ./env/lib/python3.7/site-packages (from google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (4.6)
Requirement already satisfied: setuptools>=40.3.0 in ./env/lib/python3.7/site-packages (from google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (49.2.0)
Requirement already satisfied: pyasn1-modules>=0.2.1 in ./env/lib/python3.7/site-packages (from google-auth->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples]==0.6.0) (0.2.8)
Requirement already satisfied: asn1crypto>=0.21.0 in ./env/lib/python3.7/site-packages (from cryptography>=1.3.4; extra == "secure"->urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.6.0) (1.4.0)
[... cut ...] %3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fgcm&state=wzH8stukZD9DmqXFfAa7sb00s4TR5n&prompt=consent&access_type=offline
Enter the authorization code: 4/2QHLs4wFlkO60lTqYn6cETK-T5RyX97ZH-wxKLQZLqXAsc01KCTh4dU
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json

Auto modification of the service file is not feasible. Manually check your username, project id and model id.....

Finished installing Google Assistant.......

shivasiddharth, 2020-07-29:
Can you run the assistant now and see. Looks like only now the RPi.GPIO has been installed.

nkchokshi, 2020-07-29:
> Can you run the assistant now and see. Looks like only now the RPi.GPIO has been installed.

Hello i have removed entire GassistPi folder and installed all again and after installation, i started manually. 

 (env) pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'assitancemirror-mirrorassistance-d68vsy' --project_id 'assitancemirror' 

/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_p [... cut ...]
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0182e5f0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01858788] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 149, in <module>
    if GPIO!=None:
NameError: name 'GPIO' is not defined

shivasiddharth, 2020-07-29:
My Pi 3 is working absolutely fine. Need to see if this is an isolated incident with Pi 4. i will leave this issue open

nkchokshi, 2020-07-30:
> My Pi 3 is working absolutely fine. Need to see if this is an isolated incident with Pi 4. i will leave this issue open

Ive formatted again, and installed direct google assistant sdk, and installed google assistant, which started working, but i am now able to find solution of how to integrate it with "Hello google" currently its "click to run"

shivasiddharth, 2020-07-30:
If you are in a hurry and dont want me to investigate the cause of this issue use this branch https://github.com/shivasiddharth/GassistPi/tree/Just-Google-Assistant It has the Ok Google wakeword enabled.

deena1982, 2020-08-01:
Hi, just my 2 cents on this issue, on pi4 , I got the same error.
```
(env) pi@raspberrypi:~ $ python
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/env/lib/python3.7/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
>>> 
```
so I just commented a line in actions.py when importing GPIO and an exception happens and the master branch works fine now.

```
try:
    import RPi.GPIO as GPIO
except Exception as e:
    #if str(e) == 'No module named \'RPi\'':
        GPIO = None
```

nkchokshi, 2020-08-01:
> Hi, just my 2 cents on this issue, on pi4 , I got the same error.
>
> ```
> (env) pi@raspberrypi:~ $ python
> Python 3.7.3 (default, Dec 20 2019, 18:57:59)
> [GCC 8.3.0] on linux
> Type "help", "copyright", [...]

now that i have commented the line, i am getting same error in
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 75, in <module>
    if GPIO!=None:
NameError: name 'GPIO' is not defined

nkchokshi, 2020-08-02:
> Hi, just my 2 cents on this issue, on pi4 , I got the same error.
>
> ```
> (env) pi@raspberrypi:~ $ python
> Python 3.7.3 (default, Dec 20 2019, 18:57:59)
> [GCC 8.3.0] on linux
> Type "help", "copyright", [...]

@shivasiddharth No I'm not in hurry sir, I was just finishing the magic mirror so wanted to add assistant along with it. Anyway it started working after commenting line but a question though. Commenting this, mean that it will also disable GPIO from code? I mean then we will not be able to control gpio through Google assistant? 

Also In your old videos you mentioned to control home lights with Google assistant only with correcting the "kitchen lights", "bedroom light" from main.py arrays. How do I define now? Or now have to do with nodemcu and iffft?

shivasiddharth, 2020-08-02:
The GPIO version that this project uses is currently 0.6.3 which supports boards only till Pi 3B+. If you need GPIO support for Pi 4 you need to update the RPi.GPIO version. Updating might present some errors with the existing code which may have to be troubleshooted.

nkchokshi, 2020-08-02:
Hi i have tried to add LED indicator and tried to safe shutdown with commands but they are not working. The LED indicators are not working and when i tried safe shutdown it returned with error of voice commands not supported. 

I am running currently GPIO version 0.7. 

any way around to the issue? and make it work, at least the indicators?

shivasiddharth, 2020-08-03:
> i tried safe shutdown it returned with error of voice commands not supported.

It is not an error. It is just that you have not adhered to the command syntax. Check the README for the command syntax.

nkchokshi, 2020-08-03:
But it should automatically light up the indication lights when talking and listening? On GPIO 5 & 6? I have manually tried and check if I can on off those leds. Those are working but not working in GassistPi. 

Or do I have to activate them?

shivasiddharth, 2020-08-03:
There are couple of things that could have gone wrong:
1. You could have installed the latest RPi.GPIO outside the environment and not inside the environment. Installation within the environment should be performed after using `source env/bin/activate`.
2. You could have used the Board Pin numbers 5 and 6 instead of BCM GPIO 5 and 6. GassistPi uses BCM numbers.

nkchokshi, 2020-08-03:
> There are couple of things that could have gone wrong:
>
> 1. You could have installed the latest RPi.GPIO outside the environment and not inside the environment. Installation within the environment [...]

Hello sir thank you so much for amazing help, that worked. Actually ive installed RPi outside of env. and that was causing issue. and now the indicator LEDs are working. 

Now final issue that i am facing and its completely out of code i think. But i have connected 4 channels relay with it, connected fan, and 2 lights.  Now issue is all connections are by default gets ON automatically as soon as assistant starts.  As i have added their BCM numbers in Config file. 

Even in environment, when i give command to light "GPIO.output(37,GPIO.HIGH) it actually OFF the switch instead of turning it ON, and it automatically OFFs the light and 2 second later it starts the light.

shivasiddharth, 2020-08-03:
It might be like one of those sainsmart like relay modules which have inverse logic. You need to change the logic in the board if you can or modify the code. Its an issue with your relay module not with the codes.

commit 20229b3e (2020-08-04): Error Fix

1. Fix GPIO errors associated with RPi 4
2. Enable GPIO usage for RPi4

Fixes #1027
closed 2020-08-04 by shivasiddharth