shivasiddharth/GassistPi #1028: Radio support for aac format radios

deena-din, 2020-08-01:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
```
pi@raspberrypi:~/GassistPi/src $ git status
On branch master
Your branch is up to date with 'origin/master'.
```

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03112
Serial		: 1000000015654299
Model		: Raspberry Pi 4 Model B Rev 1.2

```
```
cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:
```
It is a feature request actually to support radio stations which stream aac format audio. like 
https://playerservices.streamtheworld.com/api/livestream-redirect/ARRAHMANLITE_S01.aac

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Sample code working on Pi 4 with python mpd library which play aac and mp3
```
import mpd
station='https://playerservices.streamtheworld.com/api/livestream-redirect/ARRAHMANLITE_S01.aac'
station2='http://playerservices.streamtheworld.com/api/livestream-redirect/SKYRADIO.mp3'

mpdplayer = mpd.MPDClient()
mpdplayer.connect("localhost", 6600)
mpdplayer.clear()
mpdplayer.add(station)
mpdplayer.play()

```

deena-din, 2020-08-01:
changes I have made:
actions.py
```
#!/usr/bin/env python

#This is different from AIY Kit's actions
#Copying and Pasting AIY Kit's actions commands will not work

from kodijson import Kodi, PLAYER_VIDEO
from googleapiclient.discovery import build
from googleapiclient.errors import HttpError
from oauth2client.tools import argparser
from spotipy.oauth2 import SpotifyClientCredentials
import spotipy.util as util
import spotipy.oauth2 as oauth2
from pushbullet import Pushbullet
from mediaplayer import api
from youtube_search_engine import google_cloud_api_key
from googletrans import Translator
from youtube_search_engine import youtube_search
from youtube_search_engine import youtube_stream_link
from google.cloud import texttospeech
from gtts import gTTS
import requests
import mediaplayer
import mpd
import os
import os.path
try:
    import RPi.GPIO as GPIO
except Exception as e:
    #if str(e) == 'No module named \'RPi\'':
        GPIO = None
import time
import re
import subprocess
import aftership
import feedparser
import json
import urllib.request
import pafy
import pychromecast
import spotipy
import pprint
import yaml

ROOT_PATH = os.path.realpath(os.path.join(__file__, '..', '..'))
USER_PATH = os.path.realpath(os.path.join(__file__, '..', '..','..'))

with open('{}/src/config.yaml'.format(ROOT_PATH),'r', encoding='utf8') as conf:
    configuration = yaml.load(conf)

with open('{}/src/lang.yaml'.format(ROOT_PATH),'r', encoding='utf8') as lang:
    langlist = yaml.load(lang)

TTSChoice=''
if configuration['TextToSpeech']['Choice']=="Google Cloud":
    if not os.environ.get("GOOGLE_APPLICATION_CREDENTIALS", ""):
        if configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']!="ENTER THE PATH TO YOUR TTS CREDENTIALS FILE HERE":
            os.environ['GOOGLE_APPLICATION_CREDENTIALS'] = configuration['TextToSpeech']['Google_Cloud_TTS_Credentials_Path']
            TTSChoice='GoogleCloud'
            # Instantiates a client
            client = texttospeech.TextToSpeechClient()
        else:
            print("Set the path to your Google cloud text to speech credentials in the config.yaml file. Using gTTS for now.....")
            TTSChoice='GTTS'
    else:
        TTSChoice='GoogleCloud'
        # Instantiates a client
        client = texttospeech.TextToSpeechClient()
[... 1877 lines cut ...]
        with open('{}/.trackqueue.json'.format(USER_PATH),'r') as input_file:
            tracks= json.load(input_file)
        print(tracks[trackid])
        urlid = youtube_search(tracks[trackid])
        if urlid is not None:
            fullurl = "https://www.youtube.com/watch?v=" + urlid
            audiostream, videostream = youtube_stream_link(fullurl)
            streamurl = audiostream
            self.media_player(streamurl)

    def deezer_player(self,trackid):
        with open('{}/.trackqueue.json'.format(USER_PATH),'r') as input_file:
            tracks= json.load(input_file)
        print(tracks[trackid])
        urlid = youtube_search(tracks[trackid])
        if urlid is not None:
            fullurl = "https://www.youtube.com/watch?v=" + urlid
            audiostream, videostream = youtube_stream_link(fullurl)
            streamurl = audiostream
            self.media_player(streamurl)

    def check_delete(self,file):
        if os.path.isfile(file):
            os.system("sudo rm "+file)
```

deena-din, 2020-08-01:
changes I have made till now
actions.py
```
#import mpd for radio
import mpd
mpdplayer = mpd.MPDClient()

#Radio Station Streaming
def radio(phrase):
    conv = None
    for num, name in reversed(list(enumerate(stnname))):
        if name.lower() in phrase:
            station=stnlink[num]
            conv=stnradio[num]
            print (station)
            break
    if conv is not None:
        say("Tuning into " + conv)
        vlcplayer.media_manager(station,'Radio')
        print("python type of station: {}".format(type(station)))
        #vlcplayer.media_player(station)
        mpdplayer.connect("localhost", 6600)
        mpdplayer.clear()
        mpdplayer.add(station)
        mpdplayer.play()

    else:
        say("Station not found")
```

mediaplayer.py
```
    def stop_vlc(self):
        print('stopping vlc')
        if os.path.isfile("{}/.player.json".format(USER_PATH)):
            with open('{}/.player.json'.format(USER_PATH),'r') as input_file:
                  playerinfo= json.load(input_file)
            musictype=playerinfo[3]
            print ("MUSIC TYPE Is : {}".format(musictype))
            if musictype == "Radio":
                print("Music type is RADIO")
                actions.mpdplayer_stop()
                self.libvlc_player.stop()
            else:
                self.libvlc_player.stop()
        else:
            self.libvlc_player.stop()
```
this code is crashing though, when I start Radio, and it is a work-in-progress and I thought I can share the details here to discuss

shivasiddharth, 2020-08-05:
Does mpd have volume and playback controls ? Sorry I am not familiar with MPD.

deena-din, 2020-08-08:
Hi @shivasiddharth , thanks for the reply. mpd/mpc does have volume and plaback controls.

Actually  I did observe another thing. When I startGassist [master branch] and immediately say play Radio 2 (where I have configured the aac radio), It plays ... 😃 / So when the first command command is Radio , the existing mp3 lib that Gassist[master branch without any changes] is using is able to play aac radios. Thats awesome. It is just when I ask it play aac radio after other voice commands it crashes. I will post the crash logs etc, shortly.

dinesh-kumar-c, 2020-08-08:
when I tried to simulate the crash, aac radio is working fine :) . I don't know how. Thanks for the inputs, if I get crash further I wil capture the stacks and post here.

dinesh-kumar-c, 2020-11-17:
Hi @shivasiddharth  - the crash is back again :) and I managed to capture the traceback

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play radio 3"}

stopping vlc
https://playerservices.streamtheworld.com/api/livestream-redirect/ARRAHMANLITE_S01.aac
Fatal Python error: Segmentation fault

Thread 0x5d9ff460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0xb6fa9ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
```

dinesh-kumar-c, 2020-11-17:
the mp3 Radio is also crashing at the same point.

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Radio 1"}

stopping vlc
http://playerservices.streamtheworld.com/api/livestream-redirect/SKYRADIO.mp3
Fatal Python error: Segmentation fault

Thread 0x6028d460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0xb6fb7ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
```

shivasiddharth, 2020-11-17:
Could be an issue with gtts check the version of gtts you have.

dinesh-kumar-c, 2020-11-18:
thanks very much for the input @shivasiddharth , after upgrading gTTS, the radio is working fine now. I am wondering if I should update all pypi packages. Please advise.

```
    Uninstalling gTTS-2.0.2:
      Successfully uninstalled gTTS-2.0.2
Successfully installed gTTS-2.2.1
```

dinesh-kumar-c, 2020-11-18:
@shivasiddharth , I updated a few packages actually as I am using pi4 - RPi.GPIO and youtube-dl ( looks like the youtube download is breaking for copyrighted songs? ) https://github.com/blackjack4494/yt-dlc/issues/13

dinesh-kumar-c, 2020-11-18:
this issue can be closed and I can open another for the youtube playback.

closed 2020-11-18 by dinesh-kumar-c