shivasiddharth/GassistPi #1031: Feature request for Magic Mirror

nkchokshi, 2020-08-05:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
GassistPi

Hi Sir i want to make request if you can please make module for magic mirror? or your existing "Google Assistant SDK Smart Screen" integration over magic mirror iframe? 

I have now google assistant setup working fine in background but if we can get on screen display like google assistant smart screen on magic mirror, that will be great.

Also is it possible, i have GassistPi full installed, i want to add google assistant smart screen in it, is it possible? then ill port iframe code to magic mirror?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

nkchokshi, 2020-08-05:
Hi Sir yes i have seen the Smart Screen branch, is there any way i can merge that with Main branch of GassistPi? I have already installed main GassistPi with magic mirror. 

Can i just add functions from main.py of Google-Assistant-SDK-Smart-Screen to GassistPI main.py, will it work? 

I will then add Iframe code to magic mirror to reference the browser window.

shivasiddharth, 2020-08-05:
Yes you can merge. If you know how the git works, you can clone my repo and work on that otherwise you need to manually compare the files and make changes.

This issue forum is for real issues and not for doubt clarifications. Gitter is for non-issue stuff. Please stick to the rules.