shivasiddharth/GassistPi #1033: [Headless startup issue] Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setting.)

Reni0013, 2020-08-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
**Master Branch**

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   

**processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 000000003ec8578a
Model		: Raspberry Pi 3 Model B Rev 1.2**

**PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"**

### Describe the bug:
```
**Headless auto start is not working. i tried many troubleshooting but none of them are working. systemctl status shows that unit part of .service part is bad configured**
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
**Yes**

### 2. Is the issue related to assistant installation ?
**No, As i am able to run Google assistance with manual command**

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
**No, This is the problem**

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
sudo systemctl status gassistpi.service  

```

### Paste the contents of the terminal below
```
● gassistpi.service - Google Assistant-Voice
   Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setti
   Active: inactive (dead)
lines 1-3/3 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setting.)
   Active: inactive (dead)

```

solidssss, 2020-08-08:
I had the same issue. I had to reflash rasbpian from scratch and to it all again

Reni0013, 2020-08-08:
> I had the same issue. I had to reflash rasbpian from scratch and to it all again

Have you manage to solve after reflashing raspbian os? Because i did it couple of times but don't know somehow i return to this same error.

shivasiddharth, 2020-08-09:
Attach the contents of gassistpi.service file located inside the systemd folder within the GassistPi project folder directory. Do not change anything post it as such.

Reni0013, 2020-08-10:
Thanks a lot for your response. Here it is. i have go through many searches for this issue and one of the issue which was previously discussed here and it was solved by removing **' '**  sing after **--device_model_id** so i did it and now i am copy pasting same content without correcting it so you have better idea if i am doing anything wrong. 

```
[Unit]
Description=Google Assistant-Voice

[Service]
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id assistantreni-eaglesmarthub-gbvix6
WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

```

shivasiddharth, 2020-08-10:
You need to have those quotes just like how you have it for the project_id. Looks like you have tinkered with the systemd file heavily. Why are pieces of the unit file missing ?

Reni0013, 2020-08-10:
When i boot raspberry pi then some text commands shows up for very few seconds before raspberry pi logo appears telling that unit file is configured bad. so i think removing other content under unit section may solve this problem but i was wrong.

Reni0013, 2020-08-10:
I'll try to put quotes on model id same as project id and revert back with the results. Anything more should i have to change? Should i again correct all changes which i made to the normal one For better troubleshooting?

shivasiddharth, 2020-08-10:
You need to revert back to original systemd file that came with the project.
Add back all the parts that you removed.
Run sudo systemctl disable gassistpi.service
Re-do the headless setup steps.

Reni0013, 2020-08-10:
```
pi@raspberrypi:~ $ sudo systemctl disable gassistpi.service
Removed /etc/systemd/system/multi-user.target.wants/gassistpi.service.
pi@raspberrypi:~ $ cd /home/${USER}   
pi@raspberrypi:~ $ sudo chmod +x ./GassistPi/scripts/service-installer.sh  
pi@raspberrypi:~ $ sudo ./GassistPi/scripts/service-installer.sh    
pi@raspberrypi:~ $ sudo systemctl enable gassistpi.service  
Created symlink /etc/systemd/system/multi-user.target.wants/gassistpi.service → /lib/systemd/system/gassistpi.service.
pi@raspberrypi:~ $ sudo systemctl start gassistpi.service  
Failed to start gassistpi.service: Unit gassistpi.service has a bad unit file setting.
See system logs and 'systemctl status gassistpi.service' for details.
pi@raspberrypi:~ $ systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setti
   Active: inactive (dead)

Aug 10 17:40:32 raspberrypi systemd[1]: gassistpi.service: Service has more than
Aug 10 17:43:39 raspberrypi systemd[1]: gassistpi.service: Service has more than
lines 1-6/6 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setting.)
   Active: inactive (dead)

Aug 10 17:40:32 raspberrypi systemd[1]: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services. Refusing.
Aug 10 17:43:39 raspberrypi systemd[1]: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services. Refusing.

```

Still Getting Same

shivasiddharth, 2020-08-10:
Its not the same. You dont need to have the ExecStart= with the pushbutton script. Remove that line and follow the steps that i mentioned previously again. Also share the service file here.

Reni0013, 2020-08-11:
Service is now working after following your instructions but still GassistPi is not starting (Google voice greeting with "HI" ) and i gave command to just check if GassistPi is working in background without startup sound but it was not running in background also.

**Service File**

[```
Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/__USER__/env
Environment=PATH=/home/__USER__/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id 'assistantreni-eaglesmarthub-gbvix6'

WorkingDirectory=/home/__USER__/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=__USER__

[Install]
WantedBy=multi-user.targetsudo systemctl status gassistpi.service 
```

**Terminal Output of sudo systemctl status gassistpi.service** 

```
pi@raspberrypi:~ $ sudo systemctl status gassistpi.service  
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Tue 2020-08-11 11:58:49 IST; 20s ago
 Main PID: 1027 (python)
    Tasks: 5 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─1027 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Aug 11 11:59:09 raspberrypi python[1027]: Traceback (most recent call last):
Aug 11 11:59:09 raspberrypi python[1027]:   File "/home/pi/GassistPi/src/main.py
Aug 11 11:59:09 raspberrypi python[1027]:     logging.basicConfig(format='%(asct
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__i
Aug 11 11:59:09 raspberrypi python[1027]:     h = FileHandler(filename, mode)
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__i
Aug 11 11:59:09 raspberrypi python[1027]:     StreamHandler.__init__(self, self.
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__i
Aug 11 11:59:09 raspberrypi python[1027]:     return open(self.baseFilename, sel
Aug 11 11:59:09 raspberrypi python[1027]: PermissionError: [Errno 13] Permission
lines 1-18/18 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-08-11 11:58:49 IST; 20s ago
 Main PID: 1027 (python)
    Tasks: 5 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─1027 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistantreni --device_model_id assistantreni-eaglesmarthub-gbvix6

Aug 11 11:59:09 raspberrypi python[1027]: Traceback (most recent call last):
Aug 11 11:59:09 raspberrypi python[1027]:   File "/home/pi/GassistPi/src/main.py", line 118, in <module>
Aug 11 11:59:09 raspberrypi python[1027]:     logging.basicConfig(format='%(asctime)s : %(levelname)s : %(message)s', level=logging.DEBUG , filename='/tmp/GassistPi.log')
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__init__.py", line 1900, in basicConfig
Aug 11 11:59:09 raspberrypi python[1027]:     h = FileHandler(filename, mode)
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__init__.py", line 1092, in __init__
Aug 11 11:59:09 raspberrypi python[1027]:     StreamHandler.__init__(self, self._open())
Aug 11 11:59:09 raspberrypi python[1027]:   File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
Aug 11 11:59:09 raspberrypi python[1027]:     return open(self.baseFilename, self.mode, encoding=self.encoding)
Aug 11 11:59:09 raspberrypi python[1027]: PermissionError: [Errno 13] Permission denied: '/tmp/GassistPi.log'
```

**I have nothing change in Main.py file still i think this error shows log file has some permission issues**

[... 1026 lines cut ...]
1025.                                     device_model_id, device_id, args.nickname)
1026.                     pathlib.Path(os.path.dirname(args.device_config)).mkdir(
1027.                         exist_ok=True)
1028.                     with open(args.device_config, 'w') as f:
1029.                         json.dump({
1030.                             'last_device_id': device_id,
1031.                             'model_id': device_model_id,
1032.                         }, f)
1033.                 else:
1034.                     print(WARNING_NOT_REGISTERED)
1035. 
1036.             for event in events:
1037.                 if event.type == EventType.ON_START_FINISHED and args.query:
1038.                     assistant.send_text_query(args.query)
1039.                 self.process_event(event)
1040. 
1041.         if custom_wakeword:
1042.             self.detector.terminate()
1043. 
1044. 
1045. if __name__ == '__main__':
1046.     try:
1047.         Myassistant().main()
1048.     except Exception as error:
1049.         logging.exception(error)

shivasiddharth, 2020-08-11:
You should change the `__USER__` in the service file to `pi`. Make that change and go through the service installation process again.  I automated all these things in the installation script.

Reni0013, 2020-08-11:
Hi, Thanks for your prompt responses. i made those changes and again go through headless startup as you told and now getting this..

pi@raspberrypi:~ $ sudo systemctl status gassistpi.service  
```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Tue 2020-08-11 13:12:09 IST; 3s ago
 Main PID: 2536 (python)
    Tasks: 1 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─2536 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packag
Aug 11 13:12:11 raspberrypi python[2536]:   options=None, file=DESCRIPTOR),
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packag
Aug 11 13:12:11 raspberrypi python[2536]:   index=0, containing_type=None, field
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packag
Aug 11 13:12:11 raspberrypi python[2536]:   serialized_end=727,
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packag
Aug 11 13:12:11 raspberrypi python[2536]:   options=None, file=DESCRIPTOR),
Aug 11 13:12:12 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packag
Aug 11 13:12:12 raspberrypi python[2536]:   from collections import Mapping, Con
lines 1-18/18 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-08-11 13:12:09 IST; 3s ago
 Main PID: 2536 (python)
    Tasks: 1 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─2536 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistantreni --device_model_id assistantreni-eaglesmarthub-gbvix6

Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:201: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descr
Aug 11 13:12:11 raspberrypi python[2536]:   options=None, file=DESCRIPTOR),
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descr
Aug 11 13:12:11 raspberrypi python[2536]:   index=0, containing_type=None, fields=[]),
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptor
Aug 11 13:12:11 raspberrypi python[2536]:   serialized_end=727,
Aug 11 13:12:11 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descr
Aug 11 13:12:11 raspberrypi python[2536]:   options=None, file=DESCRIPTOR),
Aug 11 13:12:12 raspberrypi python[2536]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 
Aug 11 13:12:12 raspberrypi python[2536]:   from collections import Mapping, Container

```
**These python errors were also come before but GassistPi used to start and work normally even with these errors but now it is not. Now i can not start GassistPi manually with the command.**

**Here what i get when i tried to start  GassistPi with manual command**

```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'assistantreni-eaglesmarthub-gbvix6' --project_id 'assistantreni' 
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
  File "/usr/lib/python3.7/logging/__init__.py", line 1900, in basicConfig
    h = FileHandler(filename, mode)
  File "/usr/lib/python3.7/logging/__init__.py", line 1092, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
PermissionError: [Errno 13] Permission denied: '/tmp/GassistPi.log'
```
Do i need to install GassistPi again ( sudo  ./GassistPi/scripts/gassist-installer.sh ) ?

Reni0013, 2020-08-11:
Sir, Do you need output of **source env/bin/activate & pip3 freeze** ?

shivasiddharth, 2020-08-11:
Run the following and then check if the manual start works.
```
sudo systemctl stop gassistpi.service
sudo rm /tmp/GassistPi.log
```

Reni0013, 2020-08-11:
wow, After running these comand now i am able to start GassistPi manually.

Reni0013, 2020-08-11:
Now can i start systemctl gassistpi.service?

shivasiddharth, 2020-08-11:
Start the service and revert back.

Reni0013, 2020-08-11:
Still not starting after boot.

**sudo systemctl status gassistpi.service**

```
pi@raspberrypi:~ $ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Tue 2020-08-11 22:38:35 IST; 18s ago
 Main PID: 1116 (python)
    Tasks: 1 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─1116 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:!-- begin
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:version: 
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:logging t
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:initializ
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:logged ou
Aug 11 22:38:53 raspberrypi python[1116]: /home/pi/env/lib/python3.7/site-packag
Aug 11 22:38:53 raspberrypi python[1116]:   warnings.warn(_MISSING_FILE_MESSAGE.
Aug 11 22:38:53 raspberrypi python[1116]: 2020-08-11 22:38:53,753 - gmusicapi.Mo
Aug 11 22:38:53 raspberrypi python[1116]: WARNING:gmusicapi.Mobileclient1:could 
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:failed to
lines 1-18/18 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-08-11 22:38:35 IST; 18s ago
 Main PID: 1116 (python)
    Tasks: 1 (limit: 1944)
   CGroup: /system.slice/gassistpi.service
           └─1116 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistantreni --device_model_id assistantreni-eaglesmarthub-gbvix6

Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:version: 13.0.0
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:initialized
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:logged out
Aug 11 22:38:53 raspberrypi python[1116]: /home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
Aug 11 22:38:53 raspberrypi python[1116]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
Aug 11 22:38:53 raspberrypi python[1116]: 2020-08-11 22:38:53,753 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Aug 11 22:38:53 raspberrypi python[1116]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Aug 11 22:38:53 raspberrypi python[1116]: INFO:gmusicapi.Mobileclient1:failed to authenticate
```

shivasiddharth, 2020-08-11:
Post output of `journalctl -u gassistpi.service`

shivasiddharth, 2020-08-11:
Share the contents of the service file that you have used.

Reni0013, 2020-08-12:
Output of **journalctl -u gassistpi.service**

```
pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Thu 2019-02-14 15:41:59 IST, end at Wed 2020-08-12 09:50:09 IST
Aug 11 23:17:07 raspberrypi systemd[1]: Started Google Assistant-Voice.
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   serialized_pb=b'\n\x15google/api/http
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   serialized_end=121,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-package
lines 1-23...skipping...
-- Logs begin at Thu 2019-02-14 15:41:59 IST, end at Wed 2020-08-12 09:50:09 IST. --
Aug 11 23:17:07 raspberrypi systemd[1]: Started Google Assistant-Voice.
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. P
Aug 11 23:17:13 raspberrypi python[481]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. 
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. 
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Pleas
Aug 11 23:17:13 raspberrypi python[481]:   serialized_end=121,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
Aug 11 23:17:13 raspberrypi python[481]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Aug 11 23:17:13 raspberrypi python[481]:   file=DESCRIPTOR,
[... cut ...]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id 'assistantreni-eaglesmarthub-gbvix6'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

shivasiddharth, 2020-08-12:
Start from a fresh image and see.

Reni0013, 2020-08-12:
Fresh image means? I am not getting you. You mean flash new raspdian os? Can you please explain bit more?

shivasiddharth, 2020-08-12:
Starting fresh means, erasing the sd card, writing the os image on the sd card and installing the assistant from scratch again.

Reni0013, 2020-08-12:
Okay sir, I will do that and let you know the results.

Reni0013, 2020-08-12:
**Done with flashing fresh new raspbian os but Still it is not starting automatically but can start manually by command**

**Output of sudo systemctl status gassistpi.service**

```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-08-13 00:35:47 IST; 466ms ago
 Main PID: 2869 (python)
    Tasks: 1 (limit: 2077)
   Memory: 4.3M
   CGroup: /system.slice/gassistpi.service
           └─2869 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistantreni --device_model_id assistantreni-eaglesmarthub-gbvix6

Aug 13 00:35:47 raspberrypi systemd[1]: Started Google Assistant-Voice.
```

**Output of manual command**

```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id 'assistantreni-eaglesmarthub-gbvix6'
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: assistantreni-eaglesmarthub-gbvix6
device_id: 1D2C96FA19B8D65013863C3FE2E8392B

Registering...Done.

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED
```

Manually assistant is working fine but not starting on boot. This is the same that i was facing earlier and then i edit service file very badly  ;-(

Reni0013, 2020-08-12:
@shivasiddharth 

**Here is the gassistpi.service file that i currently have**

```
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id 'assistantreni-eaglesmarthub-gbvix6'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

shivasiddharth, 2020-08-13:
Post output of 
`journalctl -u service-name.service`

aryaputhra, 2020-08-13:
I have the same issue - I think it is something wrong with the gassistpi.service file. I have wiped clean my SD card, did a fresh install of Raspbian OS and going through your github codes, will keep you posted if the exact error pops up... it starts manually okay, but only the headless start fails unfortunately

aryaputhra, 2020-08-13:
okay, after redoing the whole thing - service now auto-starts, but it automatically keeps restarting (i.e. the starting jingles and "hi" keeps on coming, indicating the service has restarted) - how do I mute the jingles and "hi"?

shivasiddharth, 2020-08-13:
@aryaputhra This thread is going nowhere. Instead of looking into the issue that is restarting the service, you want to cut off the jingles ? 

Post the output if https://github.com/shivasiddharth/GassistPi/issues/1033#issuecomment-673209562

Reni0013, 2020-08-13:
> Post output of
> `journalctl -u gassistpi.service`

@shivasiddharth  **Here is the ouput**

```
pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Thu 2020-08-13 12:26:35 IST, end at Thu 2020-08-13 12:28:05 IST
Aug 13 12:26:40 raspberrypi systemd[1]: Started Google Assistant-Voice.
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   serialized_pb=b'\n\x15google/api/http
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   serialized_end=121,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-package
lines 1-23...skipping...
-- Logs begin at Thu 2020-08-13 12:26:35 IST, end at Thu 2020-08-13 12:28:05 IST. --
Aug 13 12:26:40 raspberrypi systemd[1]: Started Google Assistant-Voice.
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going t
Aug 13 12:26:49 raspberrypi python[484]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going 
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going 
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go
Aug 13 12:26:49 raspberrypi python[484]:   serialized_end=121,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
[... cut ...]
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:102: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is goi
Aug 13 12:26:49 raspberrypi python[484]:   file=DESCRIPTOR,
Aug 13 12:26:49 raspberrypi python[484]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:114: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to
Aug 13 12:26:49 raspberrypi python[484]:   serialized_end=144,
lines 1-54
```

Reni0013, 2020-08-13:
> okay, after redoing the whole thing - service now auto-starts, but it automatically keeps restarting (i.e. the starting jingles and "hi" keeps on coming, indicating the service has restarted) - how do I mute the jingles and "hi"?

I done the same before but even after that service was not started automatically. Can you tell us what audio setup you are using? @shivasiddharth  Does audio setup creates any problem for service file?

shivasiddharth, 2020-08-13:
The output of the journal command that you have posted above does not contain all the lines. Allow the gassistpi service to restart as many times as it can and give it some time after the last start attempt and then use the journal command.

Reni0013, 2020-08-14:
Okay, Doing that. Allow me some time.

Reni0013, 2020-08-16:
**That's all what i got**

```
pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Sun 2020-08-16 13:13:51 IST, end at Sun 2020-08-16 13:48:00 IST. --
Aug 16 13:13:51 raspberrypi python[10238]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create u
Aug 16 13:13:51 raspberrypi python[10238]:   options=None, file=DESCRIPTOR),
Aug 16 13:13:52 raspberrypi python[10238]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated
Aug 16 13:13:52 raspberrypi python[10238]:   from collections import Mapping, Container
Aug 16 13:13:57 raspberrypi python[10238]: /home/pi/env/lib/python3.7/site-packages/yaml/constructor.py:126: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, a
Aug 16 13:13:57 raspberrypi python[10238]:   if not isinstance(key, collections.Hashable):
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:version: 13.0.0
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:initialized
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:logged out
Aug 16 13:13:57 raspberrypi python[10238]: /home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
Aug 16 13:13:57 raspberrypi python[10238]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
Aug 16 13:13:57 raspberrypi python[10238]: 2020-08-16 13:13:57,377 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Aug 16 13:13:57 raspberrypi python[10238]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Aug 16 13:13:57 raspberrypi python[10238]: INFO:gmusicapi.Mobileclient1:failed to authenticate
Aug 16 13:13:58 raspberrypi python[10238]: [027dd6d0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Aug 16 13:13:58 raspberrypi python[10238]: [02803bf8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Aug 16 13:13:59 raspberrypi python[10238]: USB-DAC
Aug 16 13:13:59 raspberrypi python[10238]: Initializing GPIOs 5 and 6 for assistant activity indication
Aug 16 13:13:59 raspberrypi python[10238]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Aug 16 13:13:59 raspberrypi python[10238]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Aug 16 13:13:59 raspberrypi python[10238]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
[... cut ...]
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 13:13:59 raspberrypi python[10238]: ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
lines 1-54
```

solidssss, 2020-08-16:
I too got this error now again for the second time on my raspberry pi 3b after disabling autoboot by running stop and disable on systemctl gassistpi.service. When I tried to enable and start them again, I get:
$sudo systemctl start gassistpi.service
`Failed to start gassistpi.service: Unit gassistpi.service has a bad unit file setting.`

Running  **sudo systemctl status gassistpi.service** give me:
```
gassistpi.service - Google Assistant-Voice
   Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setting.)
   Active: inactive (dead)
```

Aug 16 09:56:17 raspberrypi systemd[1]: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services

Running **journalctl -u gassistpi.service** gives me the same output.

Lats time I had to feflash my SD  card, but I don't want to reconfigure everything again... Please make a fix for this.

Reni0013, 2020-08-16:
Another one

pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Sun 2020-08-16 16:15:43 IST, end at Sun 2020-08-16 16:47:36 IST
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noup
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noup
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib confmisc.c:1281:(snd_func_ref
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:4568:(_snd_config_eval
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:5036:(snd_config_expan
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noup
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib confmisc.c:1281:(snd_func_ref
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:4568:(_snd_config_eval
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:5036:(snd_config_expan
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noup
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server socket err = 
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server request chann
Aug 16 16:15:43 raspberrypi python[6703]: jack server is not running or cannot b
Aug 16 16:15:43 raspberrypi python[6703]: JackShmReadWritePtr::~JackShmReadWrite
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server socket err = 
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server request chann
Aug 16 16:15:43 raspberrypi python[6703]: jack server is not running or cannot b
Aug 16 16:15:43 raspberrypi python[6703]: JackShmReadWritePtr::~JackShmReadWrite
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_o
lines 1-23...skipping...
-- Logs begin at Sun 2020-08-16 16:15:43 IST, end at Sun 2020-08-16 16:47:36 IST. --
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Aug 16 16:15:43 raspberrypi python[6703]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server socket err = No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server request channel
Aug 16 16:15:43 raspberrypi python[6703]: jack server is not running or cannot be started
Aug 16 16:15:43 raspberrypi python[6703]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server socket err = No such file or directory
Aug 16 16:15:43 raspberrypi python[6703]: Cannot connect to server request channel
Aug 16 16:15:43 raspberrypi python[6703]: jack server is not running or cannot be started
Aug 16 16:15:43 raspberrypi python[6703]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
Aug 16 16:15:43 raspberrypi python[6703]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
[... cut ...]
Aug 16 16:15:43 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Aug 16 16:15:43 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Aug 16 16:15:44 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Aug 16 16:15:44 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 1584.
Aug 16 16:15:44 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Aug 16 16:15:44 raspberrypi systemd[1]: Started Google Assistant-Voice.
lines 1-54

Reni0013, 2020-08-16:
> I too got this error now again for the second time on my raspberry pi 3b after disabling autoboot by running stop and disable on systemctl gassistpi.service. When I tried to enable and start them again, I get:
> $sudo systemctl start gassistpi.service
> `Failed to start gassistpi.service: Unit gassistpi.service has a bad unit file setting.`
> 
> Running **sudo systemctl status gassistpi.service** give me:
> 
> ```
> gassistpi.service - Google Assistant-Voice
>    Loaded: bad-setting (Reason: Unit gassistpi.service has a bad unit file setting.)
>    Active: inactive (dead)
> 
> Aug 16 09:56:17 raspberrypi systemd[1]: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services
> ```
> 
> Running **journalctl -u gassistpi.service** gives me the same output.
> I can't start the assistant manually either now...
> 
> Lats time I had to reflash my SD card, but I don't want to reconfigure everything again... Please make a fix for this.

Maybe you also have 2 execstart file path under systemd > gassistpi.service file. Remove push.py path if you are not using raspberry pi zero. If it also doesn't solves your problem then better to wait for master's ( shivasiddharth's ) reply

solidssss, 2020-08-16:
That did not work. I reinstalled Raspbian again instead... but now when I run the script manually it does not ouput any text. I really need it to do it for testing. Not sure why it's not working, I can wake it with Ok google and talk to it , but the terminal does not output any text when I talk to it. It's just stuck like this:
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED
```

shivasiddharth, 2020-08-16:
> That did not work. I reinstalled Raspbian again instead... but now when I run the script manually it does not ouput any text. I really need it to do it for testing. Not sure why it's not working, I can wake it with Ok google and talk to it , but the terminal does not output any text when I talk to it. It's just stuck like this:
> 
> ```
> ON_MUTED_CHANGED:
>   {"is_muted": false}
> 
> ON_MEDIA_STATE_IDLE
> 
> ON_START_FINISHED
> ```
> 
> I'm using the same client.json and config.yaml file from before i reflashed Raspbian

If it is replying, it must be from your service running in the background. You cannot have both running.

shivasiddharth, 2020-08-16:
@Reni0013 , I am unable to replicate your issue. This is the last thing that I can suggest. Share the output of:
`sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistantreni' --device_model_id 'assistantreni-eaglesmarthub-gbvix6'`

solidssss, 2020-08-16:
> If it is replying, it must be from your service running in the background. You cannot have both running.

If I try to kill the auto-start GassistPi process, it just keeps starting. If I run sudo systemctl stop gassistpi.service   and sudo systemctl disable gassistpi.service I will most definetly get the **Loaded: bad-setting** error again for the third time.

shivasiddharth, 2020-08-17:
@solidssss Seems like you are working with pre-conceived notions. Whoever told you that `sudo systemctl stop gassistpi.service` will give a "bad-setting error" they are wrong. And, this is not the right place to discuss personalizations.

shivasiddharth, 2020-08-17:
@Reni0013 Took a second look at the journal. Your assistant is infact working. You have not setup the audio properly. That is why you are getting 
```
Aug 16 16:15:43 raspberrypi python[6703]: [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Aug 16 16:15:43 raspberrypi python[6703]: Fatal Python error: Aborted
Aug 16 16:15:43 raspberrypi python[6703]: Current thread 0x76f04ad0 (most recent call first):
```

closed 2020-08-17 by shivasiddharth

CharlesssQJ, 2021-11-10:
Run
 journalctl -u gassistpi.service
if the command line shows: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed f
delete the floowing line:
ExecStart=/home/__USER__/env/bin/python -u /home/__USER__/GassistPi/src/pushbutton.py --project-id 'created-project-id'  --device-model-id 'saved-model-id'
This fix the problem for me.
I also changed all '_USER_' to pi as well as manually typed my project-id and model-id

commit f4b38972 (2020-08-17): Update main.py

1. Remove old logs.
2. Reset log file permission.

Fixes #1033