shivasiddharth/GassistPi #1035: amixer: Cannot find the given element from control default

ashhad, 2020-08-25:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 1000000036e72503
Model		: Raspberry Pi 4 Model B Rev 1.2
pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Cannot setup audio. Using 3.5 mm jack for output and USB mic for input. This is the error.

pi@raspberrypi:~ $ sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
Renaming /home/pi/.asoundrc to /home/pi/.asoundrc.bak...
amixer: Cannot find the given element from control default

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
pi@raspberrypi:~ $ sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
Renaming /home/pi/.asoundrc to /home/pi/.asoundrc.bak...
amixer: Cannot find the given element from control default

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2020-08-25:
Post the output of:
```
aplay -l
arecord -l
```

ashhad, 2020-08-25:
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

I tried the standard google assistant from [here](https://pimylifeup.com/raspberry-pi-google-assistant/) and it works well. 

This is my .asoundrc file.
 
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:2,0"
    rate 48000
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}

shivasiddharth, 2020-08-25:
If you are Ok with that working assistant, feel free to close the issue then. If you need all the frills and fringes of the master branch, that audio config will not work with this project.

ashhad, 2020-08-25:
How do I make it work then ? This project is much better than the basic one. I bought the pi only for this.

shivasiddharth, 2020-08-25:
Do:
`sudo nano /home/pi/.asoundrc`

paste the following and save

```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:2,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
Then do:
`sudo cp /home/pi/.asoundrc /etc/asound.conf`

Run the speaker test and it should work.

npalladino100, 2020-08-26:
I believe this issue is caused by a change in one of the recent raspbian updates.  They treat each audio output as a separate ALSA device now.  This link provides more information on the change: [https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/](url).

I was trying to figure this out yesterday but I ended up just using an older version of the OS which works fine.  Here's a link to the raspbian version that I'm using: [http://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/](url)

shivasiddharth, 2020-08-26:
@npalladino100 Yup. If the entire buster release treated the devices as separate entities its easier to modify the script. I will just add a comment in the README.

closed 2020-08-26 by shivasiddharth

ashhad, 2020-08-27:
still the same issue  with amixer error. Speaker-test was working then also 

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2020-08-27:
What do you mean by still the same issue ? After doing what are you getting the amixer issue ?

Jayant-2019, 2023-03-20:
Hey @shivasiddharth i am facing the same issue. It is showing amixer: Cannot find the given element from control default issue after running the below command.
sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh

kindly help me to get install the goggle assistance.

commit 66085f66 (2020-08-26): May2020 audio

Fixes #1035