shivasiddharth/GassistPi #1037: Google Assistant fails to start in smart screen,

joshianita, 2020-08-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
https://github.com/shivasiddharth/GassistPi/tree/Google-Assistant-SDK-Smart-Screen

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  

cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000ec7b6db8
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

`
 cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"``
```
Paste your Board and OS details here.   
```

### Describe the bug:
```When i manually start the GA by typing in terminal /home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id 'replace this with the project id '--device_model_id 'replace this with the model id'
It starts up and with OK google, it opens a browser blank page if browser is open, if browser is closed it does not do anything. terminal outputs the following messages
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes/No
yes
### 2. Is the issue related to assistant installation ?
Yes/No
yes
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
no
### 4. Is the assistant service restarting automatically?
Yes/No
no
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below

/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR
```

shivasiddharth, 2020-08-28:
Why parts of the terminal contents are missing ? After conversation turn has started, you should have the command that you gave. How does the assistant respond when you give the command ? What does it say ?

joshianita, 2020-08-28:
This is what is outputted in the terminal

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR

Regards

Sanjay Joshi

joshianita, 2020-08-28:
This is what is outputted on the terminal when I manually start the assistant and then say ok Google

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR

joshianita, 2020-08-28:
This is all is seen in the terminal. The browser window also opens to a blank window and assistant logo/symbol also does not appear. 

I do not get any response from assistant.

Just when first time I manually start then I can hear the music and the Hi of assistant.

shivasiddharth, 2020-08-28:
Have you enabled the Assistant API in the google projects console ? When the assistant opens for the first time, it should have the logo. Wait for the assistant to fully start. You should get a prompt on the terminal that assistant can start conversation only after that you should start the interaction. 

I need to see the full terminal contents. What you are pasting here is only a part of the terminal contents (just the conversation part). There will be some info above this.

So did you try and give a second command after this or are you stopping with this one command. Assistant will speakout something. Assistant's volume is independent of the Pi volume. The first command should be "Hey google set volume to maximum".

Work on all these and revert back.

joshianita, 2020-08-28:
Ok, will do a fresh install in the night and send the terminal output.

Currently midway into installing Alexa from developer.amazon.com

joshianita, 2020-08-28:
ok, so did another install on a fresh raspbian buster and same error, this time GA responds by saying " cant reach internet right now, check ur modem or router and try again"  although i am connected by wifi.

this is the terminal response to manual loading and then voice commands

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'magic-assistant-xxxxxx-magic-google-assistant-xxxxxx' --project_id 'magic-assistant-xxxxxx'
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...] go away. Please use get/find descriptors from generated code or query the descriptor_pool.
[... 174 lines cut ...]
ON_ASSISTANT_ERROR

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

None
ON_ASSISTANT_ERROR

This activity shows in my google dashboard under google assistant api, but with 89% errors

[image]

shivasiddharth, 2020-08-28:
So as such the scripts are working fine. There is nothing i can do here. 

Either your network is down (pi may be connected to network, but it may not have internet access) or your api calls are getting blocked by your ISP (It has happened before). In the past we have had instances were there was heavy downloading going within the network which resulted in the same  kind of error. You could create a hotspot on your mobile and connect Pi to it and then check. 

These are just pointers for you to look upon. I cannot speculate on the cause too much.  All that I can do is leave the issue open.

joshianita, 2020-08-29:
Thanks for your insight about network. Realised i was running it through PiHole and checked blocklists but couldnt find any blocking related to G Assist, still connected it direct and g Assist started working like you said it should.

Unfortunately it shuts down unexpectedly randomly with the following error.
Went to Assistant.py and line 18 is self_shutdown = True and changed that to False but still it shuts down unexpectedly. cannot understand how to replicate shutdown .

follwing is terminal output

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'magic-assistant-xxxxxx-magic-google-assistant-xxxxxx' --project_id 'magic-assistant-xxxxxx'
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the de [... cut ...]
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}

Fatal Python error: Segmentation fault

Thread 0x76effad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 391 in main
  File "/home/pi/GassistPi/src/main.py", line 396 in <module>
Segmentation fault

shivasiddharth, 2020-08-29:
Random shutdowns are difficult to replicate that too with the segmentation fault not showing the real reason. As the
 ```
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": ""}
```
is blank, i would suspect it to be a hardware or an audio rate issue. Again, its just a speculation. Cannot do much about this.

joshianita, 2020-08-29:
Well have been playing with it trying to replicate the issue.

Found out that the issue is if Assistant is listening for a query but doesnt find anything then shuts unexpectedly.

This happens in 2 instances 

Scenario one ---- if I say the hotword Ok Google which appears on terminal like this
ON_CONVERSATION_TURN_STARTED
and then dont give any query so it returns
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": ""}  
and then shuts down unexpectedly

Second Scenario : if i make a query and after answering it GA waits for follow on queries like this ON_CONVERSATION_TURN_FINISHED:
{"with_follow_on_turn": true} 
but i dont say anything and then it returns the same 
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": ""}  
and then shuts down unexpectedly

possible solutions 
1. best way possible : to check programming that when no query received stop listening but **dont** shut down 
2. Jugaad technology : after every answer from Asisstant say "cancel" to make sure it stops listening.
3. From google Assistant settings : try and stop follow on command system where it stops listening after every answer

Sorry for my noobiness :), but yes I am a complete noob to all things programming, linux and raspnberry pi. This is my 3rd week with a Rasbperry Pi, bought 2 this month. RPi4 using as Pihole,Samba server and Kodi on Raspex OS. This RPi 3b+ planning to use as a Magic Mirror with Alexa and GAssist. 

btw any way to run both Alexa and GA concurrently as they apparently dont want to share the same Mic and Hdmi i think.

shivasiddharth, 2020-08-29:
Hmm I see what is happening.  Will try a fix.

shivasiddharth, 2020-08-29:
I think I may have fixed the issue. Delete the existing GassistPi folder, clone it again and directly manually start it. No need to run through the installation process again.

Check out my Assistants-Pi git for both Alexa and GA

closed 2020-08-29 by shivasiddharth

joshianita, 2020-08-29:
Thank you for your response

Meantime I have gone to my Google account and disabled continued
conversation so now much more reliable.

Thanks again for all your help and support.

joshianita, 2020-08-29:
Wow, you are the man.

Tried for a couple of instances of direct command as well as continued
conversation, now has a new output
No speech was deciphered.....
And then waits for Hotwire again.

Now will try to run both concurrently.

Btw amazon also has a smart screen SDK but apparently does not work on
voice hot word, need to press "a" key on keyboard, any idea if voice
commands can be started.

All thanks to gurus like u I feel great about learning new tech just for
fun, whole nearing my retirement age . In this instance I already have an
echo dot as well as home mini so these projects are just for fun 😁

shivasiddharth, 2020-08-29:
smart screen sdk with voice activation 
demo: https://www.youtube.com/watch?v=OLc5EpjWpuw
project: https://github.com/shivasiddharth/DIY-Echo-Show

joshianita, 2020-08-29:
Thanks.

Ok heard a lot about snowboy implementation, so this is how .

commit f05ec6fa (2020-08-29): Update main.py

1. Fixed crashes when no command is given.

Fixes #1037