shivasiddharth/GassistPi #1039: rasp4 sound not work after run USB MIC AND AUDIO JACK users script

lenunes, 2020-08-29:
### Which branch of GassistPi are you using ?
from Master at 08/29

### Board and OS details:    
$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03112
Serial          : 10000000db7f587b
Model           : Raspberry Pi 4 Model B Rev 1.2
--------------------------------------------------------------
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
```
Paste your Board and OS details here.   
Raspberry Pi 4 - 4GB
Raspberry Pi OS (32-bit)

### Describe the bug:
I followed the documentation in order to setup and install GassistPi on my Raspberry pi 4 with audio jack and usb mic previous working.
After INSTALL AUDIO CONFIG FILES step (with USB-MIC-JACK from Extra/May2020) rasp audio stop work. Also, after run audio config files I noticed that Audio output has only "Analog" and "Hdmi".

After ./gassist-installer.sh the following message has been shown:
"Auto modification of the service file is not feasible. Manually check your username, project id and model id..."

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.
~/projects $ tree
.
└── GassistPi
    ├── audio-drivers
    │   ├── AIY-HAT
    │   │   └── scripts
    │   │       ├── asound.conf
    │   │       ├── configure-driver.sh
    │   │       └── install-alsa-config.sh
    │   ├── CUSTOM-VOICE-HAT
    │   │   └── scripts
    │   │       ├── asound.conf
    │   │       ├── custom-voice-hat.sh
    │   │       └── install-i2s.sh
    │   ├── USB-DAC
    │   │   └── scripts
    │   │       ├── asound.conf
    │   │       └── install-usb-dac.sh
    │   ├── USB-MIC-HDMI
    │   │   └── scripts
    │   │       ├── asound.conf
    │   │       ├── configure.sh
    │   │       └── install-usb-mic-hdmi.sh
    │   ├── USB-MIC-JACK
    │   │   └── scripts
    │   │       ├── asound.conf
    │   │       └── usb-mic-onboard-jack.sh
    │   └── USB-MIC-JACK_backup
    │       └── scripts
    │           ├── asound.conf
    │           └── usb-mic-onboard-jack.sh
    ├── Extras
    │   ├── assistant.py
    │   ├── gmusicauth.py
    │   ├── IR-Sensor.py
    │   ├── May2020
    │   │   ├── USB-MIC-HDMI
    │   │   │   └── scripts
    │   │   │       ├── asound.conf
    │   │   │       ├── configure.sh
    │   │   │       └── install-usb-mic-hdmi.sh
    │   │   └── USB-MIC-JACK
    │   │       └── scripts
    │   │           ├── asound.conf
    │   │           └── usb-mic-onboard-jack.sh
    │   └── train_snowboy.py
    ├── GoogleAssistantImages
    │   ├── GoogleAssistantBars.gif
    │   ├── GoogleAssistantBarsTransparent.gif
    │   ├── GoogleAssistantDots.gif
    │   ├── GoogleAssistantDotsTransparent.gif
    │   ├── GoogleAssistant.gif
    │   └── GoogleAssistantTransparent.gif
    ├── LICENSE
    ├── Neopixel Indicator
    │   ├── Neopixel for GassistPi.fzz
    │   └── Neopixel_Google_Home.ino
    ├── README.md
    ├── Requirements
    │   ├── GassistPi-pip-requirements.txt
    │   └── GassistPi-system-requirements.txt
    ├── sample-audio-files
    │   ├── Bong.wav
    │   ├── Device-Off-Female.wav
    │   ├── Device-Off-Male.wav
    │   ├── Device-On-Female.wav
    │   ├── Device-On-Male.wav
    │   ├── Fb.wav
    │   ├── Mic-Off-Female.wav
    │   ├── Mic-Off-Male.wav
    │   ├── Mic-On-Female.wav
    │   ├── Mic-On-Male.wav
    │   ├── Pi-Close-Male.wav
    │   ├── Pi-Close.wav
    │   ├── snowboy.wav
    │   ├── Startup-Female.wav
    │   ├── Startup-Male.wav
    │   └── Tring.wav
    ├── scripts
    │   ├── gassist-installer.sh
    │   ├── service-installer.sh
    │   └── update.sh
    ├── src
    │   ├── actions.py
    │   ├── apa102.py
    │   ├── audiorecorder.py
    │   ├── colours.json
    │   ├── config.yaml
    │   ├── indicator.py
    │   ├── keywords_de.yaml
    │   ├── keywords_en.yaml
    │   ├── keywords_es.yaml
    │   ├── keywords_fr.yaml
    │   ├── keywords_it.yaml
    │   ├── keywords_nl.yaml
    │   ├── keywords_sv.yaml
    │   ├── lang.yaml
    │   ├── main.py
    │   ├── mediaplayer.py
    │   ├── pushbutton.py
    │   ├── resources
    │   │   ├── alexa_02092017.umdl
    │   │   ├── alexa.umdl
    │   │   ├── Buster-wrapper
    │   │   │   ├── snowboydetect.py
    │   │   │   └── _snowboydetect.so
    │   │   ├── common.res
    │   │   ├── ding.wav
    │   │   ├── dong.wav
    │   │   ├── jarvis.umdl
    │   │   ├── smart_mirror.umdl
    │   │   ├── snowboy.raw
    │   │   ├── snowboy.umdl
    │   │   ├── snowboy.wav
    │   │   └── Stretch-wrapper
    │   │       ├── snowboydetect.py
    │   │       └── _snowboydetect.so
    │   ├── snowboydecoder.py
    │   ├── snowboydetect.py
    │   ├── _snowboydetect.so
    │   └── youtube_search_engine.py
    ├── systemd
    │   └── gassistpi.service
    └── version

30 directories, 92 files
```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/pi/env/bin/python -u /home/pi/projects/GassistPi/src/pushbutton.py --device-model-id rasp-ghome-embedded-rasp-ghome-embedded-2hocqk --project-id rasp-ghome-embedded
```

### Paste the contents of the terminal below
```
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-08-29 03:11:59,717 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Initializing GPIOs 5 and 6 for assistant activity indication
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
ERROR (Input():snowboy-io.cc:315) Fail to open input file "/home/pi/GassistPi/src/resources/alexa.umdl"
terminate called after throwing an instance of 'std::runtime_error'
  what():  ERROR (Input():snowboy-io.cc:315) Fail to open input file "/home/pi/GassistPi/src/resources/alexa.umdl"

[stack trace: ]
Backtrace has not been implemented yet.

Aborted

```

joshianita, 2020-08-29:
admittedly i am a complete zero in all things programming and Raspberry Pi , but venturing to comment cos had the same issue myself.
Run lsusb to see ur usb devices are connected
run aplay -l to find out card number and device number of playback
run arecord -l to find out card number and device number of mic

sudo nano ~/.asoundrc 
sudo nano /etc/asound.conf

and change value of "hw :cardnumber,deviceno"
for both playback and record  and hopefully it will work

shivasiddharth, 2020-08-29:
1. The model id and project id should be within quotes. 
2. Looks like the authentication has not gone through properly.

Was there any interruption during the installation process ? Did you stop and start ?

w7boz, 2022-04-11:
Similar issue. 
Checking OS Compatability
You are running the installer on Stretch=
=Your board does not support voice control.

See attachment
[OSMC errors.txt](https://github.com/shivasiddharth/GassistPi/files/8463618/OSMC.errors.txt)
.