shivasiddharth/GassistPi #1041: RPI zero w stucks at grpcio installation

beastnani, 2020-08-30:
Using cached grpcio-1.31.0.tar.gz (20.0 MB)
Requirement already satisfied, skipping upgrade: six>=1.5.2 in ./env/lib/python3.7/site-packages (from grpcio) (1.15.0)
Skipping wheel build for grpcio, due to binaries being disabled for it.
Installing collected packages: grpcio
    Running setup.py install for grpcio ... |

closed 2020-08-30 by close-issue-app[bot]

shivasiddharth, 2020-08-30:
Raspberry Pi Zero will take time to install grpcio (a lot of time say 1 hr). It has already been discussed before. Please go through existing issues before you open a new one.

beastnani, 2020-08-30:
where is it

Electronlibre2012, 2020-10-11:
> 
> 
> but after 5 hours it does not get install pi's led dims and terminal get's stuck

Hello, for me it takes 5 hours too, and after its ok ;)