shivasiddharth/GassistPi #1043: Youtube streaming causes crash after adding Respeaker, worked fine with USB audio card.

butteryak, 2020-09-01:
### Which branch of GassistPi are you using ?
recent 8/29 master

### Board and OS details:    

Paste your Board and OS details here.   

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000008327e12b
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
When initiating gassist to play a song on youtube, gassist crashes.   It worked fine when I was previously using a USB sound card, but when I recently added a Respeaker 4-mic linear array, it now crashes when trying to play a song off youtube. 

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes, I could not find an issue similiar to mine with the same errors. 

### 2. Is the issue related to assistant installation ?
[... 214 lines cut ...]
2020-09-01 11:28:34,213 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2020-09-01 11:28:34,213 : DEBUG : Saved to /tmp/female-say.mp3
2020-09-01 11:28:35,956 : WARNING : file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/__init__.py", line 44, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2020-09-01 11:28:35,966 : DEBUG : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2020-09-01 11:28:37,872 : DEBUG : URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=sade+on+youtub&part=id%2Csnippet&key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&alt=json

shivasiddharth, 2020-09-02:
If you hide your Youtube key, I have no way of seeing if its working or not. Run the command again and post things as such without removing anything.

butteryak, 2020-09-02:
sorry, my bad, thought I was being secure ;-)  thanks much for any input you can offer :-) 

here is the log file:

2020-09-01 19:39:42,340 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2020-09-01 19:39:42,352 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2020-09-01 19:39:46,742 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2020-09-01 19:40:14,374 : DEBUG : text: Getting youtube link
2020-09-01 19:40:14,375 : DEBUG : lang: en
2020-09-01 19:40:14,375 : DEBUG : slow: False
2020-09-01 19:40:14,376 : DEBUG : lang_check: True
2020-09-01 19:40:14,376 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x6f697ae0>, <function end_of_line at 0x6f697d20>, <function abbreviations at 0x6f697e40>, <function word_sub at 0$
2020-09-01 19:40:14,377 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |¿|‥|\\)|¡|、|\\]|\\[|…|—|，|\\($
2020-09-01 19:40:14,390 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2020-09-01 19:40:16,980 : DEBUG : http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2020-09-01 19:40:16,999 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-09-01 19:40:19,159 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2020-09-01 19:40:20,001 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2020-09-01 19:40:21,301 : DEBUG : http://translate.google.com:80 "GET //translate/releases/twsfe_w_20200825_RC00/r/js/translate_m.js HTTP/1.1" 200 569982
2020-09-01 19:40:25,916 : DEBUG : langs: {'af': 'Afrikaans', 'sq': 'Albanian', 'ar': 'Arabic', 'hy': 'Armenian', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'hr': 'Croatian', 'cs': 'Czech$
2020-09-01 19:40:25,917 : DEBUG : pre-processing: <function tone_marks at 0x6f697ae0>
2020-09-01 19:40:25,918 : DEBUG : pre-processing: <function end_of_line at 0x6f697d20>
2020-09-01 19:40:25,919 : DEBUG : pre-processing: <function abbreviations at 0x6f697e40>
2020-09-01 19:40:25,923 : DEBUG : pre-processing: <function word_sub at 0x6f6a3270>
2020-09-01 19:40:25,923 : DEBUG : text_parts: 1
2020-09-01 19:40:25,929 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-09-01 19:40:27,666 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2020-09-01 19:40:27,889 : DEBUG : payload-0: {'ie': 'UTF-8', 'q': 'Getting youtube link', 'tl': 'en', 'ttsspeed': 1, 'total': 1, 'idx': 0, 'client': 'tw-ob', 'textlen': 20, 'tk': '123121.468987'}
2020-09-01 19:40:27,906 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-09-01 19:40:29,746 : DEBUG : https://translate.google.com:443 "GET /translate_tts?ie=UTF-8&q=Getting+youtube+link&tl=en&ttsspeed=1&total=1&idx=0&client=tw-ob&textlen=20&tk=123121.468987 HTT$
2020-09-01 19:40:29,756 : DEBUG : headers-0: {'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Accept-Encoding': '$
2020-09-01 19:40:29,758 : DEBUG : url-0: https://translate.google.com/translate_tts?ie=UTF-8&q=Getting+youtube+link&tl=en&ttsspeed=1&total=1&idx=0&client=tw-ob&textlen=20&tk=123121.468987
2020-09-01 19:40:29,759 : DEBUG : status-0: 200
2020-09-01 19:40:29,759 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2020-09-01 19:40:29,760 : DEBUG : Saved to /tmp/female-say.mp3
2020-09-01 19:40:31,529 : WARNING : file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/__init__.py", line 44, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
**2020-09-01 19:40:31,533 : DEBUG : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo
2020-09-01 19:40:33,523 : DEBUG : URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=sade+on+youtub&part=id%2Csnippet&key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo&alt=json**

and uncut manual start output:

 /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'gooass2-9eca2-gooass2-wqtwc3' --project_id 'gooass2-9eca2' 
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/fin [... cut ...]
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f4bad0 (most recent call first):
  File "/usr/lib/python3.7/sre_compile.py", line 247 in _compile
  File "/usr/lib/python3.7/sre_parse.py", line 431 in _parse_sub
  File "/usr/lib/python3.7/sre_parse.py", line 841 in _parse
  File "/usr/lib/python3.7/sre_parse.py", line 221 in getwidth
Aborted

shivasiddharth, 2020-09-02:
Do: 
`sudo apt-get install pulseaudio`

Restart the Pi and check again.

butteryak, 2020-09-02:
This is the result in terminal, it might be working, **but it would appear I have lost my audio output.**....mic still works though

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Sade on YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
sade on youtub
https://r1---sn-8p5n5u-p5qe.googlevideo.com/videoplayback?expire=1599037323&ei=KgtPX9yCPcf2D9uLkMAM&ip=184.21.206.140&id=o-AEzaKbWGee4AklKnB2eW9O5NyWPAfU--Po6FsP6ReplI&itag=251&source=youtube&requiressl=yes&mh=oq&mm=31%2C29&mn=sn-8p5n5u-p5qe%2Csn-vgqs7nes&ms=au%2Crdu&mv=m&mvi=1&pl=24&gcr=us&initcwndbps=58750&vprv=1&mime=audio%2Fwebm&ns=jekamQTRC6nfxt-1TbcnWmoE&gir=yes&clen=3976489&dur=241.061&lmt=1574971033150076&mt=1599015637&fvip=4&keepalive=yes&c=WEB&txp=5531432&n=pYdhb7ExsFiGorM3DG&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cgcr%2Cvprv%2Cmime%2Cns%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRQIgfqOzzREyHZ2fHgy7JO-q7P1qAAv-LryRcrwf1qdVyckCIQDiSJ62oJLFOSwcko0vgoHc8Pc-QzI1sXeT6Qbyt5Cyww%3D%3D&sig=AOq0QJ8wRQIhAOjqEUcoxFuwEcMn0Cfu8GYaJp13RtSe5uXKZCjUNAZ-AiA50KMI-qoccbSpAgr50_EHLpl0Z4PC-AHbrw1aae7ipQ==&ratebypass=yes
/home/pi/env/lib/python3.7/site-packages/vlc.py:1825: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  if not any(getargspec(callback)[:2]):  # list(...)
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

shivasiddharth, 2020-09-02:
use alsaconfig on your terminal and set the pulseaudio or audio out volume to maximum.

butteryak, 2020-09-02:
alsaconfig command not found, Perhaps you mean "alsamixer"  I assumed so, and opened alsamixer, and set pulseaudio volume to 90%.  So, the result is that audio works, but not out of the Respeaker output, but the standard 3.5mm jack on the Pi.  I'm guessing you might want to see the 
**asound.conf config:**   I'm guessing there is an issue with how asound.conf is setup?  I don't know enough about the more complex configs, so thanks much for your help+++  maybe I'm incorrect in that assumption, please let me know if there is anything else that you might need to help me troubleshoot. much appreciated+++

defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "dmixer"
    capture.pcm "multiapps"
}

pcm.multiapps {
     type plug
     slave.pcm {
         type dsnoop
         slave {
             rate 48000
             format S32_LE
             pcm "hw:seeed8micvoicec"
         }
         ipc_key 666666
     }
 }

pcm.dmixer {
    type plug
    slave {
        pcm {
            type dmix
            ipc_key 555555
            slave {
                pcm "hw:seeed8micvoicec"
                format S32_LE
                channels 8
            }
            bindings {
                0 0
                1 1
                2 2
                3 3
                4 4
                5 5
                6 6
                7 7
            }
        }
        channels 8
        format S32_LE
        rate 48000
    }
    ttable.0.0 1
    ttable.1.1 1
    ttable.0.2 1
    ttable.1.3 1
    ttable.0.4 1
    ttable.1.5 1
    ttable.0.6 1
    ttable.1.7 1
}

**as well as aplay -L:**

null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
multiapps
dmixer
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dmix:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Hardware device with all software conversions
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Hardware device with all software conversions
usbstream:CARD=seeed8micvoicec
    seeed-8mic-voicecard
    USB Stream Output

**here also aplay -l**

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed8micvoicec [seeed-8mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-0035-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2020-09-02:
Is your Respeaker 4 Mic HAT different from the 4 Mic HAT that has LEDs and just 4 mics ? The Respeaker 4 Mic HAT that i know does not have a audio jack.

butteryak, 2020-09-02:
aplay --device plughw:CARD=seeed8micvoicec,DEV=0 a.wav    outputs test wav to Respeaker   (just FYI)

butteryak, 2020-09-02:
it is this one...

https://wiki.seeedstudio.com/ReSpeaker_4-Mic_Linear_Array_Kit_for_Raspberry_Pi/

shivasiddharth, 2020-09-03:
This the 8 Mic Hat. I need to re-write the audio config files for this HAT. If you can try and test the config files, we can have this fixed.

shivasiddharth, 2020-09-03:
The audio config files that you have posted is not formatted properly. Can you paste it properly formatted ?

shivasiddharth, 2020-09-03:
Uninstall the existing seeed card installation using the uninstall option. Then delete the seeed-voicecard folder. 

Remove pulseaudio using:
`sudo apt-get purge pulseaudio`

Now try the fix.
`git clone https://github.com/shivasiddharth/seeed-voicecard -b 6mic-8mic-fix`

Go through the usual seeed-voicecard installation process.

Reboot and try the Google Assistant.

butteryak, 2020-09-03:
Ok, I've done what you suggested.   As of now the soundcard and mics are working.   Youtube and radio streaming are not working though, I tried youtube a couple of times, and I got two different errors

**youtube-first error:**

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Sade on YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
sade on youtub
https://r4---sn-8p5n5u-525e.googlevideo.com/videoplayback?expire=1599173866&ei=iSBRX6PgPMSfDffOhJAH&ip=184.21.206.140&id=o-AKpVCF6zXa5u8UtFFb19y11MS5FWRDwCcCKlhfFXrT1s&itag=251&source=youtube&requiressl=yes&mh=oq&mm=31%2C29&mn=sn-8p5n5u-525e%2Csn-vgqsknez&ms=au%2Crdu&mv=m&mvi=4&pl=24&gcr=us&initcwndbps=143750&vprv=1&mime=audio%2Fwebm&gir=yes&clen=3976489&dur=241.061&lmt=1574971033150076&mt=1599152208&fvip=4&keepalive=yes&c=WEB&txp=5531432&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cgcr%2Cvprv%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRQIge8RkXsP7kK2Tq6xEARkOkHYz1uSdHxl_ws1Spc_UwXQCIQCntSqHlm9HyfW2NtkLSqcQ3qN9ehy7D25zDm3JhAVRRQ%3D%3D&sig=AOq0QJ8wRQIhAN22sLvuIoJSjiqYasKBmtOTSc8OYezxtNBUQ2Ua6vQWAiBHl7Ed3OUVWS_gZgNrYQESyteoYC_nAAkOQolnT630PQ==&ratebypass=yes
[02c135c8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
/home/pi/env/lib/python3.7/site-packages/vlc.py:1825: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  if not any(getargspec(callback)[:2]):  # list(...)
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

**youtube-second error:** 

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Sade on YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
sade on youtub
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x5defe460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 32 in get
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 157 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x5d2ff460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x6771b460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
[... 204 lines cut ...]
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/__init__.py", line 44, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2020-09-03 10:11:47,992 : DEBUG : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo
2020-09-03 10:11:55,096 : DEBUG : URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=sade+on+youtub&part=id%2Csnippet&key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo&alt=json

shivasiddharth, 2020-09-03:
The first youtube error is not actually an error. Its just a warning. You can see that an audio stream has been produced. And it should have been played normally. Same is the case with the radio station as well.

The second one is actually the error.

Can you paste the audio config files here ?

butteryak, 2020-09-03:
you bet, here yah go:

defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:seeed8micvoicec"
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:seeed8micvoicec"
        channels 8
    }
    ipc_key 666666
}

butteryak, 2020-09-03:
actually, it's interesting.  I tried youtube about 10 times, it actually played once.  but all other times, radio/youtube appears to be playing but not sound:  same result:

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Sade on YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
sade on youtub
https://r4---sn-8p5n5u-525e.googlevideo.com/videoplayback?expire=1599180716&ei=TDtRX5PoMs7kwQHB04Vg&ip=184.21.206.140&id=o-AEZNCSnhDYwME1C-cu2MPLa7GpqNndzN-3eIo4eTekRg&itag=251&source=youtube&requiressl=yes&mh=oq&mm=31%2C29&mn=sn-8p5n5u-525e%2Csn-vgqsknez&ms=au%2Crdu&mv=m&mvi=4&pl=24&gcr=us&initcwndbps=137500&vprv=1&mime=audio%2Fwebm&gir=yes&clen=3976489&dur=241.061&lmt=1574971033150076&mt=1599159047&fvip=4&keepalive=yes&beids=9466588&c=WEB&txp=5531432&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cgcr%2Cvprv%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRQIgFOZXAeFmp-tsh_mz_YMHabNWtRHgKxKfe1KRv9RK8p4CIQDZ3mlEOjVTjhKVLq5jwn0UEGS2Q60Twnw7AEM0elXBOA%3D%3D&sig=AOq0QJ8wRQIhAN0XDgkj9-CqHZGzfudONvSAX9j0si0b0HlaVQOk93K2AiAKMaSuE5wT-yiQBXH0ScEUwzgzo77SoEBslXgDvLuffA==&ratebypass=yes
[01b575e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
/home/pi/env/lib/python3.7/site-packages/vlc.py:1825: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  if not any(getargspec(callback)[:2]):  # list(...)
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

butteryak, 2020-09-03:
It's rather interesting really.....
youtube worked a few times, maybe 1 outa 15 or so. **but mostly getting this error:**

python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x60afe460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x624ff460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x67729460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f79ad0 (most recent call first):
  File "/usr/lib/python3.7/sre_parse.py", line 255 in get
  File "/usr/lib/python3.7/sre_parse.py", line 529 in _parse
  File "/usr/lib/python3.7/sre_parse.py", line 426 in _parse_sub
  File "/usr/lib/python3.7/sre_parse.py", line 816 in _parse
  File "/usr/lib/python3.7/sre_parse.py", line 426 in _parse_sub
  File "/usr/lib/python3.7/sre_parse.py", line 938 in parse
  File "/usr/lib/python3.7/sre_compile.py", line 764 in compile
  File "/usr/lib/python3.7/re.py", line 286 in _compile
  File "/usr/lib/python3.7/re.py", line 173 in match
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/jsinterp.py", line 88 in interpret_expression
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/jsinterp.py", line 56 in interpret_statement
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/jsinterp.py", line 258 in resf
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1402 in <lambda>
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1338 in _extract_signature_function
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1426 in _decrypt_signature
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 2107 in _real_extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/common.py", line 530 in extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/YoutubeDL.py", line 797 in extract_info
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 40 in _fetch_basic
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_shared.py", line 97 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 31 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/pafy.py", line 124 in new
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 125 in youtube_stream_link
  File "/home/pi/GassistPi/src/actions.py", line 1175 in YouTube_No_Autoplay
  File "/home/pi/GassistPi/src/main.py", line 738 in custom_command
  File "/home/pi/GassistPi/src/main.py", line 389 in process_event
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

**the radio urls mostly appear to be working, about 1 out of 10 tries does not (I'm chalking this up to internet being slow) typically I see this output when calling radio:**

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Radio 1"}

stopping vlc
http://live.str3am.com:2470/live32
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
[0156c0e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
/home/pi/env/lib/python3.7/site-packages/vlc.py:1825: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  if not any(getargspec(callback)[:2]):  # list(...)
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

[5e003c48] mpeg4audio demux packetizer: AAC channels: 1 samplerate: 24000

butteryak, 2020-09-10:
I've been working with this a bit, but with no luck.   Radio seems to work fine.  although every time I try to play a song on youtube, I'm consistently getting this error:

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
sade on youtub
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x5dffe460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 75 in RunDetection
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 162 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x62834460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x676e0460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f29ad0 (most recent call first):
  File "/usr/lib/python3.7/re.py", line 183 in search
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/common.py", line 987 in _search_regex
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1398 in _parse_sig_js
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1326 in _extract_signature_function
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1426 in _decrypt_signature
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 2107 in _real_extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/common.py", line 530 in extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/YoutubeDL.py", line 797 in extract_info
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 40 in _fetch_basic
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_shared.py", line 97 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 31 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/pafy.py", line 124 in new
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 125 in youtube_stream_link
  File "/home/pi/GassistPi/src/actions.py", line 1175 in YouTube_No_Autoplay
  File "/home/pi/GassistPi/src/main.py", line 738 in custom_command
  File "/home/pi/GassistPi/src/main.py", line 389 in process_event
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

Once, I got this slightly different error:

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Sade YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...
[... 10 lines cut ...]
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76fb1ad0 (most recent call first):
  File "/usr/lib/python3.7/re.py", line 183 in search
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/common.py", line 987 in _search_regex
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1398 in _parse_sig_js
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1326 in _extract_signature_function
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 1426 in _decrypt_signature
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/youtube.py", line 2107 in _real_extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/extractor/common.py", line 530 in extract
  File "/home/pi/env/lib/python3.7/site-packages/youtube_dl/YoutubeDL.py", line 797 in extract_info
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 40 in _fetch_basic
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_shared.py", line 97 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/backend_youtube_dl.py", line 31 in __init__
  File "/home/pi/env/lib/python3.7/site-packages/pafy/pafy.py", line 124 in new
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 125 in youtube_stream_link
  File "/home/pi/GassistPi/src/actions.py", line 1175 in YouTube_No_Autoplay
  File "/home/pi/GassistPi/src/main.py", line 738 in custom_command
  File "/home/pi/GassistPi/src/main.py", line 389 in process_event
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Aborted

shivasiddharth, 2020-09-10:
I am held up with personal work. Fixing this issue needs good amount of time, which i dont have now.

butteryak, 2020-09-10:
thanks for the update+ I totally understand, thanks for your efforts so far.   

I have a bit of time to play around with it, if you have a minute, if you could point me in the direction of what I could look at to make it work, maybe I can see what I can figure out.  if not no worries.

butteryak, 2020-10-19:
After quite a bit of poking around and trying some things, I've got the Seeed card working, and I can play radio (still having some issues with youtube, but I think it's a different issue, not sure)  I know you said you are busy, so just keeping you updated on my progress.

I was figuring the issue was that radio and youtube needed pulseaudio to work properly so I made sure it was installed, I think the key here was to set my sinks and sources in pulse audio.  

```
Server String: /run/user/1000/pulse/native
Library Protocol Version: 32
Server Protocol Version: 32
Is Local: yes
Client Index: 11
Tile Size: 65496
User Name: pi
Host Name: raspberrypi
Server Name: pulseaudio
Server Version: 12.2
Default Sample Specification: s16le 2ch 44100Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.platform-soc_sound.multichannel-output
Default Source: alsa_input.platform-soc_sound.multichannel-input
Cookie: b9b3:d72d

```
Also making sure asoundrc was setup properly:   
```
defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "dmixer"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixer {
    type plug
    slave {
        pcm {
            type dmix
            ipc_key 555555
            slave {
                pcm "hw:seeed8micvoicec"
                period_time 0
                period_size 1024
                buffer_size 8192
                format S32_LE
                channels 8
            }
            bindings {
                0 0
                1 1
                2 2
                3 3
                4 4
                5 5
                6 6
                7 7
            }
        }
        channels 8
        format S32_LE
        rate 48000
[... 59 lines cut ...]
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ModuleNotFoundError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/__init__.py", line 44, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.7/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    "file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth"
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2020-10-19 14:21:11,100 : DEBUG : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo
2020-10-19 14:21:13,722 : DEBUG : URL being requested: GET https://www.googleapis.com/youtube/v3/search?q=sade+on+youtub&part=id%2Csnippet&key=AIzaSyCjKvTf8C0cAc3lfXVmdmpyCPl2Uz_l4Qo&alt=json
```

Basically I get the error above, and Gassist restarts.

shivasiddharth, 2020-12-03:
Howdy. This issue was opened long time back.  I have updated the seeed installer with your modified audio config. Thanks. Do you need help with any stuff or are we good to close this ?