shivasiddharth/GassistPi #1045: Music from Youtube or Deezer randomly has crackle

william2313, 2020-09-03:
### Which branch of GassistPi are you using ?
Master

### Board and OS details:    
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03112
Serial          : 100000001d84cd69
Model           : Raspberry Pi 4 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
About half of the time, when I play music from Youtube (or Deezer) the sound has a crackle in it (at the peaks) as if the volume is too loud for the speakers. However, the volume only gets slightly louder than normal and is certainly not too loud for my speakers. I can play music from my Pi in other ways without problems and about half of the time the music playing through GassistPi also sounds fine.  
I'm having great trouble finding out what causes this since it appears kind of randomly.  Sometimes I can start music 2 or 3 times in a row without crackle, then the next time there is crackle again. Sometimes there is crackle many times in a row. Music starts either with or without crackle and it doesn't seem to start or stop halfway a song. It only changes when I stop music and then give a new music command. 
The voice from GA sounds fine. Also news is always being played just fine. 
Changing the assistant volume has no effect on this. I also tried changing the volume in the .mediavolume.json file from 90 to 30. This caused the volume to be much lower but still included the crackle (also at lower volume).
I suspect the problem has something to do with how VLCPlayer manages volume.
BTW, I use PulseAudio. 

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
It doesn't crash and doesn't give any errors.

shivasiddharth, 2020-09-03:
Random stuffs are always difficult to solve and I cannot guarantee you a solution. What is your audio setup ?

william2313, 2020-09-03:
I use USB-MIC-JACK.
I have applied the May2020 audiofix.

shivasiddharth, 2020-09-03:
Cant do much. Will leave the issue open to see if someone else reports something. It would suggest you to try and change the audio to HDMI to see if the crackling appears.

william2313, 2020-09-03:
I use the Pi headless, so can't test HDMI. 
Could you perhaps give any pointers to where in the GassistPi / assistant code or config files the VLC player's volume / audio settings are being controlled?