shivasiddharth/GassistPi #1047: No module named 'kodijson'

MWHM2120, 2020-09-03:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master
### Board and OS details:    
raspberri pi 3 b +
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   

 cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000000c9db511

### Describe the bug:
when i need to start the GA i take this error 

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
.
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
.
```

### Paste the contents of the terminal below
```
.
```

shivasiddharth, 2020-09-04:
I am closing this issue for not sticking to the Git guidelines. 

1. The issue template has not been filled in properly.
2. This issue has already been discussed in a number of issues already. The user has clearly not made an attempt to go through the closed issues. Had you searched for "ModuleNotFoundError: No module named 'kodijson'" in the issues a number of issues would have popped up.

closed 2020-09-04 by shivasiddharth

MWHM2120, 2020-09-04:
sir but the solution in #1027 is for just GA branch i am asking for master branch