shivasiddharth/GassistPi #1049: ModuleNotFoundError: No module named 'kodijson'

MWHM2120, 2020-09-06:
IMPORTANT NOTICE
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED

Which branch of GassistPi are you using ?
master

Board and OS details:
raspberri pi 3 b +
Open a terminal and run

cat /proc/cpuinfo
cat /etc/os-release
Paste your Board and OS details here.

cat /proc/cpuinfo
processor : 0
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 1
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 2
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 3
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : BCM2835
Revision : a020d3
Serial : 000000000c9db511

Describe the bug:
when i need to start the GA i take this error

Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 20, in
from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

Is the issue related to assistant installation ?
Yes
If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 20, in
from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

If answer to question 2 is No, then proceed further.
3. Does the assistant service start normally?
No

Is the assistant service restarting automatically?
No
If answer to question 4 is Yes, then paste the contents of the terminal below.
.
If answer to question 4 is No, then proceed further.
Manually start the assistant. For guidelines to start the assistant manually refer this
Paste the command below, that crashed the assistant
.
Paste the contents of the terminal below

MWHM2120, 2020-09-06:
sir i tried todo the same in #80 but stil having the same error 

pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ sudo pip install kodi-json
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: kodi-json in /usr/local/lib/python2.7/dist-packages (1.0.0)
Requirement already satisfied: requests in /usr/lib/python2.7/dist-packages (from kodi-json) (2.21.0)
(env) pi@raspberrypi:~ $ 
(env) pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'indea-95833-myasisstant-e6dz07' --project_id 'indea-95833'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'
(env) pi@raspberrypi:~ $ sudo pip install kodi-json

shivasiddharth, 2020-09-06:
OS details have not been provided.

MWHM2120, 2020-09-06:
what can i give you more then the name ??

shivasiddharth, 2020-09-07:
I dont see any information at all regarding the OS.

MWHM2120, 2020-09-07:
i am using raspbian 

Raspberry Pi OS (previously called Raspbian) is the Foundation’s official supported operating system. You can install it with NOOBS or download the image below and follow our installation guide.

Raspberry Pi OS comes pre-installed with plenty of software for education, programming and general use. It has Python, Scratch, Sonic Pi, Java and more.

The Raspberry Pi OS with Desktop image contained in the ZIP archive is over 4GB in size, which means that these archives use features which are not supported by older unzip tools on some platforms. If you find that the download appears to be corrupt or the file is not unzipping correctly, please try using 7Zip (Windows) or The Unarchiver (Macintosh). Both are free of charge and have been tested to unzip the image correctly.

is that Enough ??

shivasiddharth, 2020-09-07:
Remove your SD card from Pi, erase it load the OS again and go through the installation again.

I dont want text book definitions and I do not entertain arrogance or sarcasms here. Instead of pasting text book definitions, you could have just said Buster or Stretch or whatever with the date of OS release that you are using.

closed 2020-09-07 by shivasiddharth