shivasiddharth/GassistPi #1053: ModuleNotFoundError: No module named 'kodijson'

MWHM2120, 2020-09-07:
IMPORTANT NOTICE
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED

Which branch of GassistPi are you using ?
master

Board and OS details:
raspberri pi 3 b +
Open a terminal and run

cat /proc/cpuinfo
cat /etc/os-release
Paste your Board and OS details here.

cat /proc/cpuinfo
processor : 0
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 1
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 2
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 3
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : BCM2835
Revision : a020d3
Serial : 000000000c9db511

Describe the bug:
when i need to start the GA i take this error

Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 20, in
from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

Is the issue related to assistant installation ?
Yes
If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 20, in
from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'

If answer to question 2 is No, then proceed further.
3. Does the assistant service start normally?
No

Is the assistant service restarting automatically?
No
If answer to question 4 is Yes, then paste the contents of the terminal below.
.
If answer to question 4 is No, then proceed further.
Manually start the assistant. For guidelines to start the assistant manually refer this
Paste the command below, that crashed the assistant
.
Paste the contents of the terminal below

MWHM2120, 2020-09-07:
sir i am sorry to be kind of annoying but i am a young boy and it takes a long time to reach for me to this level something like 2 years and i still a junior and i dont know what is Buster or Stretch means i am annoying you alittle bit because i think i am to close to finish the project 

pls tell me if there any another any that you menshend to me because i have alot important things in the sd card 

help me ASAP

shivasiddharth, 2020-09-08:
Please stop opening new issues again and again. 

I see that you are using the Pi 3 B+ you would have used the Stretch release of the OS before migrating to Buster. If you are not familiar with what the Buster is, then troubleshooting is going to become really difficult. The master branch is for more advanced usage. If you want something simpler try the https://github.com/shivasiddharth/GassistPi/tree/Just-Google-Assistant branch.

closed 2020-09-08 by shivasiddharth